ucb-bar/chipyard #486: Using a custom cpu core in chipyard

KlepD-SAL, 2020-03-17:
Hi,

I am currenty planning to implement a custom (simple) RISC-V CPU core and using the rocket chip / chipyard SoC environment (Busses, Debugger, ROM, Bridges, ...) to make it usable in real-life applications (targeting FPGAs).
To keep it simple in the beginning, the plan is to skip "advanced" features like virtual memory and branch prediction.

The documentation states that the Rocket Core can be replaced with a BOOM core or any custom core but I could not find any further documentation on how to start or proceed with the custom one.

The question is where to start in your environment:

- Can I use the RocketTile and just replace the core with my own or do I have to create a custom Tile like with the Ariane core?

- If I can use it, what is the best way to get the interface my custom core will be attatched to, so I can build up on that? Or should I choose a totally different approach?

- Can the RocketTile be simplified to skip/ignore all logic concerning the "advanced" features?

Any hint or help will be helpful!

Thanks a lot and cheers,
KlepD

alonamid, 2020-04-29:
I would recommend looking at the Ariane integration PR (#448 ), especially if you're integration a Verilog core. 
In particular, you should look at how ArianeTile, ArianeTileKey and ArianTileParams are defined.

abejgonzalez, 2020-04-29:
I'll also note that #505 modifies how Ariane is integrated slightly.

closed 2020-08-05 by alonamid