ucb-bar/chipyard #497: Verilator simulation does not dump waveform

abdullahyildiz, 2020-03-25:
I used `make debug` command in `$CHIPYARD/sims/verilator`
  to elaborate `RocketConfig-debug` to produce the executable `simulator-chipyard-RocketConfig-debug`.

When I run the following command, no waveform is dumped into VCD file:

```
$ /eda/riscv/chipyard/sims/verilator/simulator-chipyard-RocketConfig-debug -v rv64ui-p-add.vcd /eda/riscv/target/share/riscv-tests/isa/rv64ui-p-add

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 44607
[UART] UART0 is here (stdin/stdout).

```
The size of the generated VCD file is zero.

`-rw-rw-r-- 1 edaman edaman        0 Mar 25 13:58 rv64ui-p-add.vcd`

colinschmidt, 2020-03-25:
I don't believe you should have a space between `-v` and the name of your vcd.
See how the makefile specifies it here: https://github.com/ucb-bar/chipyard/blob/2c0928cb9c7c7f56f2d362c0a57d05561a872fa9/sims/verilator/Makefile#L34

abdullahyildiz, 2020-03-25:
@colinschmidt,

Yes, you're right. I missed that.

closed 2020-03-25 by abdullahyildiz