ucb-bar/chipyard #514: Including Prerequisites in "Quick Start" or Script

Matthias-Raudonis, 2020-04-15:
**Impact**: software

**Description**

Following the Quick Start guide from a clean Ubuntu 18 LTS install, one encounters repeated compile errors because some packages are missing. 

In "1.4. Initial Repository Setup" I found some prerequisites are mentioned. 

I am suggesting a reference from the Quick Start Section to the above section. Why not make a script and include it into the Quick Start Process?

**What is a motivating example for changing the behavior?**
It takes a much shorter time to compile if there are no repeated errors.

Matthias-Raudonis, 2020-04-17:
Note: This was the cause of issues #455 and #436

alonamid, 2020-04-27:
Fixed by #518

closed 2020-04-27 by alonamid