ucb-bar/chipyard #515: Required Prerequisites for clean install

Matthias-Raudonis, 2020-04-15:
In https://chipyard.readthedocs.io/en/dev/Chipyard-Basics/Initial-Repo-Setup.html#requirements the  dependencies are listed.

In https://github.com/riscv/riscv-gnu-toolchain#prerequisites there is another listing with different Prerequisites.

Then, there seem to be additional packages like pkg-config seem to be also missing.

Shouldn't the chipyard have at least the dependencies of the toolchain?

Which are all the packages that are required for a clean install?

alonamid, 2020-04-29:
When I looked into this in the past, I believe some of the requirements of the GNU toolchain were included within other requirements listed in the Chipyard page.
Have you installed all the dependencies listed within the Chipyard initial setup docs, and still encountered a dependency error?

faddat, 2020-08-09:
To make it easier for users to set up an environment, I have been trying to use precompiled binaries, and that has led me to Arch LInux.  Arch has nice fresh builds of the GNU toolchain.

closed 2020-08-17 by alonamid