ucb-bar/chipyard #524: Error simulating  with verilator  in branch nvdla-integration

bg193, 2020-04-22:
I encountered the following error when running make in /sims/verilator. In nvdla-integration branch.

%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347042:1: Cannot find file containing module: 'NV_BLKBOX_SRC0_X'
347042 | NV_BLKBOX_SRC0_X testInst_mbist_Wa_w0_0 (.Y(pre_mbist_Wa_w0_0));
       | ^~~~~~~~~~~~~~~~
%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347042:1: This may be because there's no search path specified with -I<dir>.
347042 | NV_BLKBOX_SRC0_X testInst_mbist_Wa_w0_0 (.Y(pre_mbist_Wa_w0_0));
       | ^~~~~~~~~~~~~~~~
        ... Looked in:
             NV_BLKBOX_SRC0_X
             NV_BLKBOX_SRC0_X.v
             NV_BLKBOX_SRC0_X.sv
             /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/NV_BLKBOX_SRC0_X
             /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/NV_BLKBOX_SRC0_X.v
             /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/NV_BLKBOX_SRC0_X.sv
%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347043:1: Cannot find file containing module: 'AN2D4PO4'
347043 | AN2D4PO4 UJ_DFTQUALIFIER_mbist_Wa_w0_0 (.Z(mbist_Wa_w0[0]), .A1(pre_mbist_Wa_w0_0), .A2(DFT_clamp) );
       | ^~~~~~~~
%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347045:1: Cannot find file containing module: 'NV_BLKBOX_SRC0_X'
347045 | NV_BLKBOX_SRC0_X testInst_mbist_Wa_w0_1 (.Y(pre_mbist_Wa_w0_1));
       | ^~~~~~~~~~~~~~~~
%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347046:1: Cannot find file containing module: 'AN2D4PO4'
347046 | AN2D4PO4 UJ_DFTQUALIFIER_mbist_Wa_w0_1 (.Z(mbist_Wa_w0[1]), .A1(pre_mbist_Wa_w0_1), .A2(DFT_clamp) );
       | ^~~~~~~~
%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347048:1: Cannot find file containing module: 'NV_BLKBOX_SRC0_X'
347048 | NV_BLKBOX_SRC0_X testInst_mbist_Wa_w0_2 (.Y(pre_mbist_Wa_w0_2));
       | ^~~~~~~~~~~~~~~~
%Error: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:347049:1: Cannot find file containing module: 'AN2D4PO4'
347049 | AN2D4PO4 UJ_DFTQUALIFIER_mbist_Wa_w0_2 (.Z(mbist_Wa_w0[2]), .A1(pre_mbist_Wa_w0_2), .A2(DFT_clamp) );

abejgonzalez, 2020-04-22:
Hi, it looks like you need to change what files are included in the `preprocessed` file (all files should be included by this makefile: https://github.com/ucb-bar/nvdla-wrapper/blob/master/src/main/resources/Makefile). Double-check that you are using the right NVDLA hash/repo.

With that being said, this branch is still in active development and is very unstable. Until it is fully tested and working I would advise against using that branch.

bg193, 2020-04-23:
@abejgonzalez 
I checked my local code , its same with  https://github.com/ucb-bar/nvdla-wrapper. 
 
I did a test to change vsrc.mk: https://github.com/ucb-bar/nvdla-wrapper/blob/master/src/main/resources/vsrc.mk

NVDLA_RAM_SUBDIR = $(NVDLA_RAMS)
to
NVDLA_RAM_SUBDIR = fpga/model 

Then this step pass，but get other fail:

%Warning-UNSIGNED: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:134032:43: Comparison is constant due to unsigned arithmetic
                                                                                                                                                            : ... In instance TestHarness.dut.system.nvdla.u_nvdla.nvdla_top.u_partition_o.u_NV_NVDLA_cdp.u_wdma
134032 | assign dat0_fifo0_rd_prdy = dat0_rdy & (0 <= req_chn_size);
       |                                           ^~
%Warning-UNSIGNED: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:134257:83: Comparison is constant due to unsigned arithmetic
                                                                                                                                                            : ... In instance TestHarness.dut.system.nvdla.u_nvdla.nvdla_top.u_partition_o.u_NV_NVDLA_cdp.u_wdma
134257 | assign dat0_vld  = dat_en & dat0_fifo_rd_pvld & !(is_last_beat & (is_beat_num_odd < 0)); 
       |                                                                                   ^
%Warning-UNSIGNED: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:134258:73: Comparison is constant due to unsigned arithmetic
                                                                                                                                                            : ... In instance TestHarness.dut.system.nvdla.u_nvdla.nvdla_top.u_partition_o.u_NV_NVDLA_cdp.u_wdma
134258 | assign dat0_rdy  = dat_en & dat_rdy & !(is_last_beat & (is_beat_num_odd < 0)); 
       |                                                                         ^
%Warning-CASEINCOMPLETE: /home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/nvdla_large.preprocessed.v:69168:5: Case values incompletely covered (example pattern 0x1)
69168 |     case(reg2dp_pixel_format)
      |     ^~~~
%Error: Exiting due to 8281 warning(s)
Makefile:119: recipe for target '/home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/VTestHarness.mk' failed
make: *** [/home/bg193/works/chipyard2/sims/verilator/generated-src/chipyard.TestHarness.LargeNVDLARocketConfig/chipyard.TestHarness.LargeNVDLARocketConfig/VTestHarness.mk] Error 1

abejgonzalez, 2020-04-25:
Ok I should have addressed the issue in the latest `nvdla-integration` branch. Let me know if you have any more issues.

closed 2020-05-07 by bg193