ucb-bar/chipyard #527: centos-req.sh for centos6?

austinharris, 2020-04-24:
The firesim documentation says to use the FPGA 1.6.0 AMI for the manager instance which is centos6. I had to adjust the centos-req.sh script to install git216/git222 instead of git, and I also needed the kmod package in order for linux to build. Is 1.6.0 still the recommended version, or could we update the centos-req.sh script to support both?

alonamid, 2020-04-24:
I think we prepared the centos-req script based on centos7.
When you set up a FIreSim manager on an EC2 instances, you need to run the FireSim AWS machine launch script anyways when you set up the instance, even before you clone the repo (https://docs.fires.im/en/latest/Initial-Setup/Setting-up-your-Manager-Instance.html). That script includes the getting the proper version of git for centos6. 
Do you use centos6 in any other non-FireSim scenario?

sagark, 2020-04-24:
To clarify Alon's answer, the way I'd recommend using chipyard on AWS instances for FireSim is to skip the chipyard system requirements section and instead follow the instructions for setting up a FireSim manager instance here: https://docs.fires.im/en/latest/Initial-Setup/Setting-up-your-Manager-Instance.html#launching-a-manager-instance
This also assumes you've followed the prior steps in the FireSim documentation for setting up your AWS account. If not you should follow all the steps in the FireSim docs up to and including the aforementioned link. At that point, you'll have an instance that contains all the dependencies for running Chipyard and FireSim.

Then, you can continue with chipyard setup from here: https://chipyard.readthedocs.io/en/latest/#setting-up-the-chipyard-repo without having to install any additional prerequisites.

Thanks for pointing this discrepancy, we'll clarify this in the docs.

austinharris, 2020-04-24:
I did do the firesim setup first, and then tried to run firesim-setup.sh --fast which failed building linux for the modinfo command (I guess by default /usr/sbin isn't on the path? I don't think I did anything special beyond launch the default AMI). I naively ran the centos-req.sh script thinking perhaps there was a package in there I was missing and that's when I ran across the git issues. I guess the solution is just to put modinfo on your path.

FWIW I also went through most of the firesim setup before realizing that if I'm using a modified chipyard it is preferable (?) to start from a chipyard repo that has the firesim submodule rather than the other way around.

alonamid, 2020-04-24:
Just to clarify, are you doing the firesim setup with Chipyard, or as an independent repository? (aka "firesim-as-top").

austinharris, 2020-04-24:
I went through just the firesim setup to the point where I was able to launch a build and then went to the chipyard docs which recommended initializing FireSim as a library in Chipyard. I'm fine with either way, it was just not clear to me which way to go with when looking through both sets of docs.

sagark, 2020-05-09:
@austinharris were you able to get things running on the FPGA Dev AMI?

austinharris, 2020-05-09:
Yes, thanks!

closed 2020-05-09 by austinharris