ucb-bar/chipyard #530: sourcing env.sh in zsh after 1.2.0 update is very slow

baltazarortiz, 2020-04-26:
**Impact**: software

**What is the current behavior?**
Sourcing `env.sh` after updating to 1.2.0 runs for ~20-30 seconds and then outputs `env-riscv-tools.sh:5: job table full` before finishing. Hitting control-C after a second or two still results in everything being sourced properly.

**What is the expected behavior?**
The script should return near-instantaneously as it did pre-update.

**Please tell us about your environment:**

* chipyard 1.2.0 tag + a few local changes since, but this started happening as soon as I updated to 1.2.0. I did update from a 1.1.0 repo with local changes, but they were all in tests/generators, nothing to do with the scripts.
* ubuntu 18.04
* zsh 5.4.2 (with oh-my-zsh)

**Other information**

davidbiancolin, 2020-04-26:
This smells like an infinite recursion, possibly because those env.sh files weren't properly updated.  Does this happen if you use a fresh clone of chipyard?

Can you provide `env.sh` and `env-riscv-tools.sh`?

baltazarortiz, 2020-04-27:
My `env.sh` and `env-riscv-tools.sh` are identical, which might in fact be my issue :)

```
export CHIPYARD_TOOLCHAIN_SOURCED=1                                             
export RISCV=/home/ubuntu//chipyard/riscv-tools-install            
export PATH=${RISCV}/bin:${PATH}                                                   
export LD_LIBRARY_PATH=${RISCV}/lib${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}         
source $( realpath $(dirname ${BASH_SOURCE[0]:-${(%):-%x}}) )/env-riscv-tools.sh
```

davidbiancolin, 2020-04-27:
they should look like:
`PATH=$( realpath $(dirname ${BASH_SOURCE[0]:-${(%):-%x}}) )/software/firemarshal:$PATH
source $( realpath $(dirname ${BASH_SOURCE[0]:-${(%):-%x}}) )/env-riscv-tools.sh`

and 
`export CHIPYARD_TOOLCHAIN_SOURCED=1
export RISCV=/home/ubuntu/chipyard/riscv-tools-install
export PATH=${RISCV}/bin:${PATH}
export LD_LIBRARY_PATH=${RISCV}/lib${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
`

For env.sh, and env-riscv-tools.sh respectively in 1.2.

davidbiancolin, 2020-04-27:
so there should be no recursive source :P

davidbiancolin, 2020-04-27:
I'm not totally sure why this happened yet, but in general i'd suggest doing a fresh clone initially when changing versions.

baltazarortiz, 2020-04-27:
Thanks! Do you have a recommendation for upgrading chipyard with local changes (i.e. changes to the TestHarness and RocketConfigs and things like that) in addition to a custom generator? I suppose I could have fresh cloned, manually copied all my changes in, and force pushed, but that didn't feel great.

baltazarortiz, 2020-04-27:
(I can open this discussion as a separate issue if you prefer)

baltazarortiz, 2020-04-27:
Oh also, I realized why they were the same - env.sh was a link to env-risc-tools :facepalm:

alonamid, 2020-10-19:
Should be fixed now that the build-toolchains script generates and env.sh file that calls other source files rather than appending to the end of the generated env.sh file

closed 2020-10-19 by alonamid