ucb-bar/chipyard #531: how to combine hwacha and gemmini in the same tile

sq4693, 2020-04-27:
HI,

my CONFIG is to put hwacha and gemmini in the same tile, but I can not generate the hwacha RTL, How to solve this, thanks.

class GemminiHwachaRocketConfig extends Config(
  new chipyard.iobinders.WithUARTAdapter ++
  new chipyard.iobinders.WithTieOffInterrupts ++
  new chipyard.iobinders.WithBlackBoxSimMem ++
  new chipyard.iobinders.WithTiedOffDebug ++
  new chipyard.iobinders.WithSimSerial ++
  new testchipip.WithTSI ++
  new chipyard.config.WithNoGPIO ++
  new chipyard.config.WithBootROM ++
  new chipyard.config.WithUART ++
  new chipyard.config.WithL2TLBs(1024) ++
  new gemmini.DefaultGemminiConfig ++ 
  new hwacha.DefaultHwachaConfig ++                        // use Hwacha vector accelerator
  new freechips.rocketchip.subsystem.WithNoMMIOPort ++
  new freechips.rocketchip.subsystem.WithNoSlavePort ++
  new freechips.rocketchip.subsystem.WithInclusiveCache ++
  new freechips.rocketchip.subsystem.WithNExtTopInterrupts(0) ++
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++
  new freechips.rocketchip.system.BaseConfig)

jerryz123, 2020-04-29:
Can you clarify what error you are seeing?

jerryz123, 2020-04-30:
The error was that the config fragments defined in Gemmini and Hwacha overrode the list of accelerators, instead of appending to it. The fixes are here:
https://github.com/ucb-bar/gemmini/pull/18/commits/ef0276086504c09f3cea2a30025f85ce999b3458
https://github.com/ucb-bar/hwacha/pull/19/commits/9cfcd4ddeda386536f100c22609b318079b64cec

You can apply these patches to the local Gemmini and Hwacha repos you are using. The fix will appear in Chipyard master in a future release.

sq4693, 2020-04-30:
Yes,i found its been overrided. and i have sovle it.
Thanks!

PR #580 (2020-05-29): Bump Submodules
**Related issue**: Fixes #538, Fixes #531 

**Type of change**: other enhancement

**Impact**: rtl change + software change

**Release Notes**
This is bumping all submodules in Chipyard to their `master` branches in preparation for the CY 1.3 release.
closed 2020-07-07 by alonamid