ucb-bar/chipyard #539: Multi-Clock Support for NVDLA

abejgonzalez, 2020-04-29:
**Impact**: rtl

**Description**
Have the NVDLA accelerator be in a different clock domain than the uncore/tile clocks.

**What is a motivating example for changing the behavior?**
Wanted in #505

jerryz123, 2020-08-26:
Since multiclock support is being worked on right now, I'm interested in what the use case for this is. What clocks does the NVDLA want?

adoerflinger, 2022-01-11:
I'm really happy about the multiclock support being now integrated in Chipyard and I can see how that works for tiles, buses, and some other IPs. However, I am still not sure how you can move accelerators such as NVDLA and Gemmini into a different clock domain, without changing the frequency of a whole bus. I thought one would need to modify `ClockSinkDomain` or `ClockSinkNode` and add an async crossing type. However, there is no clock sink defined for Gemmini and NVDLA. Has anyone done this yet?

Yuxin-Yu, 2023-11-09:
Hi @jerryz123 .Is there any new progress regarding NVDLA's support for multiple clocks?

jerryz123, 2023-11-10:
This can be done now. The nvdla wrapper should be modified such that it instantiates the NVDLA within its own ClockSinkDomain. The ClockSinkDomain can be attached to the subsystem AsyncClockGrouosNode. And clock crossings should be added to the NVDLA's tilelink interfaces.

Yuxin-Yu, 2023-11-10:
Do you have any relevant reference examples? I have just been in contact with Chipyard and I am not very familiar with diplomacy @jerryz123

Yuxin-Yu, 2023-12-29:
Hi @adoerflinger ，have you solved this?

Yuxin-Yu, 2024-01-02:
Hi @jerryz123 @abejgonzalez ,As jerryz123 said，I have modified the packaging code of NVDLA, where the NVDLA.scala file is as follows:
```
// See LICENSE for license details.
package nvidia.blocks.dla

import chisel3._
import freechips.rocketchip.config._
import freechips.rocketchip.diplomacy._
import freechips.rocketchip.amba.axi4._
import freechips.rocketchip.amba.apb._
import freechips.rocketchip.devices.tilelink.{BasicBusBlockerParams, TLClockBlocker}
import freechips.rocketchip.tilelink._
import freechips.rocketchip.interrupts._
import freechips.rocketchip.prci._
import freechips.rocketchip.subsystem._
import nvidia.blocks.ip.dla._
import sifive.blocks.util.{DeviceAttachParams, DeviceParams}
import freechips.rocketchip.regmapper._

case class NVDLAParams(
  config: String,
  raddress: BigInt=0x10040000L,
  synthRAMs: Boolean = false,
  dtsFrequency: BigInt = 15000000,
  ramSize:BigInt = 1
) extends DeviceParams

class NVDLA(params: NVDLAParams)(implicit p: Parameters) extends LazyModule {

//  def dtsFrequency = params.dtsFrequency
//  def fixedClockOpt = dtsFrequency.map(f => ClockParameters(freqMHz = f.toDouble / 1000000.0))

  val blackboxName = "nvdla_" + params.config
  val hasSecondAXI = params.config == "large"
  val dataWidthAXI = if (params.config == "large") 256 else 64

  // DTS
  val dtsdevice = new SimpleDevice("nvdla",Seq("nvidia,nv_" + params.config))

  // dbb TL
  val dbb_tl_node = TLIdentityNode()

  // dbb AXI
  val dbb_axi_node = AXI4MasterNode(
    Seq(
      AXI4MasterPortParameters(
        masters = Seq(AXI4MasterParameters(
          name    = "NVDLA DBB",
          id      = IdRange(0, 256))))))

  // TL <-> AXI
  (dbb_tl_node
    := TLBuffer()
    := TLWidthWidget(dataWidthAXI/8)
    := AXI4ToTL()
    := AXI4UserYanker(capMaxFlight=Some(16))
    := AXI4Fragmenter()
    := AXI4IdIndexer(idBits=3)
    := AXI4Buffer()
    := dbb_axi_node)

  // cvsram AXI
  val cvsram_axi_node = if (hasSecondAXI) Some(AXI4MasterNode(
    Seq(
      AXI4MasterPortParameters(
        masters = Seq(AXI4MasterParameters(
          name    = "NVDLA CVSRAM",
          id      = IdRange(0, 256)))))))
  else None

[... 168 lines cut ...]
      case _: AsynchronousCrossing => where.ibus.fromAsync
    }) := nvdla.int_node

    nvdla
  }
}
```
The Periphery.scala file is as follows:
```
import chisel3._
import freechips.rocketchip.config.Field
import freechips.rocketchip.subsystem.BaseSubsystem
import freechips.rocketchip.diplomacy.{BufferParams, LazyModule, LazyModuleImp}
import freechips.rocketchip.tilelink.{TLBuffer, TLIdentityNode}

case object NVDLAKey extends Field[Option[NVDLAParams]](None)
case object NVDLAFrontBusExtraBuffers extends Field[Int](0)

trait CanHavePeripheryNVDLA { this: BaseSubsystem =>
 val nvdla = p(NVDLAKey).map { case key =>
   NVDLAAttachParams(key).attachTo(this)
  }
}
```
It looks like there's no problem, and Verilog code can also be generated normally, but I don't know how to define the frequency of this clock group. May I ask how to define the frequency?