ucb-bar/chipyard #540: Documentation (PDF) sections do not match the online version

cousteaulecommandant, 2020-04-30:
The documentation in <https://chipyard.readthedocs.io/en/latest/> has this very nice layout with a "front matter" containing a quick start with the install steps, getting help, and a table of contents, and then all the sections (1. Chipyard basics; 2. Simulation; 3. Included RTL generators; etc.).

However, the automatically generated PDF documentation (<https://chipyard.readthedocs.io/_/downloads/en/latest/pdf/>) has a strange layout, where the whole front matter is section "1. Quick Start", with subsections "1.1. System requirements; 1.2. Setting up the Chipyard Repo; ...; 1.6. Table of Contents".  Then, all the content of the document appears as sub-sub-sections of the ToC (1.6.1. Chipyard basics; 1.6.2. Simulation; 1.6.3. Included RTL generators; etc.).  Basically all the content is in a single subsection, except for the introduction.  This makes navigating the PDF document a bit impractical and confusing.

Would it be possible to make "Quick Start" an unnumbered section / preface / front matter / "section 0" when generating the PDF document, and then everything in the ToC in their own sections, so that it doesn't get in the way of the actual document hierarchy and produces a proper PDF with useful bookmarks (with 3 levels of section numbering / PDF bookmarks)?  (I don't know if ReStructuredText/Sphinx supports this.)

alonamid, 2020-05-26:
We have actually removed the Quick Start section from the preface in the next Chipyard release (you can preview it https://chipyard.readthedocs.io/en/dev/ ) since we noticed the Quick Start section did not provide sufficient information about system requirements, and as a result users did not read the initial repository setup section. Therefore we decided to merge the "Quick Start" preface back into the "Initial Repository Setup" section, with a noticeable redirection to it from website landing page.

I haven't checked yet if this will help the PDF version, but we will happily accept a PR if you find a fix for this.

cousteaulecommandant, 2020-05-26:
It's done something on the PDF.  The PDF still has everything inside section 2, but at least it has removed one numbering level (now the superfluous prefix is 2. rather than 1.6.).  I don't think I'll be able to PR anything because I have zero experience with Sphinx.