ucb-bar/chipyard #555: Invalid build URI

ConstantPark, 2020-05-19:
This is my error symptom.
I don't know "Invalid build URI"

[info] Loading settings for project chipyard-build from plugins.sbt ...
[info] Loading project definition from /home/esoc/1.Workspace_1/constant/0.RISCV/mmio/temp/chipyard/project
[info] Updating ProjectRef(uri("file:/home/esoc/1.Workspace_1/constant/0.RISCV/mmio/temp/chipyard/project/"), "chipyard-build")...
[info] Done updating.
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] Loading settings for project dsptools from build.sbt ...
[info] Loading settings for project barstoolsMacros from build.sbt ...
[info] Loading settings for project mdf from build.sbt ...
[info] Loading settings for project boom from build.sbt ...
[info] Loading settings for project hwacha from build.sbt ...
[info] Loading settings for project icenet from build.sbt ...
[info] Loading settings for project testchipip from build.sbt ...
[info] Loading settings for project hardfloat from build.sbt ...
[info] Loading settings for project chisel_testers from build.sbt ...
[info] Loading settings for project treadle from build.sbt ...
[info] Loading settings for project firrtl_interpreter from build.sbt ...
[info] Loading settings for project chisel from build.sbt ...
[info] Loading settings for project chipyard from build.sbt ...
[error] java.lang.RuntimeException: Invalid build URI (no handler available): file:/home/esoc/1.Workspace_1/constant/0.RISCV/mmio/temp/chipyard/sims/firesim/sim/
[error]         at scala.sys.package$.error(package.scala:26)
[error]         at sbt.internal.Load$.$anonfun$builtinLoader$1(Load.scala:481)
[error]         at sbt.internal.BuildLoader.$anonfun$apply$3(BuildLoader.scala:240)
[error]         at scala.Option.getOrElse(Option.scala:121)
[error]         at sbt.internal.BuildLoader.apply(BuildLoader.scala:240)
[error]         at sbt.internal.Load$.loadURI$1(Load.scala:546)
[error]         at sbt.internal.Load$.loadAll(Load.scala:562)
[error]         at sbt.internal.Load$.loadURI(Load.scala:492)
[error]         at sbt.internal.Load$.load(Load.scala:471)
[error]         at sbt.internal.Load$.$anonfun$apply$1(Load.scala:251)
[error]         at sbt.internal.Load$.timed(Load.scala:1395)
[error]         at sbt.internal.Load$.apply(Load.scala:251)
[error]         at sbt.internal.Load$.defaultLoad(Load.scala:69)
[error]         at sbt.BuiltinCommands$.liftedTree1$1(Main.scala:829)
[error]         at sbt.BuiltinCommands$.doLoadProject(Main.scala:829)
[error]         at sbt.BuiltinCommands$.$anonfun$loadProjectImpl$2(Main.scala:800)
[error]         at sbt.Command$.$anonfun$applyEffect$4(Command.scala:142)
[error]         at sbt.Command$.$anonfun$applyEffect$2(Command.scala:137)
[error]         at sbt.Command$.process(Command.scala:181)
[error]         at sbt.MainLoop$.processCommand(MainLoop.scala:151)
[error]         at sbt.MainLoop$.$anonfun$next$2(MainLoop.scala:139)
[error]         at sbt.State$$anon$1.runCmd$1(State.scala:246)
[error]         at sbt.State$$anon$1.process(State.scala:250)
[error]         at sbt.MainLoop$.$anonfun$next$1(MainLoop.scala:139)
[error]         at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error]         at sbt.MainLoop$.next(MainLoop.scala:139)
[error]         at sbt.MainLoop$.run(MainLoop.scala:132)
[error]         at sbt.MainLoop$.$anonfun$runWithNewLog$1(MainLoop.scala:110)
[error]         at sbt.io.Using.apply(Using.scala:22)
[error]         at sbt.MainLoop$.runWithNewLog(MainLoop.scala:104)
[error]         at sbt.MainLoop$.runAndClearLast(MainLoop.scala:59)
[error]         at sbt.MainLoop$.runLoggedLoop(MainLoop.scala:44)
[error]         at sbt.MainLoop$.runLogged(MainLoop.scala:35)
[error]         at sbt.StandardMain$.runManaged(Main.scala:138)
[error]         at sbt.xMain.run(Main.scala:89)
[error]         at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:109)
[error]         at xsbt.boot.Launch$.withContextLoader(Launch.scala:128)
[error]         at xsbt.boot.Launch$.run(Launch.scala:109)
[error]         at xsbt.boot.Launch$$anonfun$apply$1.apply(Launch.scala:35)
[error]         at xsbt.boot.Launch$.launch(Launch.scala:117)
[error]         at xsbt.boot.Launch$.apply(Launch.scala:18)
[error]         at xsbt.boot.Boot$.runImpl(Boot.scala:41)
[error]         at xsbt.boot.Boot$.main(Boot.scala:17)
[error]         at xsbt.boot.Boot.main(Boot.scala)
[error] Invalid build URI (no handler available): file:/home/esoc/1.Workspace_1/constant/0.RISCV/mmio/temp/chipyard/sims/firesim/sim/
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? q

Is there anyone who can solve this problem ?

colinschmidt, 2020-05-19:
What version of chipyard are you using?
What step are you running when you encounter this issue?
Did you run the initial setup steps?

ConstantPark, 2020-05-21:
I fixed that issue. It was a version issue of Verilator. It would be nice to announce that there is no problem with which version works. More 4.x version makes a problem.

closed 2020-05-21 by colinschmidt

abdullahyildiz, 2020-06-23:
I got the same error on CentOS Linux release 7.7.1908 (Core).

```
$ verilator --version
Verilator 4.034 2020-05-03 rev v4.034
```

The error occurs when I run `make` in `sims/verilator`.

jerryz123, 2020-12-23:
Sorry for late response. 
This is likely due to bad git version causing early termination of `init-submodules-no-riscv-tool.sh`.
Please upgrade to git version > 1.8.5.

shi27feng, 2021-02-12:
I got the same error message even it's been upgraded to git version 1.8.5:

❯ cd sims/verilator
❯ make
Running with RISCV=/home/feng/Workspace/chipyard/riscv-tools-install
cd /home/feng/Workspace/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -jar /home/feng/Workspace/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/home/feng/Workspace/chipyard/tools  ";project utilities; runMain utilities.GenerateSimFiles -td /home/feng/Workspace/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -sim verilator"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.7 (Oracle Corporation Java 15.0.2)
[info] loading global plugins from /home/feng/.sbt/1.0/plugins
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/feng/Workspace/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/feng/Workspace/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] loading settings for project chisel3-build from plugins.sbt ...
[info] loading project definition from /home/feng/Workspace/chipyard/tools/chisel3/project
[info] loading settings for project chisel from build.sbt ...
[info] loading settings for project firrtl-build from plugins.sbt ...
[info] loading project definition from /home/feng/Workspace/chipyard/tools/firrtl/project
[info] loading settings for project firrtl from build.sbt ...
[info] resolving key references (55367 settings) ...
[info] set current project to chipyardRoot (in build file:/home/feng/Workspace/chipyard/)
[info] set current project to utilities (in build file:/home/feng/Workspace/chipyard/)
[info] compiling 27 Scala sources to /home/feng/Workspace/chipyard/generators/rocket-chip/hardfloat/target/scala-2.12/classes ...
[error] ## Exception when compiling 27 sources to /home/feng/Workspace/chipyard/generators/rocket-chip/hardfloat/target/scala-2.12/classes
[error] java.io.IOError: java.lang.RuntimeException: /packages cannot be represented as URI
[error] java.base/jdk.internal.jrtfs.JrtPath.toUri(JrtPath.java:176)
[error] scala.tools.nsc.classpath.JrtClassPath.asURLs(DirectoryClassPath.scala:213)
[error] scala.tools.nsc.classpath.AggregateClassPath.$anonfun$asURLs$1(AggregateClassPath.scala:64)
[error] scala.collection.TraversableLike.$anonfun$flatMap$1(TraversableLike.scala:244)
[error] scala.collection.Iterator.foreach(Iterator.scala:941)
[error] scala.collection.Iterator.foreach$(Iterator.scala:941)
[error] scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[error] scala.collection.IterableLike.foreach(IterableLike.scala:74)
[error] scala.collection.IterableLike.foreach$(IterableLike.scala:73)
[error] scala.collection.AbstractIterable.foreach(Iterable.scala:56)
[error] scala.collection.TraversableLike.flatMap(TraversableLike.scala:244)
[error] scala.collection.TraversableLike.flatMap$(TraversableLike.scala:241)
[error] scala.collection.AbstractTraversable.flatMap(Traversable.scala:108)
[error] scala.tools.nsc.classpath.AggregateClassPath.asURLs(AggregateClassPath.scala:64)
[error] scala.tools.nsc.typechecker.Macros.findMacroClassLoader(Macros.scala:78)
[error] scala.tools.nsc.typechecker.Macros.findMacroClassLoader$(Macros.scala:77)
[error] scala.tools.nsc.Global$$anon$4.findMacroClassLoader(Global.scala:477)
[error] scala.reflect.macros.runtime.MacroRuntimes.$anonfun$defaultMacroClassloaderCache$2(MacroRuntimes.scala:68)
[error] scala.reflect.internal.SymbolTable$perRunCaches$$anon$2.apply(SymbolTable.scala:477)
[error] scala.reflect.macros.runtime.MacroRuntimes.defaultMacroClassloader(MacroRuntimes.scala:70)
[error] scala.reflect.macros.runtime.MacroRuntimes.defaultMacroClassloader$(MacroRuntimes.scala:70)
[error] scala.tools.nsc.Global$$anon$4.defaultMacroClassloader(Global.scala:477)
[error] scala.reflect.macros.runtime.MacroRuntimes$MacroRuntimeResolver.resolveRuntime(MacroRuntimes.scala:88)
[error] scala.reflect.macros.runtime.MacroRuntimes.$anonfun$standardMacroRuntime$3(MacroRuntimes.scala:50)
[error] scala.collection.mutable.MapLike.getOrElseUpdate(MapLike.scala:209)
[error] scala.collection.mutable.MapLike.getOrElseUpdate$(MapLike.scala:206)
[... cut ...]
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
[error] 	at java.base/java.lang.Thread.run(Thread.java:832)
[error] (hardfloat / Compile / compileIncremental) java.io.IOError: java.lang.RuntimeException: /packages cannot be represented as URI
[error] Total time: 5 s, completed Feb 12, 2021, 10:23:55 AM
make: *** [/home/feng/Workspace/chipyard/common.mk:86: /home/feng/Workspace/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/sim_files.f] Error 1

tymcauley, 2021-02-12:
Not the exact same error, but I've been getting some strange "build URI" errors as well. For one, if I clone a fresh chipyard repo and run `init-submodules-no-riscv-tool.sh`, I can successfully build the default simulator, (`cd sims/verilator && make` works). However, if I `cd` to the repo's root directory and start up `sbt`, I get this error during startup:

```
$ sbt
[info] welcome to sbt 1.4.4 (Ubuntu Java 11.0.10)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/tynan/code/hw/chipyard/project
...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
java.lang.RuntimeException: Invalid build URI (no handler available): file:///home/tynan/workspace/chisel3/
... (stacktrace here)
[error] Invalid build URI (no handler available): file:///home/tynan/workspace/chisel3/
[error] Use 'last' for the full log.
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
```

So, `/home/tynan` is just the path to my home directory (but as you can see, I've checked out the repo to `/home/tynan/code/hw/chipyard`). Not sure where that path is getting picked up from. I'll try and do some more digging on this...

I'm currently on the `master` branch (release 1.4), running on an Ubuntu 20.04 machine. Let me know if you'd like me to try anything else.

tymcauley, 2021-02-12:
Looks like when located in the repo root, this works:

```
$ sbt -Dsbt.sourcemode=true -Dsbt.workspace=$(pwd)/tools
...
sbt:chipyardRoot>
```

but this does _not_ work:

```
$ sbt -Dsbt.sourcemode=true -Dsbt.workspace=./tools
...
java.lang.RuntimeException: Invalid build URI (no handler available): file:/home/tynan/code/hw/chipyard/sims/firesim/sim/tools/chisel3/
... (stacktrace)
[error] Invalid build URI (no handler available): file:/home/tynan/code/hw/chipyard/sims/firesim/sim/tools/chisel3/
[error] Use 'last' for the full log.
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
```

So, the absolute path is required.

tymcauley, 2021-02-12:
Also, adding a `.sbtopts` file fixes the issue:

```
$ cat .sbtopts
-Dsbt.sourcemode=true
-Dsbt.workspace=$PWD/tools
$ sbt
...
sbt:chipyardRoot>
```

tymcauley, 2021-03-05:
I also _just_ realized that there's a `launch-sbt` Make target, so you can run this command to start an SBT shell with the right options set:

```
make -C sims/verilator launch-sbt
```

davide-giacomini, 2021-04-02:
I got the same two errors of @tymcauley and @shi27feng. I did what @tymcauley wrote and now I can run `make launch-sbt` and everything goes smooth. I also added the `.sbtopts` file and I can run `sbt` from the root direcotry without problems.

I have an other problem though. If I run in any directory `make launch-sbt` there are no problems. However, if I load the sbt changes from IntelliJ Idea this error shows up:
`[error] Invalid build URI (no handler available): file:/home/davide/Projects/chipyard/$PWD/tools/chisel3/`

Moreover, if I try, inside the folder `fpga`, running `make SUB_PROJECT=vcu118 bitstream`, the exact same error typology of @shi27feng shows up. More precisely, the error is the subsequent: 
`make: *** [/home/davide/Projects/chipyard/common.mk:86: /home/davide/Projects/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/sim_files.f] Error 1`

I am simply trying to see how the bitstream is generated for the board above. I am new to Chipyard.

Thank you very much in advance.