ucb-bar/chipyard #561: Syntax Error running "make" in Verilator with default examples

skenkare6, 2020-05-21:
**Impact**: rtl | **software** | unknown | other

**What is the current behavior?**

I've tried running either the command `make` or ` make CONFIG=Sha3RocketConfig` in chipyard/sims/verilator after installing verilator through apt-get. For either, it runs until the part where it wants to run all the tests, where I get this message:

```
%Error: /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/SimUART.v:39: syntax error, unexpected ';'
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/bin/verilator_bin --cc --exe -O3 -CFLAGS ' -O1 -std=c++11 -I/root/chipyard/riscv-tools-install/include -I/root/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR' -CFLAGS '-I/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig -include /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig.plusArgs -include /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/verilator.h' -LDFLAGS ' -L/root/chipyard/riscv-tools-install/lib -Wl,-rpath,/root/chipyard/riscv-tools-install/lib -L/root/chipyard/tools/DRAMSim2 -Wl,-rpath,/root/chipyard/tools/DRAMSim2 -L/root/chipyard/sims/verilator -lfesvr -lpthread -ldramsim' --top-module TestHarness --assert --output-split 10000 --output-split-cfuncs 100 -f /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/sim_files.common.f /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig.top.v /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig.harness.v /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig.top.mems.v /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig.harness.mems.v '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' -o /root/chipyard/sims/verilator/simulator-chipyard-Sha3RocketConfig -Mdir /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig -CFLAGS '-include /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig/VTestHarness.h'
Makefile:108: recipe for target '/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig/VTestHarness.mk' failed
make: *** [/root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.Sha3RocketConfig/chipyard.TestHarness.Sha3RocketConfig/VTestHarness.mk] Error 10
```

If I run the default as opposed to Sha3, the error is on the same line and file.

**What is the expected behavior?**

The expected behavior is that the program is compiled and the executable is created.

**Please tell us about your environment:**

- version: `chipyard git commit 2c0928c (most recent on master)`
- os: `Linux Debian`

**Other information**

I'm following the instructions from section 2.1 of the docs.

jwright6323, 2020-05-21:
What version of verilator are you using? SimUART.v has never changed since its creation (https://github.com/ucb-bar/testchipip/commits/4b15061b6fe77f6793603f799751f1f988144ef7/src/main/resources/testchipip/vsrc/SimUART.v), so it would surprise me if this is a problem with the verilog and not a verilator problem.

skenkare6, 2020-05-21:
@jwright6323 Oh, interesting, thanks for finding that!! It's Verilator 3.916 2017-11-25 rev verilator_3_914-65-g0478dbd, so looks like it's pretty old. Let me try installing a more up to date version and I'll update this soon with if the issue is fixed. Thank you!!

jwright6323, 2020-05-21:
Cool, let us know if that works. For reference, you can check the version of verilator we use for CI here:
https://github.com/ucb-bar/chipyard/blob/2c0928cb9c7c7f56f2d362c0a57d05561a872fa9/.circleci/defaults.sh#L24

Note that the `dev` branch is our development branch, and we've bumped it to `v4.034` there to get around some other issues, which will be reflected in the next release.

saarilakenkare, 2020-05-23:
Yup, installing that version worked perfectly. Thank you!!!

skenkare6, 2020-05-23:
Yup, works perfectly. Thank you!!!

closed 2020-05-23 by skenkare6