ucb-bar/chipyard #564: Makefile comment line-lengths (style discussion)

ssteffl, 2020-05-24:
In [a recent PR](https://github.com/ucb-bar/chipyard/pull/562), I rewrote the verilator Makefile, and in the course of doing so I shorted all line-lengths to < 80 characters. including comment lines.

It was suggested I open up a style ticket to discuss why we are using 89-character comment lines in the first place, and probe for any objections to reducing these to < 80 characters (either 72 or 78 characters).

Many editors allow vertical window splitting to view code side by side (including vim, which I use). I use this feature 100% of the time i have vim open. But long lines in files make viewing these files annoying. 

I believe it is unlikely that shortening the comment lines will have any adverse effects on those who do not vertically split their editor window, while only benefiting those who do.

jwright6323, 2020-05-24:
> why we are using 89-character comment lines in the first place

I'm 99% sure 89 is arbitrary- somebody held the `#` key down for a few seconds and every subsequent change just copied that line.

> I believe it is unlikely that shortening the comment lines will have any adverse effects on those who do not vertically split their editor window, while only benefiting those who do.

I, and I'm reasonably confident almost everyone else, use vertical splits and have never been bothered by this, but I have no opposition to making the change as long as it's in its own style PR and globally changes _all_ files in the repo.

colinschmidt, 2020-05-24:
I also use vertical splits constantly and it is a very slight annoyance that some lines are over 80 characters, but I encounter lots of files like that so I'm used to it.

That being said I think a single PR that changes all comment lines to 72 characters would be my preferred solution now that we are changing it.

a0u, 2020-05-24:
72 characters is also my preference.

abejgonzalez, 2020-05-25:
Yes this was arbitrary and doesn't affect me so I don't care about lengths.

davidbiancolin, 2020-05-25:
I'm fine with this for makefiles. _Hard_ veto on applying sub 80 lines to scala files.