ucb-bar/chipyard #590: Using AXI4 in GCD.scala MMIO peripheral

jalezeta, 2020-06-01:
Hello!

I am pretty new to Scala and I have been trying to attach my own simple IP to the Rocket chip. I have found that Chipyard has an example of this on which a GCD IP is attached to the TL bus. This "GCD.scala" example has an option to attach the IP, but I have seen that there is a comment stating that the Chisel code should only be built if we are using the non-AXI4 version. What is the reason for this? I have tried to integrate this example into the freedom U-500 implemented in a VC707 FPGA and it only works when the argument "useAXI4" is false. 

Thanks for your time.

jerryz123, 2020-06-01:
You should be able to attach a Chisel GCD example via AXI4, those two parameters (Chisel/Verilog and AXI4/TL) are designed to be orthogonal.

What configuration are you attempting to build? What is the error you are seeing.

jalezeta, 2020-06-01:
Thanks  @jerryz123 !

I am trying to build the "AXI4=true" and "useBlackBox=true" configuration. The problem that I am facing is that I can see my GCD block in the address map output log display during build just when I am using "AXI4=false". I was worried about the comment in "GCD.scala" in line 163 that says " // Only build if we are using the TL (nonAXI4) version". Should it work out of the box? 

[info] [0.001] Elaborating design...
Interrupt map (8 harts 6 interrupts):
  [1, 1] => uart_0
  [2, 2] => spi_0
  [3, 6] => gpio_0

mem AXI4-ID <= TL-Source mapping:
[ 0, 32) <= [ 0, 64) "TLBroadcast"

/dts-v1/;

/ {
#address-cells = <1>;
#size-cells = <1>;
compatible = "freechips,rocketchip-unknown-dev";
model = "freechips,rocketchip-unknown";
L28: aliases {
serial0 = &L19;
};
L23: chosen {
};
L27: cpus {
#address-cells = <1>;
#size-cells = <0>;
L6: cpu@0 {
clock-frequency = <0>;
compatible = "sifive,rocket0", "riscv";
d-cache-block-size = <64>;
d-cache-sets = <64>;
d-cache-size = <16384>;
d-tlb-sets = <1>;
d-tlb-size = <32>;
device_type = "cpu";
i-cache-block-size = <64>;
i-cache-sets = <64>;
i-cache-size = <16384>;
i-tlb-sets = <1>;
i-tlb-size = <32>;
mmu-type = "riscv,sv39";
next-level-cache = <&L0 &L22>;
reg = <0x0>;
riscv,isa = "rv64imafdc";
status = "okay";
timebase-frequency = <1000000>;
tlb-split;
L4: interrupt-controller {
#interrupt-cells = <1>;
compatible = "riscv,cpu-intc";
interrupt-controller;
};
};
L9: cpu@1 {
clock-frequency = <0>;
compatible = "sifive,rocket0", "riscv";
d-cache-block-size = <64>;
d-cache-sets = <64>;
d-cache-size = <16384>;
d-tlb-sets = <1>;
d-tlb-size = <32>;
device_type = "cpu";
i-cache-block-size = <64>;
i-cache-sets = <64>;
i-cache-size = <16384>;
i-tlb-sets = <1>;
i-tlb-size = <32>;
mmu-type = "riscv,sv39";
next-level-cache = <&L0 &L22>;
reg = <0x1>;
riscv,isa = "rv64imafdc";
status = "okay";
timebase-frequency = <1000000>;
tlb-split;
L7: interrupt-controller {
#interrupt-cells = <1>;
compatible = "riscv,cpu-intc";
interrupt-controller;
};
};
L12: cpu@2 {
clock-frequency = <0>;
compatible = "sifive,rocket0", "riscv";
d-cache-block-size = <64>;
d-cache-sets = <64>;
d-cache-size = <16384>;
d-tlb-sets = <1>;
d-tlb-size = <32>;
device_type = "cpu";
i-cache-block-size = <64>;
i-cache-sets = <64>;
i-cache-size = <16384>;
i-tlb-sets = <1>;
i-tlb-size = <32>;
mmu-type = "riscv,sv39";
next-level-cache = <&L0 &L22>;
reg = <0x2>;
riscv,isa = "rv64imafdc";
status = "okay";
timebase-frequency = <1000000>;
tlb-split;
L10: interrupt-controller {
#interrupt-cells = <1>;
compatible = "riscv,cpu-intc";
interrupt-controller;
};
};
L15: cpu@3 {
clock-frequency = <0>;
compatible = "sifive,rocket0", "riscv";
d-cache-block-size = <64>;
d-cache-sets = <64>;
d-cache-size = <16384>;
d-tlb-sets = <1>;
d-tlb-size = <32>;
device_type = "cpu";
i-cache-block-size = <64>;
i-cache-sets = <64>;
i-cache-size = <16384>;
i-tlb-sets = <1>;
i-tlb-size = <32>;
mmu-type = "riscv,sv39";
next-level-cache = <&L0 &L22>;
reg = <0x3>;
riscv,isa = "rv64imafdc";
status = "okay";
timebase-frequency = <1000000>;
tlb-split;
L13: interrupt-controller {
#interrupt-cells = <1>;
compatible = "riscv,cpu-intc";
interrupt-controller;
};
};
};
L22: memory@80000000 {
device_type = "memory";
reg = <0x80000000 0x40000000>;
};
L26: soc {
#address-cells = <1>;
#size-cells = <1>;
compatible = "freechips,rocketchip-unknown-soc", "simple-bus";
ranges;
L2: clint@2000000 {
compatible = "riscv,clint0";
interrupts-extended = <&L4 3 &L4 7 &L7 3 &L7 7 &L10 3 &L10 7 &L13 3 &L13 7>;
reg = <0x2000000 0x10000>;
reg-names = "control";
};
L3: debug-controller@0 {
compatible = "sifive,debug-013", "riscv,debug-013";
interrupts-extended = <&L4 65535 &L7 65535 &L10 65535 &L13 65535>;
reg = <0x0 0x1000>;
reg-names = "control";
};
L0: error-device@3000 {
compatible = "sifive,error0";
reg = <0x3000 0x1000>;
};
L17: gcd@2000 {
compatible = "ucbbar,gcd";
reg = <0x2000 0x1000>;
reg-names = "control";
};
L24: gpio@64002000 {
#gpio-cells = <2>;
#interrupt-cells = <2>;
compatible = "sifive,gpio0";
gpio-controller;
interrupt-controller;
interrupt-parent = <&L1>;
interrupts = <3 4 5 6>;
reg = <0x64002000 0x1000>;
reg-names = "control";
};
L1: interrupt-controller@c000000 {
#interrupt-cells = <1>;
compatible = "riscv,plic0";
interrupt-controller;
interrupts-extended = <&L4 11 &L4 9 &L7 11 &L7 9 &L10 11 &L10 9 &L13 11 &L13 9>;
reg = <0xc000000 0x4000000>;
reg-names = "control";
riscv,max-priority = <7>;
riscv,ndev = <6>;
};
L16: rom@10000 {
compatible = "sifive,maskrom0";
reg = <0x10000 0x2000>;
reg-names = "mem";
};
L19: serial@64000000 {
clocks = <&L18>;
compatible = "sifive,uart0";
interrupt-parent = <&L1>;
interrupts = <1>;
reg = <0x64000000 0x1000>;
reg-names = "control";
};
L20: spi@64001000 {
#address-cells = <1>;
#size-cells = <0>;
clocks = <&L18>;
compatible = "sifive,spi0";
interrupt-parent = <&L1>;
interrupts = <2>;
reg = <0x64001000 0x1000>;
reg-names = "control";
L21: mmc@0 {
compatible = "mmc-spi-slot";
disable-wp;
reg = <0x0>;
spi-max-frequency = <20000000>;
voltage-ranges = <3300 3300>;
};
};
L18: tlclk {
#clock-cells = <0>;
clock-frequency = <50000000>;
clock-output-names = "tlclk";
compatible = "fixed-clock";
};
};
};

Generated Address Map
      0 -     1000 ARWX  debug-controller@0
   2000 -     3000 ARW   gcd@2000
   3000 -     4000 ARWXC error-device@3000
  10000 -    12000  R X  rom@10000
2000000 -  2010000 ARW   clint@2000000
c000000 - 10000000 ARW   interrupt-controller@c000000
64000000 - 64001000 ARW   serial@64000000
64001000 - 64002000 ARW   spi@64001000
64002000 - 64003000 ARW   gpio@64002000
80000000 - c0000000  RWXC memory@80000000

jerryz123, 2020-06-01:
It looks like AXI4RegisterRouter class used by the GCD example does not add an entry to the DTS, or to the generated address map.

However, the generated hardware appears to have access to the GCD device.

jalezeta, 2020-06-02:
Thanks @jerryz123!

I finally checked the generated verilog file and I could see that the HW is generated correctly as you said. There should not be any problem in this aspect then.

However, I tried adapting this GCD into my own example and I could see that a "MyAdderAXI4" module is generated, but the Adder internal logic is not showing up. 

This is the code that I have slightly modified from "GCD.scala". I am using the non Blackbox version and AXI4 version. 

`package myadder

import chisel3._
import chisel3.util._
import chisel3.experimental.{IntParam, BaseModule}
import freechips.rocketchip.amba.axi4._
import freechips.rocketchip.subsystem.BaseSubsystem
import freechips.rocketchip.config.{Parameters, Field, Config}
import freechips.rocketchip.diplomacy._
import freechips.rocketchip.regmapper.{HasRegMap, RegField}
import freechips.rocketchip.tilelink._
import freechips.rocketchip.util.UIntIsOneOf

// DOC include start: MyAdder params
case class MyAdderParams(
  address: BigInt = 0x2000,
  width: Int = 32,
  useAXI4: Boolean = true,
  useBlackBox: Boolean = true)
// DOC include end: MyAdder params

// DOC include start: MyAdder key
case object MyAdderKey extends Field[Option[MyAdderParams]](None)
// DOC include end: MyAdder key

class MyAdderIO(val w: Int) extends Bundle {

  val in1 = Input(UInt(w.W))
  val in2 = Input(UInt(w.W))
  val sum = Output(UInt(w.W))
  val carry_out = Output(Bool())
  //val busy = Output(Bool())

}

trait MyAdderTopIO extends Bundle {
 // val MyAdder_busy = Output(Bool())
}

trait HasMyAdderIO extends BaseModule {
  val w: Int
  val io = IO(new MyAdderIO(w))
}

// DOC include start: MyAdder blackbox
class MyAdderBlackBox(val w: Int) extends BlackBox(Map("WIDTH" -> IntParam(w))) with HasBlackBoxResource
  with HasMyAdderIO
{
  setResource("/src/vsrc/myAdderBlackBox.v")
}
// DOC include end: MyAdder blackbox

// DOC include start: MyAdder chisel
class MyAdderChiselModule(val w: Int) extends Module
  with HasMyAdderIO
{
  
  val in1   = Reg(UInt(w.W))
  val in2   = Reg(UInt(w.W))
  val sum   = Reg(UInt(w.W))
  val carry_out = Reg(UInt(1.W))

  in1 := io.in1
  in2 := io.in2
  sum := in1+in2 	
  io.sum := sum
  io.carry_out := Bool(true)
  //io.busy := Bool(true)
}
// DOC include end: MyAdder chisel

// DOC include start: MyAdder instance regmap

trait MyAdderModule extends HasRegMap {
  val io: MyAdderTopIO
  implicit val p: Parameters
  def params: MyAdderParams

  val in1 = Reg(UInt(params.width.W))
  val in2 = Reg(UInt(params.width.W))
  val sum = Reg(UInt(params.width.W))
  val carry_out = Reg(UInt(1.W))

  val impl = if (params.useBlackBox) {
    Module(new MyAdderBlackBox(params.width))
  } else {
    Module(new MyAdderChiselModule(params.width))
  }
 
  impl.io.in1 := in1
  impl.io.in2 := in2
  sum := impl.io.sum
  carry_out := impl.io.carry_out
  //io.MyAdder_busy := impl.io.busy

  regmap(
    0x00 -> Seq(
      RegField.w(params.width, in1)), // a read-only register capturing current status
    0x04 -> Seq(
      RegField.w(params.width, in2)), // a plain, write-only register
    0x08 -> Seq(
      RegField.w(params.width, sum)), // write-only, y.valid is set on write
    0x0C -> Seq(
      RegField.r(1, carry_out))) // read-only, MyAdder.ready is set on read
}
// DOC include end: MyAdder instance regmap

// DOC include start: MyAdder router
class MyAdderTL(params: MyAdderParams, beatBytes: Int)(implicit p: Parameters)
  extends TLRegisterRouter(
    params.address, "MyAdder", Seq("ucbbar,MyAdder"),
    beatBytes = beatBytes)(
      new TLRegBundle(params, _) with MyAdderTopIO)(
      new TLRegModule(params, _, _) with MyAdderModule)

class MyAdderAXI4(params: MyAdderParams, beatBytes: Int)(implicit p: Parameters)
  extends AXI4RegisterRouter(
    params.address,
    beatBytes=beatBytes)(
      new AXI4RegBundle(params, _) with MyAdderTopIO)(
      new AXI4RegModule(params, _, _) with MyAdderModule)
// DOC include end: MyAdder router

// DOC include start: MyAdder lazy trait
trait CanHavePeripheryMyAdder { this: BaseSubsystem =>
  private val portName = "MyAdder"

  // Only build if we are using the TL (nonAXI4) version
  val MyAdder = p(MyAdderKey) match {
    case Some(params) => {
      if (params.useAXI4) {
        val MyAdder = LazyModule(new MyAdderAXI4(params, pbus.beatBytes)(p))
        pbus.toSlave(Some(portName)) {
          MyAdder.node :=
          AXI4Buffer () :=
          TLToAXI4 () :=
          // toVariableWidthSlave doesn't use holdFirstDeny, which TLToAXI4() needsx
          TLFragmenter(pbus.beatBytes, pbus.blockBytes, holdFirstDeny = true)
        }
        Some(MyAdder)
      } else {
        val MyAdder = LazyModule(new MyAdderTL(params, pbus.beatBytes)(p))
        pbus.toVariableWidthSlave(Some(portName)) { MyAdder.node }
        Some(MyAdder)
      }
    }
    case None => None
  }
}
// DOC include end: MyAdder lazy trait

// DOC include start: MyAdder imp trait
trait CanHavePeripheryMyAdderModuleImp extends LazyModuleImp {
  val outer: CanHavePeripheryMyAdder
  /*val MyAdder_busy = outer.MyAdder match {
    case Some(MyAdder) => {
      val busy = IO(Output(Bool()))
      busy := MyAdder.module.io.MyAdder_busy
      Some(busy)
    }
    case None => None
  }*/
}

// DOC include end: MyAdder imp trait

// DOC include start: MyAdder config fragment
class WithMyAdder(useAXI4: Boolean, useBlackBox: Boolean) extends Config((site, here, up) => {
  case MyAdderKey => Some(MyAdderParams(useAXI4 = useAXI4, useBlackBox = useBlackBox))
})
// DOC include end: MyAdder config fragment`

jalezeta, 2020-06-04:
The MyAdderAXI4 Module is meant to be AXI4 Slave that should then be converted to TL and connected to the main bus of the SoC.

jerryz123, 2020-06-04:
Sorry, what I meant to say is, what does the generated MyAdderAXI4 module contain? I imagine it is supposed to wrap the internal Adder module?

jalezeta, 2020-06-04:
Yes, that is the idea. The problem is that I do not see any "Adder" logic inside the MyAdderAXI4 verilog module generated. (I am attaching this to the Freedom U500 implementation in the VC707 FPGA).

`module MyAdderAXI4( 
  input         clock, 
  input         reset, 
  output        auto_in_aw_ready, 
  input         auto_in_aw_valid, 
  input         auto_in_aw_bits_id, 
  input  [13:0] auto_in_aw_bits_addr, 
  input  [11:0] auto_in_aw_bits_user, 
  output        auto_in_w_ready, 
  input         auto_in_w_valid, 
  input         auto_in_b_ready, 
  output        auto_in_b_valid, 
  output [11:0] auto_in_b_bits_user, 
  output        auto_in_ar_ready, 
  input         auto_in_ar_valid, 
  input         auto_in_ar_bits_id, 
  input  [13:0] auto_in_ar_bits_addr, 
  input  [11:0] auto_in_ar_bits_user, 
  input         auto_in_r_ready, 
  output        auto_in_r_valid, 
  output [63:0] auto_in_r_bits_data, 
  output [11:0] auto_in_r_bits_user 
);
  wire  Queue_clock; 
  wire  Queue_reset; 
  wire  Queue_io_enq_ready; 
  wire  Queue_io_enq_valid; 
  wire  Queue_io_enq_bits_read; 
  wire [63:0] Queue_io_enq_bits_data; 
  wire [12:0] Queue_io_enq_bits_extra; 
  wire  Queue_io_deq_ready; 
  wire  Queue_io_deq_valid; 
  wire  Queue_io_deq_bits_read; 
  wire [63:0] Queue_io_deq_bits_data; 
  wire [12:0] Queue_io_deq_bits_extra; 
  wire  _T_148; 
  wire  _T_150; 
  wire  _T_230_ready; 
  wire [13:0] _T_159; 
  wire [10:0] _T_224; 
  wire [8:0] _T_145_bits_index; 
  wire  _T_462; 
  wire [8:0] _T_242; 
  wire  _T_244; 
  wire  _T_151; 
  wire [12:0] _T_156; 
  wire [12:0] _T_157; 
  wire  _GEN_12; 
  wire [63:0] _GEN_14; 
  wire  _T_612_bits_read; 
  wire  _T_612_valid; 
  wire  _T_618; 
  wire [12:0] _T_612_bits_extra; 
  Queue_45 Queue ( 
    .clock(Queue_clock),
    .reset(Queue_reset),
    .io_enq_ready(Queue_io_enq_ready),
    .io_enq_valid(Queue_io_enq_valid),
    .io_enq_bits_read(Queue_io_enq_bits_read),
    .io_enq_bits_data(Queue_io_enq_bits_data),
    .io_enq_bits_extra(Queue_io_enq_bits_extra),
    .io_deq_ready(Queue_io_deq_ready),
    .io_deq_valid(Queue_io_deq_valid),
    .io_deq_bits_read(Queue_io_deq_bits_read),
    .io_deq_bits_data(Queue_io_deq_bits_data),
    .io_deq_bits_extra(Queue_io_deq_bits_extra)
  );
  assign _T_148 = auto_in_aw_valid & auto_in_w_valid; 
  assign _T_150 = auto_in_ar_valid == 1'h0; 
  assign _T_230_ready = Queue_io_enq_ready; 
  assign _T_159 = auto_in_ar_valid ? auto_in_ar_bits_addr : auto_in_aw_bits_addr; 
  assign _T_224 = _T_159[13:3]; 
  assign _T_145_bits_index = _T_224[8:0]; 
  assign _T_462 = _T_145_bits_index[0]; 
  assign _T_242 = _T_145_bits_index & 9'h1fe; 
  assign _T_244 = _T_242 == 9'h0; 
  assign _T_151 = _T_230_ready & _T_150; 
  assign _T_156 = {auto_in_ar_bits_id,auto_in_ar_bits_user}; 
  assign _T_157 = {auto_in_aw_bits_id,auto_in_aw_bits_user}; 
  assign _GEN_12 = _T_462 ? _T_244 : _T_244; 
  assign _GEN_14 = _T_462 ? 64'h100000000 : 64'h0; 
  assign _T_612_bits_read = Queue_io_deq_bits_read; 
  assign _T_612_valid = Queue_io_deq_valid; 
  assign _T_618 = _T_612_bits_read == 1'h0; 
  assign _T_612_bits_extra = Queue_io_deq_bits_extra; 
  assign auto_in_aw_ready = _T_151 & auto_in_w_valid; 
  assign auto_in_w_ready = _T_151 & auto_in_aw_valid; 
  assign auto_in_b_valid = _T_612_valid & _T_618; 
  assign auto_in_b_bits_user = _T_612_bits_extra[11:0]; 
  assign auto_in_ar_ready = Queue_io_enq_ready; 
  assign auto_in_r_valid = _T_612_valid & _T_612_bits_read; 
  assign auto_in_r_bits_data = Queue_io_deq_bits_data; 
  assign auto_in_r_bits_user = _T_612_bits_extra[11:0]; 
  assign Queue_clock = clock; 
  assign Queue_reset = reset; 
  assign Queue_io_enq_valid = auto_in_ar_valid | _T_148; 
  assign Queue_io_enq_bits_read = auto_in_ar_valid; 
  assign Queue_io_enq_bits_data = _GEN_12 ? _GEN_14 : 64'h0; 
  assign Queue_io_enq_bits_extra = auto_in_ar_valid ? _T_156 : _T_157; 
  assign Queue_io_deq_ready = _T_612_bits_read ? auto_in_r_ready : auto_in_b_ready;  
ndmodule`

jerryz123, 2020-06-04:
It looks like you made the `sum` MMIO register a write-only register, not a read-only register, which I expect is what you intended. In the current design, it looks like the output IO of `impl` is unused, which could cause that module to be optimized out.

jalezeta, 2020-06-04:
Yes! Thanks @jerryz123, I was intending to make it a read-only register, my mistake. What do you mean by `impl` being unused? Do you mean that the `trait CanHavePeripheryMyAdderModuleImp` is not instantiating the `impl` IO output `sum`? For instance:

```
trait CanHavePeripheryMyAdderModuleImp extends LazyModuleImp {
  val outer: CanHavePeripheryMyAdder
  def params: MyAdderParams
  val MyAdder_sum = outer.MyAdder match {
    case Some(MyAdder) => {
      val sum = IO(Output(UInt(params.width.W))
      sum:= MyAdder.module.io.MyAdder_sum
      Some(sum)
    }
    case None => None
  }
}
```
And modify the TopIO module:

```
trait MyAdderTopIO extends Bundle {
 def params: MyAdderParams
 val MyAdder_sum = Output(UInt(params.width.W))
}
```

I guess I should do the same for the `carry_out`, but for now I am not interested.

jerryz123, 2020-06-04:
I mean if a module's outputs are all constants, or unused, FIRRTL may eliminate the module. For your example, this module:
```
 val impl = if (params.useBlackBox) {
    Module(new MyAdderBlackBox(params.width))
  } else {
    Module(new MyAdderChiselModule(params.width))
  }
```
may be eliminated.

jalezeta, 2020-06-04:
Understood, but I don't understand why my specific example has constant or unsused examples. Do you mean that these outputs are not included in the `CanHavePeripheryMyAdderModuleImp ` and `MyAdderTopIO ` . My intention is to write to numbers into in1 and in2 registers and then read the sum register. My intention is to have the sum as the output, but not as constant or unused. Sorry for being new into Chisel :)

jerryz123, 2020-06-04:
When sum is a write-only register, the read output of `sum` from module impl is unused. The only output IO of impl which is used is `carry_out`, but `carry_out` is set to a constant.

jalezeta, 2020-06-04:
Thanks! I already fixed it and I am able to see the internal logic. I really appreciate your help  @jerryz123. I'll close the issue as it is solved now.

closed 2020-06-04 by jalezeta