ucb-bar/chipyard #591: loadMemoryFromFile broken by Chipyard transforms

baltazarortiz, 2020-06-03:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* 1.3.0
*OS:* Ubuntu 18.04
*Other:* 

* My branch demonstrating this bug by adding a `TLRAM` to the GCD example is [here](https://github.com/baltazarortiz/chipyard/tree/TLRAM_loading_bug).

* [Link I referenced that does the same thing in Rocket](https://jia.je/hardware/2020/03/17/rocket-chip-tlram-load/)

**What is the current behavior?**
* I instantiated a `TLRAM` inside my module (GCD in the example branch) and then pointed `loadMemoryFromFile` at it.

* The `TLRAM` shows up in the memory map as expected, and the annotations JSON file shows a `LoadMemoryAnnotation`. However, `LoadMemoryTransform` is not run.

* I tried adding `LoadMemoryTransform` to `TopTransforms` and/or `HarnessTransforms`, and it made no difference.

**What is the expected behavior?**
* `loadMemoryFromFile` should load a custom `TLRAM` despite the other Chipyard transforms.

**Other information**

Based on discussion in the FIRRTL chat and Chipyard mailing list, it seems that the issue is some sort of interaction between `ReplSeqMems` or the memory macrocompiler things, neither of which I am super well versed on.

However, when I removed all references to those from the makefiles, my TLRAM did get the transform applied. It was still split into 8 pieces though, which made it slightly nonobvious how I would have had to fill out the 8 files to get my memory loaded contiguously (is it just file 0 through file 7 going low to high address?). I tried using the `SimplifyMems` transform, but it didn't seem to do anything.

This also broke the actual Chipyard memory system though, so is not an actual solution, just showed that this is most likely at least part of the issue.

baltazarortiz, 2020-06-03:
(If this should be an issue against barstools instead, let me know!)

HaFred, 2020-06-27:
Is it carried out on verilator backend? How is it going now?

baltazarortiz, 2020-06-28:
Not sure what you mean re: verilator, but I still don't have this working. With thesis deadline slowly but steadily approaching I am just working around this for now with a blackbox peripheral with a giant memory in it that I can manually load. Works for my needs thankfully, but very much not integrated with the Chipyard memory system.

abejgonzalez, 2020-08-17:
@baltazarortiz Is this issue solved with this PR? https://github.com/ucb-bar/chipyard/pull/635

baltazarortiz, 2020-08-17:
@abejgonzalez yes in the sense that I can now load memory from a file, but not in the sense that #635 actually fixes this issue. The fundamental issue here that I'd be worried about in general is that the Chipyard transforms seem to somehow break other FIRRTL transforms. This isn't a part of Chipyard that I'm knowledgeable enough with to know whether it's just memory-related transforms or others as well.

abejgonzalez, 2020-08-18:
Noted. I'll keep this open then so we can look into it when we get the chance.