ucb-bar/chipyard #604: Can't build esp-tools

shirlynan, 2020-06-18:
I got errors when running ./scripts/build-toolchains.sh riscv-tools esp-tools:

/home/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++
/home/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s

How could I solve it?
Thanks for your help.

alonamid, 2020-06-19:
I believe the build-toolchains.sh script takes only one argument, not two.
You need to run the script once for each of the toolchains.

shirlynan, 2020-06-19:
> I believe the build-toolchains.sh script takes only one argument, not two.
> You need to run the script once for each of the toolchains.

Thanks for your reply. However,  even if I only use one argument, I got same errors. I can't compile riscv-tools either.

chipyardtest/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++
chipyardtest/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s

closed 2020-06-19 by shirlynan

yzt000000, 2020-12-16:
> Thanks for your reply. However, even if I only use one argument, I got same errors. I can't compile riscv-tools either. [...]

how to recover it， I had met the same problem

shirlynan, 2020-12-16:
> how to recover it， I had met the same problem

Sorry, a long time has passed, I've forgotten how I solved it. But I vaguely remember that, I seemed to have reinstalled the entire toolchain or upgraded the gcc version. It would be great if you could try again and tell me if it works.

mbs0221, 2020-12-30:
I had met similar problems when running the build-toolchains.sh script for riscv-tools as following:
`$ ./scripts/build-toolchains.sh riscv-tools`