ucb-bar/chipyard #609: Default make failed with verilator

ygkim912, 2020-06-19:
I encountered the following error when running `make` in `sims/verilator`
```
[error] java.lang.RuntimeException: Missing scala-library.jar
[error]         at scala.sys.package$.error(package.scala:30)
[error]         at sbt.Defaults$.$anonfun$scalaInstanceFromUpdate$8(Defaults.scala:800)
[error]         at scala.Option.getOrElse(Option.scala:189)
[error]         at sbt.Defaults$.file$1(Defaults.scala:800)
[error]         at sbt.Defaults$.$anonfun$scalaInstanceFromUpdate$1(Defaults.scala:802)
[error]         at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error]         at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error]         at sbt.std.Transform$$anon$4.work(Transform.scala:67)
[error]         at sbt.Execute.$anonfun$submit$2(Execute.scala:280)
[error]         at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:19)
[error]         at sbt.Execute.work(Execute.scala:289)
[error]         at sbt.Execute.$anonfun$submit$1(Execute.scala:280)
[error]         at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:178)
[error]         at sbt.CompletionService$$anon$2.call(CompletionService.scala:37)
[error] java.lang.RuntimeException: Missing scala-library.jar
[error]         at scala.sys.package$.error(package.scala:30)
[error]         at sbt.Defaults$.$anonfun$scalaInstanceFromUpdate$8(Defaults.scala:800)
[error]         at scala.Option.getOrElse(Option.scala:189)
[error]         at sbt.Defaults$.file$1(Defaults.scala:800)
[error]         at sbt.Defaults$.$anonfun$scalaInstanceFromUpdate$1(Defaults.scala:802)
[error]         at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error]         at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error]         at sbt.std.Transform$$anon$4.work(Transform.scala:67)
[error]         at sbt.Execute.$anonfun$submit$2(Execute.scala:280)
[error]         at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:19)
[error]         at sbt.Execute.work(Execute.scala:289)
[error]         at sbt.Execute.$anonfun$submit$1(Execute.scala:280)
[error]         at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:178)
[error]         at sbt.CompletionService$$anon$2.call(CompletionService.scala:37)
[error] (firrtl_interpreter / scalaInstance) Missing scala-library.jar
[error] (chisel_testers / scalaInstance) Missing scala-library.jar
```

jerryz123, 2020-06-19:
Can you delete your `~/.sbt` and `~/.ivy2` directories and retry?

ygkim912, 2020-06-19:
It also failed.
IntelliJ also cannot import sbt project.

abejgonzalez, 2020-08-17:
Is this issue solved?

ygkim912, 2020-10-07:
No, not solved yet.

alonamid, 2020-10-19:
Is this error appearing only within IntelliJ, or also when you clone and work completely within the command line?

knipegp, 2020-10-22:
I can now build a Verilator simulation using `make CONFIG=SmallBoomConfig`. Doing so required that I run `export CPATH="$CPATH":"$CHIPYARD_ROOT"/riscv-tools-install/include`. This allowed `g++` to find the fesvr header files. I expected `env.sh` to do so but using the Chipyard Docker container as a build environment may be unexpected.

abejgonzalez, 2020-10-23:
While you can use the Docker image to have most/all of the dependencies installed, we mainly use the Docker image just for CI purposes. However, with that being said... the Docker image doesn't have the Chipyard repo pre-installed on it... what "original" version of the repo are you referring to? Maybe Im missing something here? Also, what is in your `env.sh`? If the $RISCV variable is set properly then the simulator should point to the headers automatically (https://github.com/ucb-bar/chipyard/blob/a3c0e3a0b261c90618d45e0ceb38a64677af5089/sims/common-sim-flags.mk#L10).

knipegp, 2020-10-23:
Here is the content of `env.sh`:

```
# line auto-generated by init-submodules-no-riscv-tools.sh
PATH=$( realpath $(dirname ${BASH_SOURCE[0]:-${(%):-%x}}) )/software/firemarshal:$PATH
# line auto-generated by build-toolchains.sh
source $( realpath $(dirname ${BASH_SOURCE[0]:-${(%):-%x}}) )/env-riscv-tools.sh
```

I think I'm causing my own problems with my personal Docker automation so I wouldn't worry about the `RISCV` environment variable problem.

When I refer to the "original" version of the repo, I'm talking about the first working copy that I cloned and used locally. I was switching between tags and master which may have caused version discrepancies between the Chipyard repository and the git submodules that `git status` or `git clean` can't catch.

To summarize, I think my problems were just user error. I have things working now. Sorry for clogging up your issues.

abejgonzalez, 2020-10-23:
No problem at all.

closed 2020-11-23 by alonamid

SaltedFishLZ, 2020-12-12:
I think we also need to add `export LIBRARY_PATH=$RISCV/lib` to make ld work correctly for link static libraries (In this case, -lfesvr will cause an error because the library fesvr.a is a static library. $LD_LIBRARY_PATH only works for dynamic libraries *.so.)

SaltedFishLZ, 2020-12-12:
> While you can use the Docker image to have most/all of the dependencies installed, we mainly use the Docker image just for CI purposes. However, with that being said... the Docker image doesn't have [...]

I think if you can provide your docker images, that will be great since we can pull them from the docker hub or other sites.