ucb-bar/chipyard #610: Fail when verilator version is too old

jerryz123, 2020-06-19:
**Impact**: software

**Description**
When detected verilator version is too old, fail loudly, and provide user with minimum supported verilator version,

**What is a motivating example for changing the behavior?**
We've somehow become very dependent on bleeding edge versions of verilator (4.034 was only published in May of this year).
We've also stopped providing the script to download and install our own verilator.
As a result, there is a high probability that verilator builds will fail, and the current system does not indicate anywhere that this might be due to an old verilator version.

xubaqian, 2020-11-03:
Sorry to disturb you again. I install a 4.034 version of verilator, but either the latest chipyard or the `c576a7e`  version(commited on 31 May. I notice you opened this issue on 20 Jun) builds fail. The errors are:
```
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62247:74: error: expected ‘;’ at end of member declaration
 tatic void _settle__TOP__9999(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_C
                                                                     ^
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62247:76: error: ‘VL_ATTR_COLD’ does not name a type
 tic void _settle__TOP__9999(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_COL
                                                                     ^
In file included from /usr/local/share/verilator/include/verilated_vcd_c.h:26:0,
                 from /home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h:4,
                 from <command-line>:0:
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62248:19: error: ‘VL_CACHE_LINE_BYTES’ was not declared in this scope
 } VL_ATTR_ALIGNED(VL_CACHE_LINE_BYTES);
                   ^
/usr/local/share/verilator/include/verilatedos.h:37:62: note: in definition of macro ‘VL_ATTR_ALIGNED’
 # define VL_ATTR_ALIGNED(alignment) __attribute__ ((aligned (alignment)))
                                                              ^
VTestHarness.mk:74: recipe for target 'SimDRAM.o' failed
make[1]: *** [SimDRAM.o] Error 1
make[1]: Leaving directory '/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig'
Makefile:135: recipe for target '/home/xqz/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig' failed
make: *** [/home/xqz/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig] Error 2
```
I'm somewhat worry about my experiment. Can you tell me what version of chipyard and verilator could work, or how to fix the error?  Thank you in advance!

lsteveol, 2020-11-03:
@xubaqian 
I've cloned chipyard several times across Ubuntu. Currently 20.04 is my version and I have followed the following with success

https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

This commit should work 19152d3b73f1c2255ce378501f6c01f9363b75e9.

Hope that helps.

jerryz123, 2020-11-03:
@xubaqian Try deleting the `generated-src` directory, and then rerunning.

xubaqian, 2020-11-03:
@lsteveol @jerryz123 
Thank you very much! After installing these dependencies in Ubuntu 16.04, it finally works!

jerryz123, 2023-02-17:
Conda-env for most users removes need for this,

closed 2023-02-17 by jerryz123