ucb-bar/chipyard #616: Top AXI slave port

pbert519, 2020-06-29:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* 
*OS:* 

I try to generate to generate verilog wit the sim/verilator makefile, which has the AXI Slave port available in the ChipTop module. The configuration is the following:
`   class RocketConfig extends Config(`
    `new chipyard.iobinders.WithUARTAdapter ++ `                     
    `new chipyard.iobinders.WithTieOffInterrupts ++   `              
    `new chipyard.iobinders.WithBlackBoxSimMem ++  `             
    `new chipyard.iobinders.WithTiedOffDebug ++     `                
    `new chipyard.iobinders.WithSimSerial ++            `           
    `new chipyard.iobinders.WithTieOffL2FBusAXI ++`
    `new testchipip.WithTSI ++                                      `
    `new chipyard.config.WithBootROM ++                    `     
    `new chipyard.config.WithUART ++                         `       
    `new chipyard.config.WithL2TLBs(1024) ++             `      
    `new freechips.rocketchip.subsystem.WithNoMMIOPort ++  `        
    `//new freechips.rocketchip.subsystem.WithNoSlavePort ++      `   
    `new freechips.rocketchip.subsystem.WithInclusiveCache ++     ` 
    `new freechips.rocketchip.subsystem.WithNExtTopInterrupts(0) ++ `
    `new freechips.rocketchip.subsystem.WithNBigCores(1) ++        ` 
    `new freechips.rocketchip.subsystem.WithCoherentBusTopology ++  `
    `new freechips.rocketchip.system.BaseConfig)`                   

**What is the current behavior?**
During '[info] running barstools.tapeout.transforms.GenerateTopAndHarness' i get the following errors:
'[error] (run-main-0) firrtl.passes.PassExceptions:' 
'[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[ChipTop.scala 71:68:chipyard.TestHarness.RocketConfig.fir@278078.4] : [module ChipTop]  Reference system is not fully initialized.'
'[error]    : system.l2_frontend_bus_axi4.0.aw.bits.size <= VOID'
This is just an example, there is an error for every AXI Signal. I attached the full log here:
[log.txt](https://github.com/ucb-bar/chipyard/files/4846481/log.txt)

**What is the expected behavior?**
A generated verilog ChipTop module with the AXI Slave as IO.

**Other information**
Actually i tried also to activate the MMIO AXI Port, and WithSimAXIMMIO as IOBinder and i got the same errors.

jerryz123, 2020-06-29:
The error indicates that the design has top level input IOs that are undriven. This happens because you modified the config to add these additional IOs, but did not add the config iobinder fragment which specifies how to drive these IOs.

For the AXI Slave IO, you should *remove* the `freechips.rocketchip.subsystem.WithNoSlavePort` config and *add* the `chipyard.iobinders.WithTieOffL2FBusAXI` fragment to your config.
For the AXI MMIO port, you have to *remove* the `freechips.rocketchip.subsystem.WithNoMMIOPort` config and *add* the `chipyard.iobinders.WithSimAXIMMIO` fragment to your config.

dalance, 2020-06-30:
I have the same issue. I tried to remove `freechips.rocketchip.subsystem.WithNoMMIOPort` and add `chipyard.iobinders.WithSimAXIMMIO` like below:

```diff
diff --git a/generators/chipyard/src/main/scala/config/RocketConfigs.scala b/generators/chipyard/src/main/scala/config/RocketConfigs.scala
index 954c531..d279aa8 100644
--- a/generators/chipyard/src/main/scala/config/RocketConfigs.scala
+++ b/generators/chipyard/src/main/scala/config/RocketConfigs.scala
@@ -17,7 +17,8 @@ class RocketConfig extends Config(
   new chipyard.config.WithBootROM ++                             // use default bootrom
   new chipyard.config.WithUART ++                                // add a UART
   new chipyard.config.WithL2TLBs(1024) ++                        // use L2 TLBs
-  new freechips.rocketchip.subsystem.WithNoMMIOPort ++           // no top-level MMIO master port (overrides default set in rocketchip)
+  //new freechips.rocketchip.subsystem.WithNoMMIOPort ++           // no top-level MMIO master port (overrides default set in rocketchip)
+  new chipyard.iobinders.WithSimAXIMMIO ++
   new freechips.rocketchip.subsystem.WithNoSlavePort ++          // no top-level MMIO slave port (overrides default set in rocketchip)
   new freechips.rocketchip.subsystem.WithInclusiveCache ++       // use Sifive L2 cache
   new freechips.rocketchip.subsystem.WithNExtTopInterrupts(0) ++ // no external interrupts

```

The modified `RocketConfig` failed at chipyard v1.3.0, and succeed at chipyard v1.2.0.

jerryz123, 2020-06-30:
Sorry, I misunderstood the error.
It looks like you are right, 1.3.0 broke this feature. Changes in the inheritance structure of RocketChip broke the reflection-based construction of the IOBinders.
A temporary patch is to go to the definition of `WithSimAXIMMIO` and `WithTieOffL2AXIBus`, in `generators/chipyard/src/main/scala/iobinders.scala`, and change these lines
```
class WithSimAXIMem extends OverrideIOBinder({
class WithSimAXIMMIO extends OverrideIOBinder({
class WithTieOffL2FBusAXI extends OverrideIOBinder({

```
to 
```
class WithSimAXIMem extends ComposeIOBinder({
class WithSimAXIMMIO extends ComposeIOBinder({
class WithTieOffL2FBusAXI extends ComposeIOBinder({
```
Note that this solution is temporary, and should work for simple designs. A more robust solution will be pushed into dev soon.
We should also add testing for these configs so they don't fall through the holes again.

dalance, 2020-06-30:
Thank you for your quick reply.
I confirmed the patch works fine at `RocketConfig` and `BoomConfig`.

pbert519, 2020-06-30:
Thank you for the fast response. It solves the compiler errors when using a RocketConfig with slave port activated. 
Unfortunately the slave port is not exposed to the DigitalTop or ChipTop module. Is that even an expected behavior?

jerryz123, 2020-07-03:
The final fix for this problem was just merged into the `dev` branch. https://github.com/ucb-bar/chipyard/pull/618
This also fixes the ports not being exposed through both `DigitalTop` and `ChipTop`

closed 2020-07-03 by jerryz123