ucb-bar/chipyard #617: How to boot soc from spi flash?

bg193, 2020-06-30:
How to set reset vector pointer to spiflash address and run bootloader from spiflash. Can anyone help? Thanks!

jerryz123, 2020-06-30:
If you want to run your second stage bootloader out of SPI, you can modify the source code for the bootrom defined in `generators/testchipip/bootrom/bootrom.S`. There is code here which by default sets `mepc` to `DRAM_BASE`. You can change this to point to the base address of your spi flash memory instead.

jerryz123, 2020-06-30:
Alternatively, you can also set `reset_vector` by modifying the `BootROMParams` parameters set by `chipyard.config.WithBootRom`. I would create a copy of `chipyard.config.WithBootROMParams` named `WithBootFromSPI` which sets `BootROMParams` to `BootROMParams(hang = 0xspi_addr, contentFileName="path/to/dummy/file")`. Adding `WithBootFromSPI` to your custom config in place of `WithBootRom` should set the system-wide `global_reset_vector` to the custom address set by `hang`.

bg193, 2020-07-02:
> Alternatively, you can also set `reset_vector` by modifying the `BootROMParams` parameters set by `chipyard.config.WithBootRom`. I would create a copy of `chipyard.config.WithBootROMParams` named [...]

I found that  sivefive-blocks contain spiflash, does it support linear address access, So CPU can read flash memory directly like common memory?  Write bootloader at 0x20000000(flash memory mapped address) and  modify reset_vertor to 0x2000000, cpu boot from flash, did it works?

jerryz123, 2020-07-02:
It seems like this should work, the SPI Flash memory is marked as executable, so CPU should be able to execute from that region.

bg193, 2020-07-02:
Thanks , Really appreciate your help.

closed 2020-07-04 by bg193