ucb-bar/chipyard #622: Error when building: Invalid Option: --max-num-width

wky9710, 2020-07-05:
unknown

**Tell us about your environment:**
*OS:* Linux server17 4.15.0-106-generic #107~16.04.1-Ubuntu SMP Thu Jun 4 15:40:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**What is the current behavior?**
I was following instructions here, building RISCV-BOOM
https://docs.boom-core.org/en/latest/#  --> Listing 1 Quick-Start Code
and the last step:
make CONFIG=LargeBoomConfig
and I met this problem.
details below

Sorry I'm really new to this area, thanks for your patience :)

**Other information**
*****************
rm -rf /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig
mkdir -p /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig
verilator --cc --exe -O3 -CFLAGS " -O1 -std=c++11 -I/home/chengjian/opt/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR" -CFLAGS "-I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h" -LDFLAGS " -L/home/wangkunyu/work/chipyard/sims/verilator -lpthread" /home/chengjian/opt/lib/libfesvr.a /home/wangkunyu/work/chipyard/tools/DRAMSim2/libdramsim.a  --top-module TestHarness -Wno-fatal --assert --output-split 10000 --output-split-cfuncs 100 --max-num-width 1048576 -f /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/sim_files.common.f /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.top.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.harness.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.top.mems.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.harness.mems.v +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\)  -o /home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig -Mdir /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig -CFLAGS "-include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h"
%Error: Invalid Option: --max-num-width
%Error: Command Failed /usr/local/bin/verilator_bin --cc --exe -O3 -CFLAGS ' -O1 -std=c++11 -I/home/chengjian/opt/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR' -CFLAGS '-I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h' -LDFLAGS ' -L/home/wangkunyu/work/chipyard/sims/verilator -lpthread' /home/chengjian/opt/lib/libfesvr.a /home/wangkunyu/work/chipyard/tools/DRAMSim2/libdramsim.a --top-module TestHarness -Wno-fatal --assert --output-split 10000 --output-split-cfuncs 100 --max-num-width 1048576 -f /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/sim_files.common.f /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.top.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.harness.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.top.mems.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.harness.mems.v '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' -o /home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig -Mdir /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig -CFLAGS '-include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h'
Makefile:120: recipe for target '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.mk' failed
make: *** [/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.mk] Error 10
*****************

jerryz123, 2020-07-05:
This is due to incompatible verilator version. 4.034 is minimum supported version. 
In the future we would like to add a more informative warning on Verilator version mismatch #610

wky9710, 2020-07-05:
updating verilator works, but another error occurs
error info below
by the way, according to https://github.com/ucb-bar/chipyard/issues/233, I already cleaned verilator and tried again, but it didn't work and was the same error
seems another software is not updated? I'm working on my lab's server, so I'm not sure if all softwares are updated.
Many thanks (:3|____)
***************************
make VM_PARALLEL_BUILDS=1 -C /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig -f VTestHarness.mk
make[1]: Entering directory '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/home/chengjian/opt/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h  -Os -c -o SimDTM.o /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/SimDTM.cc
In file included from /usr/local/share/verilator/include/verilated.h:27:0,
                 from /usr/local/share/verilator/include/verilated_vcd_c.h:23,
                 from /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h:4,
                 from <command-line>:0:
/usr/local/share/verilator/include/verilatedos.h:259:0: warning: "__STDC_FORMAT_MACROS" redefined
 #define __STDC_FORMAT_MACROS
 ^
<command-line>:0:0: note: this is the location of the previous definition
/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/SimDTM.cc: In function ‘int debug_tick(unsigned char*, unsigned char, int*, int*, int*, unsigned char, unsigned char*, int, int)’:
/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/SimDTM.cc:26:43: error: no matching function for call to ‘dtm_t::dtm_t(PLI_INT32&, PLI_BYTE8**&)’
       dtm = new dtm_t(info.argc, info.argv);
                                           ^
In file included from /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/SimDTM.cc:3:0:
/home/chengjian/opt/include/fesvr/dtm.h:17:3: note: candidate: dtm_t::dtm_t(const std::vector<std::__cxx11::basic_string<char> >&)
   dtm_t(const std::vector<std::string>& args);
   ^
/home/chengjian/opt/include/fesvr/dtm.h:17:3: note:   candidate expects 1 argument, 2 provided
VTestHarness.mk:76: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig'
Makefile:135: recipe for target '/home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig' failed
make: *** [/home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig] Error 2

***************************

jerryz123, 2020-07-05:
Its possible there is version incompatibility between the preinstalled version of fesvr on the lab server, and the version which Chipyard requires. You should install a new copy of the RISC-V toolchain using `scripts/build-toolchains.sh`, and make sure your `RISCV` variable points to that

wky9710, 2020-07-06:
Sorry I'm here again lol
I've updated RISCV variable, and another error... 
according to #233 & #232, RISCV and env.sh are updated, verilator cleaned, but it seems the error still comes from riscv tools...
details below.
// dtm.h not found, and I indeed can't find it, but I installed riscv-tools following the steps...
Really thanks for your patience_(:з)∠)_

`wangkunyu@server17:~/work/chipyard/sims/verilator$ make CONFIG=LargeBoomConfig
make VM_PARALLEL_BUILDS=1 -C /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig -f VTestHarness.mk
make[1]: Entering directory '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/home/wangkunyu/work/chipyard/riscv-tools-install//include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h  -Os -c -o SimDTM.o /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/SimDTM.cc
In file included from /usr/local/share/verilator/include/verilated.h:27:0,
                 from /usr/local/share/verilator/include/verilated_vcd_c.h:23,
                 from /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h:4,
                 from <command-line>:0:
/usr/local/share/verilator/include/verilatedos.h:259:0: warning: "__STDC_FORMAT_MACROS" redefined
 #define __STDC_FORMAT_MACROS
 ^
<command-line>:0:0: note: this is the location of the previous definition
/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/SimDTM.cc:3:23: fatal error: fesvr/dtm.h: No such file or directory
compilation terminated.
VTestHarness.mk:76: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig'
Makefile:135: recipe for target '/home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig' failed
make: *** [/home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig] Error 2
`

jerryz123, 2020-07-06:
What are the contents of the `$RISCV` directory? `dtm.h` should be located in `$RISCV/include/fesvr/dtm.h`

wky9710, 2020-07-13:
Sorry for replying late
It was riscv-tools building error, maybe ld's version(I didn't notice the building failed orz), and I created another issue in riscv-tools [https://github.com/riscv/riscv-gnu-toolchain/issues/669](url), waiting for reply
And somehow I managed to install it on another server today... Anyhow the install succeeded. Really thanks for your patience :)

closed 2020-07-13 by wky9710

xubaqian, 2020-10-31:
@wky9710 sorry, after reading this issue I'm still not sure how to fix the error.

Because https://boringssl.googlesource.com/boringssl cannot be accessed in China, I commented out the following line in `scripts/build-toolchains.sh `
`#SRCDIR="$(pwd)/toolchains" module_all qemu --prefix="${RISCV}" --target-list=riscv64-softmmu`

Then it shows `Toolchain Build Complete`. But in the last step `make CONFIG=LargeBoomConfig` I encounter the same error as you met at first. Should I update the verilator or ld ?

xubaqian, 2020-11-01:
@jerryz123 Hello. I update the verilator version to 4.034, rebuild the toolchain and make clean in sim/verilator. I'm sure the `$RISCV` and `$PATH` are correct, but still encounter errors：
```
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62245:74: error: expected ‘;’ at end of member declaration
 tatic void _settle__TOP__9997(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_C
                                                                     ^
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62245:76: error: ‘VL_ATTR_COLD’ does not name a type
 tic void _settle__TOP__9997(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_COL
                                                                     ^
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62246:74: error: expected ‘;’ at end of member declaration
 tatic void _settle__TOP__9998(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_C
                                                                     ^
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62246:76: error: ‘VL_ATTR_COLD’ does not name a type
 tic void _settle__TOP__9998(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_COL
                                                                     ^
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62247:74: error: expected ‘;’ at end of member declaration
 tatic void _settle__TOP__9999(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_C
                                                                     ^
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62247:76: error: ‘VL_ATTR_COLD’ does not name a type
 tic void _settle__TOP__9999(VTestHarness__Syms* __restrict vlSymsp) VL_ATTR_COL
                                                                     ^
In file included from /usr/local/share/verilator/include/verilated_vcd_c.h:26:0,
                 from /home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/verilator.h:4,
                 from <command-line>:0:
/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig/VTestHarness.h:62248:19: error: ‘VL_CACHE_LINE_BYTES’ was not declared in this scope
 } VL_ATTR_ALIGNED(VL_CACHE_LINE_BYTES);
                   ^
/usr/local/share/verilator/include/verilatedos.h:37:62: note: in definition of macro ‘VL_ATTR_ALIGNED’
 # define VL_ATTR_ALIGNED(alignment) __attribute__ ((aligned (alignment)))
                                                              ^
VTestHarness.mk:74: recipe for target 'SimDRAM.o' failed
make[1]: *** [SimDRAM.o] Error 1
make[1]: Leaving directory '/home/xqz/chipyard/sims/verilator/generated-src/chipyard.TestHarness.LargeBoomConfig/chipyard.TestHarness.LargeBoomConfig'
Makefile:135: recipe for target '/home/xqz/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig' failed
make: *** [/home/xqz/chipyard/sims/verilator/simulator-chipyard-LargeBoomConfig] Error 2
```
simply removing `VL_ATTR_C` and replacing `VL_CACHE_LINE_BYTES` with a constant will pass these errors, but there will be other errors in other files. 

There must be something error... Thank you in advance.