ucb-bar/chipyard #623: About IceNet

bg193, 2020-07-05:
What is the status of the IceNet project，very little information about it，did IceNet is finished？

zhemao, 2020-07-07:
Can you elaborate on what you mean by "finished"?

IceNet is feature complete and stable. There is ongoing research using it for distributed architecture research, although not with a specific focus on IceNIC features. We will continue to fix bugs, and potentially add features if they are found to be necessary within the research work.

bg193, 2020-07-14:
> Can you elaborate on what you mean by "finished"?
>
> IceNet is feature complete and stable. There is ongoing research using it for distributed architecture research, although not with a specific focus [...]

Where is the difference between IceNet and Ethernet MAC?  If  IceNet connect PHY module can access ethernet?

zhemao, 2020-07-14:
IceNet provides an Ethernet MAC, yes. If you attach it to a PHY (assuming compatible interfaces), you can connect your SoC to the internet.

a0u, 2020-07-14:
> IceNet provides an Ethernet MAC, yes. If you attach it to a PHY (assuming compatible interfaces), you can connect your SoC to the internet.

Adding support for MII could be an interesting project for someone.

closed 2020-08-10 by bg193