ucb-bar/chipyard #624: How to add spi interface?

bg193, 2020-07-07:
Now qspi0  fixed to connect flash memory, how to add more qspi interface for other purpose？

abejgonzalez, 2020-08-17:
To create a normal SPI interface you should follow the example for the SPI flash configuration, but instead, use the normal SPI variables.

1. In the SiFive Blocks repo (https://github.com/sifive/sifive-blocks/blob/9f5e4fc79a6a8ad6987b2c4881b3e2703cd5b777/src/main/scala/devices/spi/SPIPeriphery.scala#L10), there is a `PeripherySPIKey` that you should be able to assign to in a config. fragment. Like the following: 

```
class WithSPISetup extends Config((site, here, up) => {
  case PeripherySPIKey => Seq(SPIParams(rAddress = 0xa0000))
})
```

That should create the SPI interface on the chip.

2. You should add the `with sifive.blocks.devices.spi.HasPeripherySPIFlash` trait to the `DigitalTop` module (https://github.com/ucb-bar/chipyard/blob/19152d3b73f1c2255ce378501f6c01f9363b75e9/generators/chipyard/src/main/scala/DigitalTop.scala#L15-L28).

3. You should create an IOBinder to add IOCells for the SPI interface and determine if you need to connect something to the SPI interface. You can follow the `WithSimSPIFlashModel` example (https://github.com/ucb-bar/chipyard/blob/19152d3b73f1c2255ce378501f6c01f9363b75e9/generators/chipyard/src/main/scala/IOBinders.scala#L220-L226).

Hope that gets you started.

bg193, 2020-08-29:
thanks a lot !!!

closed 2020-08-29 by bg193

Usutatsu, 2022-08-24:
I am trying to solve the same problem.
In my case I want to connect a new SPI interface to a PMOD connector.
Is there any reference or other material that can help me with a step-by-step implementation?

Also, in which file should I add the 1st step PeripherySPIKey description?