ucb-bar/chipyard #628: WithSimAXIMMIO lost part of axi port

Suedongchu, 2020-07-12:
I want to use the MMIO in GemminiRocketConfig：
```scala
class GemminiRocketConfig extends Config(
  // new chipyard.iobinders.WithUARTAdapter ++
  new chipyard.iobinders.WithTieOffInterrupts ++
  new chipyard.iobinders.WithBlackBoxSimMem ++
  new chipyard.iobinders.WithTiedOffDebug ++
  // new chipyard.iobinders.WithSimSerial ++
  // new testchipip.WithTSI ++
  new chipyard.config.WithBootROM ++
  // new chipyard.config.WithUART ++
  new chipyard.config.WithL2TLBs(1024) ++
  new gemmini.DefaultGemminiConfig ++                        // use Gemmini systolic array GEMM accelerator
  // new freechips.rocketchip.subsystem.WithNoMMIOPort ++
  new chipyard.iobinders.WithSimAXIMMIO ++
  new chipyard.iobinders.WithTieOffL2FBusAXI ++
  new freechips.rocketchip.subsystem.WithDefaultSlavePort ++
  // new freechips.rocketchip.subsystem.WithNoSlavePort ++
  new freechips.rocketchip.subsystem.WithInclusiveCache ++
  new freechips.rocketchip.subsystem.WithNExtTopInterrupts(4) ++
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++
  new freechips.rocketchip.subsystem.WithCoherentBusTopology ++
  new freechips.rocketchip.system.BaseConfig)
```
It seems that some MMIO-AXI ports are missing in the final verilog module DigitalTop
```verilog
module DigitalTop( // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410037.2]
  input         clock, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410038.4]
  input         reset, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410039.4]
  input         mem_axi4_0_aw_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_aw_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [3:0]  mem_axi4_0_aw_bits_id, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [31:0] mem_axi4_0_aw_bits_addr, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [7:0]  mem_axi4_0_aw_bits_len, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [2:0]  mem_axi4_0_aw_bits_size, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [1:0]  mem_axi4_0_aw_bits_burst, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_aw_bits_lock, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [3:0]  mem_axi4_0_aw_bits_cache, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [2:0]  mem_axi4_0_aw_bits_prot, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [3:0]  mem_axi4_0_aw_bits_qos, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input         mem_axi4_0_w_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_w_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [63:0] mem_axi4_0_w_bits_data, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [7:0]  mem_axi4_0_w_bits_strb, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_w_bits_last, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_b_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input         mem_axi4_0_b_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input  [3:0]  mem_axi4_0_b_bits_id, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input  [1:0]  mem_axi4_0_b_bits_resp, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input         mem_axi4_0_ar_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_ar_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [3:0]  mem_axi4_0_ar_bits_id, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [31:0] mem_axi4_0_ar_bits_addr, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [7:0]  mem_axi4_0_ar_bits_len, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [2:0]  mem_axi4_0_ar_bits_size, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [1:0]  mem_axi4_0_ar_bits_burst, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_ar_bits_lock, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [3:0]  mem_axi4_0_ar_bits_cache, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [2:0]  mem_axi4_0_ar_bits_prot, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output [3:0]  mem_axi4_0_ar_bits_qos, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  output        mem_axi4_0_r_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input         mem_axi4_0_r_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input  [3:0]  mem_axi4_0_r_bits_id, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input  [63:0] mem_axi4_0_r_bits_data, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input  [1:0]  mem_axi4_0_r_bits_resp, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input         mem_axi4_0_r_bits_last, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410041.4]
  input         mmio_axi4_0_aw_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  output        mmio_axi4_0_aw_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  output [3:0]  mmio_axi4_0_aw_bits_id, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  output [30:0] mmio_axi4_0_aw_bits_addr, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
[... 21 lines cut ...]
  input  [3:0]  mmio_axi4_0_r_bits_id, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  input  [63:0] mmio_axi4_0_r_bits_data, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  input  [1:0]  mmio_axi4_0_r_bits_resp, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  input         mmio_axi4_0_r_bits_last, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410042.4]
  input         resetctrl_hartIsInReset_0, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410045.4]
  input         debug_clock, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input         debug_reset, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  output        debug_clockeddmi_dmi_req_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input         debug_clockeddmi_dmi_req_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input  [6:0]  debug_clockeddmi_dmi_req_bits_addr, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input  [31:0] debug_clockeddmi_dmi_req_bits_data, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input  [1:0]  debug_clockeddmi_dmi_req_bits_op, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input         debug_clockeddmi_dmi_resp_ready, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  output        debug_clockeddmi_dmi_resp_valid, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  output [31:0] debug_clockeddmi_dmi_resp_bits_data, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  output [1:0]  debug_clockeddmi_dmi_resp_bits_resp, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input         debug_clockeddmi_dmiClock, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input         debug_clockeddmi_dmiReset, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  output        debug_ndreset, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  output        debug_dmactive, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input         debug_dmactiveAck, // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410046.4]
  input  [3:0]  interrupts // @[:chipyard.TestHarness.GemminiRocketConfig.fir@410047.4]
);
```
How can I add the lock/cache/prot/qos signals of MMIO-AXI just like  mem_axi4_0_ar_bits_lock/cache/prot /qos ？

jerryz123, 2020-07-12:
The TileLink-to-AXI4 converter cannot support those signals, for various reasons. 

```
      arw.lock  := UInt(0) // not exclusive (LR/SC unsupported b/c no forward progress guarantee)
      arw.cache := UInt(0) // do not allow AXI to modify our transactions
      arw.prot  := AXI4Parameters.PROT_PRIVILEDGED
      arw.qos   := UInt(0) // no QoS
```

https://github.com/chipsalliance/rocket-chip/blob/52843da3d8daaba5c32c488ee21d20aa2f640772/src/main/scala/tilelink/ToAXI4.scala#L170-L173

jerryz123, 2020-07-12:
To add those hard-wired signals back to `DigitalTop`, you should add `new chipyard.config.WithDontTouchPorts` to your Config definition

Suedongchu, 2020-07-12:
I add
```scala
new chipyard.iobinders.WithDontTouchPorts ++
```
It works
Thank you.
And is there a more detailed document for how the Config Class works？

jerryz123, 2020-07-12:
There is some information here: https://chipyard.readthedocs.io/en/dev/Advanced-Concepts/CDEs.html

closed 2020-08-17 by abejgonzalez