ucb-bar/chipyard #634: make Config=GemminiRocketConfig -j8 error

xiongwu12345, 2020-07-29:
Hello,I am a beginner at Chipyard. I download the repository ,and cd/sims/verilator directory,run the make Config=GemminiRocketConfig -j8 command.But get below errors: 

[info] Set current project to utilities (in build file:/home/xiongwu/Desktop/chipyard/)
[info] Compiling 1 Scala source to /home/xiongwu/Desktop/chipyard/generators/rocket-chip/macros/target/scala-2.12/classes ...
[info] Compiling 1 Scala source to /home/xiongwu/Desktop/chipyard/generators/rocket-chip/api-config-chipsalliance/build-rules/sbt/target/scala-2.12/classes ...
[info] Compiling 5 Scala sources to /home/xiongwu/Desktop/chipyard/tools/chisel3/coreMacros/target/scala-2.12/classes ...
[error] ## Exception when compiling 1 sources to /home/xiongwu/Desktop/chipyard/generators/rocket-chip/macros/target/scala-2.12/classes
[error] Error accessing /home/xiongwu/Desktop/chipyard/lib/firrtl.jar
[error] scala.tools.nsc.classpath.AggregateClassPath.$anonfun$list$3(AggregateClassPath.scala:99)
[error] scala.collection.Iterator.foreach(Iterator.scala:941)
[error] scala.collection.Iterator.foreach$(Iterator.scala:941)
[error] scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[error] scala.collection.IterableLike.foreach(IterableLike.scala:74)
[error] scala.collection.IterableLike.foreach$(IterableLike.scala:73)
[error] scala.collection.AbstractIterable.foreach(Iterable.scala:56)
[error] scala.tools.nsc.classpath.AggregateClassPath.list(AggregateClassPath.scala:87)
[error] scala.tools.nsc.util.ClassPath.list(ClassPath.scala:36)
[error] scala.tools.nsc.util.ClassPath.list$(ClassPath.scala:36)
[error] scala.tools.nsc.classpath.AggregateClassPath.list(AggregateClassPath.scala:30)
[error] scala.tools.nsc.symtab.SymbolLoaders$PackageLoader.doComplete(SymbolLoaders.scala:284)
[error] scala.tools.nsc.symtab.SymbolLoaders$SymbolLoader.complete(SymbolLoaders.scala:230)
[error] scala.reflect.internal.Symbols$Symbol.info(Symbols.scala:1542)
[error] scala.reflect.internal.Mirrors$RootsBase.init(Mirrors.scala:257)
[error] scala.tools.nsc.Global.rootMirror$lzycompute(Global.scala:74)
[error] scala.tools.nsc.Global.rootMirror(Global.scala:72)
[error] scala.tools.nsc.Global.rootMirror(Global.scala:44)
[error] scala.reflect.internal.Definitions$DefinitionsClass.ObjectClass$lzycompute(Definitions.scala:295)
[error] scala.reflect.internal.Definitions$DefinitionsClass.ObjectClass(Definitions.scala:295)
[error] scala.reflect.internal.Definitions$DefinitionsClass.init(Definitions.scala:1480)
[error] scala.tools.nsc.Global$Run.<init>(Global.scala:1199)
[error] xsbt.ZincCompiler$ZincRun.<init>(CallbackGlobal.scala:63)
[error] xsbt.CachedCompiler0.run(CompilerInterface.scala:128)
[error] xsbt.CachedCompiler0.run(CompilerInterface.scala:105)
[error] xsbt.CompilerInterface.run(CompilerInterface.scala:31)
[error] sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] java.lang.reflect.Method.invoke(Method.java:498)
[error] sbt.internal.inc.AnalyzingCompiler.call(AnalyzingCompiler.scala:237)
[error] sbt.internal.inc.AnalyzingCompiler.compile(AnalyzingCompiler.scala:111)
[error] sbt.internal.inc.AnalyzingCompiler.compile(AnalyzingCompiler.scala:90)
[error] sbt.internal.inc.MixedAnalyzingCompiler.$anonfun$compile$3(MixedAnalyzingCompiler.scala:82)
[error] scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error] sbt.internal.inc.MixedAnalyzingCompiler.timed(MixedAnalyzingCompiler.scala:133)
[error] sbt.internal.inc.MixedAnalyzingCompiler.compileScala$1(MixedAnalyzingCompiler.scala:73)
[error] sbt.internal.inc.MixedAnalyzingCompiler.compile(MixedAnalyzingCompiler.scala:116)
[error] sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileInternal$1(IncrementalCompilerImpl.scala:307)
[error] sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileInternal$1$adapted(IncrementalCompilerImpl.scala:307)
[error] sbt.internal.inc.Incremental$.doCompile(Incremental.scala:106)
[error] sbt.internal.inc.Incremental$.$anonfun$compile$4(Incremental.scala:87)
[error] sbt.internal.inc.IncrementalCommon.recompileClasses(IncrementalCommon.scala:116)
[error] sbt.internal.inc.IncrementalCommon.cycle(IncrementalCommon.scala:63)
[error] sbt.internal.inc.Incremental$.$anonfun$compile$3(Incremental.scala:89)
[error] sbt.internal.inc.Incremental$.manageClassfiles(Incremental.scala:134)
[error] sbt.internal.inc.Incremental$.compile(Incremental.scala:80)
[error] sbt.internal.inc.IncrementalCompile$.apply(Compile.scala:67)
[error] sbt.internal.inc.IncrementalCompilerImpl.compileInternal(IncrementalCompilerImpl.scala:311)
[error] sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileIncrementally$1(IncrementalCompilerImpl.scala:269)
[error] sbt.internal.inc.IncrementalCompilerImpl.handleCompilationError(IncrementalCompilerImpl.scala:159)
[error] sbt.internal.inc.IncrementalCompilerImpl.compileIncrementally(IncrementalCompilerImpl.scala:238)
[error] sbt.internal.inc.IncrementalCompilerImpl.compile(IncrementalCompilerImpl.scala:69)
[error] sbt.Defaults$.compileIncrementalTaskImpl(Defaults.scala:1549)
[error] sbt.Defaults$.$anonfun$compileIncrementalTask$1(Defaults.scala:1523)
[error] scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:40)
[error] sbt.std.Transform$$anon$4.work(System.scala:67)
[error] sbt.Execute.$anonfun$submit$2(Execute.scala:269)
[error] sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] sbt.Execute.work(Execute.scala:278)
[error] sbt.Execute.$anonfun$submit$1(Execute.scala:269)
[... 463 lines cut ...]
[error] 	at scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[error] 	at scala.collection.IterableLike.foreach(IterableLike.scala:74)
[error] Caused by: java.util.zip.ZipException: error in opening zip file
[error] 	at java.util.zip.ZipFile.open(Native Method)
[error] 	at java.util.zip.ZipFile.<init>(ZipFile.java:225)
[error] 	at java.util.zip.ZipFile.<init>(ZipFile.java:155)
[error] 	at java.util.zip.ZipFile.<init>(ZipFile.java:169)
[error] 	at scala.reflect.io.FileZipArchive.scala$reflect$io$FileZipArchive$$openZipFile(ZipArchive.scala:138)
[error] 	at scala.reflect.io.FileZipArchive.root$lzycompute(ZipArchive.scala:179)
[error] 	at scala.reflect.io.FileZipArchive.root(ZipArchive.scala:176)
[error] 	at scala.reflect.io.FileZipArchive.allDirs$lzycompute(ZipArchive.scala:212)
[error] 	at scala.reflect.io.FileZipArchive.allDirs(ZipArchive.scala:212)
[error] 	at scala.tools.nsc.classpath.ZipArchiveFileLookup.findDirEntry(ZipArchiveFileLookup.scala:76)
[error] 	at scala.tools.nsc.classpath.ZipArchiveFileLookup.list(ZipArchiveFileLookup.scala:63)
[error] 	at scala.tools.nsc.classpath.ZipArchiveFileLookup.list$(ZipArchiveFileLookup.scala:62)
[error] 	at scala.tools.nsc.classpath.ZipAndJarClassPathFactory$ZipArchiveClassPath.list(ZipAndJarFileLookupFactory.scala:56)
[error] 	at scala.tools.nsc.classpath.AggregateClassPath.$anonfun$list$3(AggregateClassPath.scala:91)
[error] (coreMacros / Compile / compileIncremental) scala.reflect.internal.FatalError: Error accessing /home/xiongwu/Desktop/chipyard/lib/firrtl.jar
[error] (rocketMacros / Compile / compileIncremental) scala.reflect.internal.FatalError: Error accessing /home/xiongwu/Desktop/chipyard/lib/firrtl.jar
[error] (rocketConfig / Compile / compileIncremental) scala.reflect.internal.FatalError: Error accessing /home/xiongwu/Desktop/chipyard/lib/firrtl.jar
[error] Total time: 3 s, completed Jul 29, 2020 10:21:13 AM
/home/xiongwu/Desktop/chipyard/common.mk:38: recipe for target '/home/xiongwu/Desktop/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/sim_files.f' failed
make: *** [/home/xiongwu/Desktop/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/sim_files.f] Error 1

Can anyone help me!Thank you so much!!

alonamid, 2020-07-31:
Do you have any previous installations of Scala cached on this machine?

colinschmidt, 2020-07-31:
Also maybe try removing the parallel make flag. It has caused issues in the past.

alonamid, 2020-08-17:
Closed due to no additional response. Feel free to re-open if needed.

closed 2020-08-17 by alonamid