ucb-bar/chipyard #637: "InclusiveCacheKey" is not defined

hyf6661669, 2020-08-01:
I met this error:
```
[error] /home/hyf/study/rv_cores/chipyard/generators/tracegen/src/main/scala/Configs.scala:7:8: object WithInclusiveCache is not a member of package freechips.rocketchip.subsystem
[error] import freechips.rocketchip.subsystem.{ExtMem, SystemBusKey, WithInclusiveCache, InclusiveCacheKey}
[error]        ^
[error] /home/hyf/study/rv_cores/chipyard/generators/tracegen/src/main/scala/Configs.scala:71:22: not found: value InclusiveCacheKey
[error]       val l2p = site(InclusiveCacheKey)
[error]                      ^
[error] two errors found
[error] (tracegen / Compile / compileIncremental) Compilation failed
```

And it seems that InclusiveCacheKey is not defined in rocket-chip/src/main/scala/subsystem/Configs.scala.

jerryz123, 2020-08-01:
`InclusiveCacheKey` is defined in `sifive-cache`. Did you initialize your submodules correctly?

hyf6661669, 2020-08-01:
> `InclusiveCacheKey` is defined in `sifive-cache`. Did you initialize your submodules correctly?

Thanks, some files in my projects are missing, problem solved !!

closed 2020-08-01 by hyf6661669