ucb-bar/chipyard #638: How to use the lates riscv-boom in chipyard

hyf6661669, 2020-08-01:
It seems that the submodule ```chipyard/generators/boom``` points to an older version of boom, but I want to use the latest version (SonicBoom). Could you please tell me how to achieve it ? Thanks!

[image]

jerryz123, 2020-08-01:
Oops, it points to a commit of BOOM right before we tagged the 3.0 release. The version there is still SonicBOOM/

hyf6661669, 2020-08-02:
> Oops, it points to a commit of BOOM right before we tagged the 3.0 release. The version there is still SonicBOOM/

If I want to use another version of BOOM, what should I do?

jerryz123, 2020-08-03:
What version of BOOM do you want to use?

hyf6661669, 2020-08-03:
> What version of BOOM do you want to use?

 For example, if I want to use a particular commit hash of BOOM, what should I do?

jerryz123, 2020-08-03:
You can go into the submodule and git checkout the commit hash you want. Keep in mind that you will run into version incompatibilities if you choose very old commits.

hyf6661669, 2020-08-03:
> You can go into the submodule and git checkout the commit hash you want. Keep in mind that you will run into version incompatibilities if you choose very old commits.

Thanks.

closed 2020-08-03 by hyf6661669