ucb-bar/chipyard #643: Are there known issues with submodules?

faddat, 2020-08-12:
On my laptop:
[image]

On the `dev` branch in CI:
https://gitlab.com/virgohardware/core/-/jobs/682118806#L8398

in CI, I do not run into issues with submodules in the `master` branch, just `dev`.  

In both cases, I haven't got the feeling that it's "working right" just yet.

faddat, 2020-08-12:
It's at least partially my script, will update so that it's no longer the case.

faddat, 2020-08-12:
On my laptop, it's because I wasn't using one of the submodule scripts, I think.  

Not quite certain about the Dockerfile though, that's likely to to take more work.

colinschmidt, 2020-08-12:
The default script ignores that submodule because not everyone has access to it.

faddat, 2020-08-13:
Ok, so in CI, I just thought to try disabling esp-tools, maybe the issue is with that step.  

**Update**: The issue was not with that step, the issue is with riscv-tools. 

Here's that step of the Dockerfile, and the one before it:

```
# Install Chipyard
RUN git clone https://github.com/ucb-bar/chipyard.git && \
        cd chipyard && \
        git checkout $BRANCH && \
        export MAKEFLAGS=-"j $(nproc)" && \
        ./scripts/init-submodules-no-riscv-tools.sh --no-firesim 1>/dev/null

# Install riscv-tools
RUN cd chipyard && \
        export MAKEFLAGS=-"j $(nproc)" && \
        ./scripts/build-toolchains.sh riscv-tools 1>/dev/null
```

On my macbook, I want to work with Chipyard in Intellij with the Scala plugin. I thought that I should let it configure itself more or less.  I will try using the script and see how that goes.

faddat, 2020-08-13:
Using  sh scripts/init-submodules-no-riscv-tools.sh  

still results in

[image]

colinschmidt, 2020-08-13:
You should be able to create commits that don't touch those submodules without initializing them.
What is the current status of those submodules in your repo?

faddat, 2020-08-13:
```
git submodule status
-651134f3c43ff8c0fa55a0f65faeb12b9500fcfa generators/ariane
-d77c2c3ff648bd5b18a932aa9f6b64dead7e3476 generators/boom
-caaf781ec9d69e45443e496046bc6ab439e3e54f generators/gemmini
-a989b69759137802b4c39e9ddebb90427455fb79 generators/hwacha
-705ca50690383aa589dc560a5e7c152af04c46ad generators/icenet
-b2b78c9f892a6196634eb3f1fbc443436c897a00 generators/nvdla
-1872f5d501221f13950aa2293939634a1e0d1735 generators/rocket-chip
-a94dcf3ae0a0440aade96bcdaa4da685352ae704 generators/sha3
-c1dee8234c23c8fc454108e59ecba20987f95cde generators/sifive-blocks
-4ebefa3e30ec44bd2f4ff82747025fb7b362b954 generators/sifive-cache
-4b15061b6fe77f6793603f799751f1f988144ef7 generators/testchipip
-c2d8e3a46e59222e115a1fdaa7267592e1d3c503 sims/firesim
-cf8b8f7d82e085b3ecb31865fa40fb3dd62554f2 software/coremark
-6c6a08f9790c660823e9f858ca87f93b2502fa44 software/firemarshal
-30290c3ed8014577fa2ad234fa1b731a082299fb software/nvdla-workload
-8f2812bebc15ae36f15cb8b80f4b50aa7d342b46 software/spec2017
-9f532293985d08b0c176d96c7b650e5f433780e1 toolchains/esp-tools/riscv-gnu-toolchain
-13384cac1e54828200067ff890f564a505a4ebb3 toolchains/esp-tools/riscv-isa-sim
-a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb toolchains/esp-tools/riscv-pk
-f1370d054389fc83974fc820985b5c51693b8f9d toolchains/esp-tools/riscv-tests
-04b249764b1bc0b784b9f7ff422084f4b53a07ef toolchains/libgloss
-4f59102571fce49af180cfc6d4cdd2b5df7bdb14 toolchains/qemu
-2855d823a6e93d50af604264b02ced951e80de67 toolchains/riscv-tools/riscv-gnu-toolchain
-543ff1245e1c2071ea9d0252a622c9cece88b7f5 toolchains/riscv-tools/riscv-gnu-toolchain-prebuilt
-9443c1dbac0301faf3a47c5e6914cc7dcb34983e toolchains/riscv-tools/riscv-isa-sim
-7c82a7b9d5b7d8b71e0a66826705ec141db718c3 toolchains/riscv-tools/riscv-openocd
-a3e4ac61d2b1ff37a22b9193b85d3b94273e80cb toolchains/riscv-tools/riscv-pk
-249796cec94d75ff10ca034153e206a319e87158 toolchains/riscv-tools/riscv-tests
-2ec7965b2ee051aaff03d5db21c6709aea4dd24e tools/DRAMSim2
-4a7cf86960425fbf0b6224865b745ff511dc5742 tools/axe
-7e6e19b8adf3b625b31b09173ecae5f634c83e1b tools/barstools
-1aa906fe168eb5ddca705ec955b27cf5c8856e4d tools/chisel-testers
-21ea734d809395962a8d3195a76377f6e44308f3 tools/chisel3
-56e2ff46b70521916c362799517f4ed8e67e9e88 tools/dromajo/dromajo-src
-211166e635861fb1937828aee38c166baf0840b5 tools/dsptools
-7c6f58d986e67b3d0662a4cd6654a68f9cc52cf9 tools/firrtl
-a881c07df6bceea462dbbd9a28e25721a1e88567 tools/firrtl-interpreter
-59b0f0f224ff4f1eb6ebb1b4dd7eaf1ab3fac2e5 tools/torture
-1c67bc846aafc3bdd707f76ead8cefd5f93e0376 tools/treadle
-9d83bbadc0caaa7f81b4929c4e32333fc5a8d900 vlsi/hammer
-f644138bab11075f267a3f1d72108da13c8a05ab vlsi/hammer-cadence-plugins
-67f57f120031e8c23d3d165f56c3a3c7a4064aa4 vlsi/hammer-mentor-plugins
-ef163445eec6362fa6a9bf6be0bd18a5d36c707e vlsi/hammer-synopsys-plugins
```

Is that the info you need?  I am not real sure about this.  

Thank you!

colinschmidt, 2020-08-13:
Hmm. that seems fine does `git status` show the `vlsi/hamme-*-plugins` at all? I've had some weird issues lately where it seems to mark them as deleted (not sure why) and I have to `git checkout -- vlsi/hammer-*-plugins` to fix it.

faddat, 2020-08-14:
We are experiencing the same thing, I think-- 

"weird issues".  

```
❯ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
❯ git commit .
error: 'generators/ariane' does not have a commit checked out
fatal: updating files failed
```

You'll note that it's now a different package, and no, I am not sure why.

colinschmidt, 2020-08-14:
It was slightly different for me, in that `git status` told me:
```
deleted:    vlsi/hammer-mentor-plugins
```
so that when I did `git checkout -- vlsi/hammer-mentor-plugins` the issue was resolved.

Perhaps this is a standard git issue that has a solution on stack overflow or other forum? Sorry if this is a silly question but have you already looked around the internet for this error message?

faddat, 2020-08-15:
Actually I've looked quite a bit for this error message and I've tried a few things but have come up empty. 

One of the things that I have tried is deinit ing whichever module is feeling troublesome, but no luck there. 

I could remove the sub module or delete it but I think that screws up git.

Let me kind of try and break down the two issues that I'm having with Chipyard sub modules:

1) My MacBook:

* Clone chipyard
* Use the noriscv script
* Unable to make commits in `dev` or `master`

2) Docker

This one is actually a bit more puzzling. I have been trying to make a Docker image for chipyard using Debian or Ubuntu.  During the riscv tools build, It dies, seemingly because of a submodule.

The only thing that I haven't tried is setting up a fresh VM, just because it's so time-consuming And so far in my experience if a script works outside of a doctor container it should also work inside.

Writing this out help me structure the situation a little bit and I guess I will try setting up a fresh VM.

faddat, 2020-08-16:
So I finished #606 and I have it on my MacBook.  

However, here's what happened from my fresh clone:

```bash
❯ git clone https://github.com/ucb-bar/chipyard
Cloning into 'chipyard'...
remote: Enumerating objects: 27, done.
remote: Counting objects: 100% (27/27), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 12061 (delta 8), reused 18 (delta 5), pack-reused 12034
Receiving objects: 100% (12061/12061), 3.75 MiB | 832.00 KiB/s, done.
Resolving deltas: 100% (6721/6721), done.
❯ cd chipyarrd
cd: no such file or directory: chipyarrd
❯ cd chipyard
❯ git checkout dev
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
❯ git branch cva6rename
❯ git checkout cva6rename
M	.circleci/check-commit.sh
M	.gitmodules
M	CHANGELOG.md
M	common.mk
M	docs/Customization/Custom-Core.rst
M	docs/Generators/Ariane.rst
M	docs/Generators/index.rst
M	generators/chipyard/src/main/scala/config/ArianeConfigs.scala
M	generators/chipyard/src/main/scala/example/TutorialTile.scala
M	generators/firechip/src/test/scala/ScalaTestSuite.scala
M	scripts/tutorial-patches/build.sbt.patch
M	sims/verilator/Makefile
Switched to branch 'cva6rename'
❯ git commit .
error: 'generators/ariane' does not have a commit checked out
fatal: updating files failed
```

I figured that if I just cloned it and made no reference to submodules, it'd work, but seems not.  I'm also going to run the init script and post the result here.

colinschmidt, 2020-08-17:
What if you just type `git commit`? Do you get the same error?

colinschmidt, 2020-08-17:
Also if its specifically that branch and only ariane now it might be because that branch is renaming the submodule, and you are missing some `git submodule sync`s.

colinschmidt, 2020-08-25:
I performed this exact script on my local mac osx laptop, in an entirely fresh directory.
I encountered the same problem when I typed `git commit .` but as expected when I typed `git commit` I had no issue.
I believe I suggested this course of action a while ago. Hopefully the same solution works for you.

```
~/ucb/tmp/chipyard$ touch testfile
~/ucb/tmp/chipyard$ git add testfile
~/ucb/tmp/chipyard$ git commit .
error: 'toolchains/esp-tools/riscv-gnu-toolchain' does not have a commit checked out
fatal: updating files failed
~/ucb/tmp/chipyard$ git commit
[master e6b8d90] add testfile
 1 file changed, 0 insertions(+), 0 deletions(-)
 create mode 100644 testfile
```

faddat, 2020-08-25:
huh.  Yes, you did suggest it, and I tried it, and it did not work.  I possibly I was in a non-fresh folder?  

Allow me to try it on my mac :)

Thanks!

faddat, 2020-08-25:
It worked. 

Thank you @colinschmidt

closed 2020-08-25 by faddat