ucb-bar/chipyard #644: What tools do you use when you work on Chipyard?

faddat, 2020-08-13:
Context: I'm trying to make HDLs more accessible to more people and am a bit of an amateur myself.  

Chipyard let me describe a custom SOC for the first time, and it was very exciting!

I did that with a pretty unconventional setup, an Arch Linux Docker image that installed all of the toolchains as binaries.  I haven't been able to get toolchain builds to work from source just yet.

I am interested in creating an **exact** guide to doing a few really basic tasks with Chipyard, like defining an SOC.  The docs that exist are good, but I am thinking of something a little bit shorter, and more exact, like:

* Get this IDE
* Get Chipyard
* Do exactly this
* boom, SOC

I may try and link it up with the CI system that I've put together, and the eventual goal would be to allow interested amateurs to try out things quickly, without the need to have a big Linux machine (my work on this has led to a pretty significant monthly spend on bare metal servers) or spend a lot of time setting up a development environment.  

So, chipyard devs:

**what tools do you use when you work on Chipyard?** 

:)

hyf6661669, 2020-08-13:
I use verilator for RTL simulation and gtkwave for examing waves. They are open-source software. Verilator is very competitive but I have to admit that gtkwave is much weaker than Verdi.

alonamid, 2020-08-17:
I believe most of the devs use traditional "old school" Linux-based development tools. For example, if I remember correctly, both @zhemao and @jerryz123 are emacs users, while @abejgonzalez , @colinschmidt and @davidbiancolin tend to use vim with some plugins like ctags. Some developers like @hngenc use IntelliJ as an IDE when working with Chisel, but most of the interaction with Chipyard is through the Linux command line and scripts. 

Regarding Linux distributions, I believe that's also a mix - I've seen @zhemao use Arch Linux, @jerryz123 use Ubuntu, and others use macs and VMs.

As you might see, the development environments are quite heterogeneous. I do think that while it might take some time to set up a development environment, it's a one time cost which enables you to set up an environment that you are comfortable working in.

faddat, 2020-08-18:
@alonamid thank you!

That was exactly the kind of information that I was looking for.  

For me setting up the development environment is really not a big deal because I've been playing with RISC-V and open hardware stuff for about five or six months now.

Our goal with https://gitlab.com/virgohardware/core is to make it so that people can learn how this stuff works with just a web browser.  They can use the GitLab Web IDE to make changes, and builds/tests will happen using GitLab CI.  

Once they've gotten hungry for more, what I'd like is for them to be able to download the exact environment used in CI and run it on their local mac/windows/linux machine, and when long builds are needed, they can push commits to a branch in that `core` repository so that they can run with acceleration like `ccache` or `distcc`.  If they are working from an existing core, they will also inherit a test suite, ready to go:

https://gitlab.com/virgohardware/core/-/pipelines/179221213/dag

Because GitLab is open source, the entire thing is forkable and people can do as they please with it.

alonamid, 2020-10-19:
Closing the issue due to lack of activity. Feel free to re-open if necessary

closed 2020-10-19 by alonamid