ucb-bar/chipyard #646: How does riscv-boom RTL simulation loads the compiled ELF file into the RAM?

hyf6661669, 2020-08-14:
I'm trying to understand the RTL simulation flow of riscv-boom. I find that you use ```SimDRAM``` in the TestHarness, and use SimSerial to load the ELF file at the start of the simulation. Since I find SimSerial would write the instructions and .rodata into some places.

The dumped wave shows that SimDRAM was not modified (axi_w_valid amd axi_aw_valid is always 0). After the core exits from the bootrom and starts to execute the ELF file, the L1 ICache would try to read instructions from somewhere. But the data is not from SimDRAM, because axi_r_bits_data is always 0. Could please tell me where does the data from SimSerial go ? And where does the L1 ICache get data from?

jerryz123, 2020-08-14:
[image]

The SerialAdapter plugs into the FrontBus. Thus, writes from the SerialAdapter allocate blocks in the  L2 cache. For small programs such as riscv-tests which fit entirely in L2, there may be no DRAM activity.

hyf6661669, 2020-08-14:
@jerryz123 Thanks for you explanation, actually I'm running dhrystone in riscv-boom so the elf file is small enough to stay in the L2 cache.
Two more questions here:
1. Is there a method to load the whole elf file into SimDRAM without occupying simulation time? Just like $readmemh() in verilog. Since I want to run many benchmarks for many times in riscv-boom, the load operation would be time-consuming. 

2. How to make verilator starts to dump waves after the elf file is loaded into the SimDRAM/L2 Cache ? At present I only find that  I can use ```-x/--dump-start``` option to tell the simulator when to start dumping waves. But for different benchmarks the code sizes are different and cycles for load operations are also different.

hyf6661669, 2020-08-14:
OK, the answer is in the document ....

closed 2020-08-14 by hyf6661669

Sai-Manish, 2024-05-28:
Hi @hyf6661669, where is document did you find the way to load the whole elf file into SimDRAM?

> @jerryz123 Thanks for you explanation, actually I'm running dhrystone in riscv-boom so the elf file is small enough to stay in the L2 cache. Two more questions here:
>
> 1. Is there a method to load [...]

jerryz123, 2024-05-28:
Chipyard docs around the LOADMEM flag