ucb-bar/chipyard #651: I met a problem when use VCS to do Gate Level Simulation

lihao1915, 2020-08-19:
context: I want to use VCS to do a Gate Level Simulation on chipyard; so I replace the top file"freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig.top.v
" with netlist file. And I use the system task"$sdf_annotate" in "freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig.harness.v".But when I "make" in "vcs" file , the error appeared:

`/home/lh/synopsys/anzhuang/vcs/bin/vcs: line 8256: 73552 Killed                  /home/lh/synopsys/anzhuang/vcs/linux64/bin/vcs1 -Mcc=gcc -Mcplusplus=g++ -Masflags= -Mcfl=' -pipe -fPIC -I/home/lh/RV-installation/rocket-tool/include -I/home/lh/ai-chip/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/lh/synopsys/anzhuang/vcs/include ' -Mxcflags=' -pipe -fPIC -I/home/lh/synopsys/anzhuang/vcs/include' -Mldflags=" -Wl,--no-as-needed -rdynamic -Wl,-E " -Mout=/home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig -Mamsrun="" -Mvcsaceobjs="" -Mobjects=" /home/lh/synopsys/anzhuang/vcs/linux64/lib/libvirsim.so /home/lh/synopsys/anzhuang/vcs/linux64/lib/liberrorinf.so /home/lh/synopsys/anzhuang/vcs/linux64/lib/libsnpsmalloc.so /home/lh/synopsys/anzhuang/verdi/share/PLI/VCS/LINUXAMD64/pli.a /home/lh/ai-chip/chipyard/tools/DRAMSim2/libdramsim.a /home/lh/RV-installation/rocket-tool/lib/libfesvr.a " -Mexternalobj="" -Msaverestoreobj=/home/lh/synopsys/anzhuang/vcs/linux64/lib/vcs_save_restore_new.o -Mcrt0="" -Mcrtn="" -Mcsrc="" -Msyslibs="-ldl -lm " -full64 -Xnotice=1 -line -P /home/lh/synopsys/anzhuang/verdi/share/PLI/VCS/LINUXAMD64/novas.tab +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -error=noZMMCM -timescale=1ns/10ps -q -q +v2k +vcs+lic+wait +notimingchecks +nospecify +vc+list +systemverilogext+.sv+.svi+.svh+.svt -assert svaext +libext+.sv +v2k +verilog2001ext+.v95+.vt+.vp +libext+.v -Xcbug=0x1 -Xpiyushb1=0x80 +libext+.v +define+VCS +define+CLOCK_PERIOD=1.0 +define+MEM_INIT_ZERO +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN -o /home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig -Xcbug=0x1 -Xpiyushb1=0x80 -picarchive +vcsd +itf+/home/lh/synopsys/anzhuang/vcs/linux64/lib/vcsdp.tab +cli+1 -debug=4 +memcbk +rad -sverilog +vpi -gen_obj -f /home/lh/ai-chip/chipyard/sims/vcs/generated-src/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig/sim_files.common.f +incdir+/home/lh/ai-chip/chipyard/sims/vcs/generated-src/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig "/home/lh/netlist/ExampleRocketSystem_sta.v" "/home/lh/ai-chip/chipyard/sims/vcs/generated-src/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig.harness.v" "/home/lh/ai-chip/chipyard/sims/vcs/mem-lib/SZAA55_2048X8X2CM8b1.v" "/home/lh/ai-chip/chipyard/sims/vcs/mem-lib/SZAA55_2048X8X4CM8b1.v" "/home/lh/ai-chip/chipyard/sims/vcs/generated-src/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig.harness.mems.v" -v /home/lh/netlist/u055lscspmvbbh_sdf30.v -v /home/lh/netlist/u055lscspmvbbl_sdf30.v -v /home/lh/netlist/u055lscspmvbbr_sdf30.v
Makefile:50: recipe for target '/home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig' failed
make: *** [/home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig] Error 137
`

colinschmidt, 2020-08-19:
There isn't enough information here. Is there another error message? What technology are you using to synthesize the design? You could consider using Hammer in the vlsi directory to run synthesis and then the gate level simulation.

lihao1915, 2020-08-19:
> There isn't enough information here. Is there another error message? What technology are you using to synthesize the design? You could consider using Hammer in the vlsi directory to run synthesis and then the gate level simulation.

Thanks for your reply!
I used DC to get the netlist file. 
And after getting the file,I changed the variable "sim_vsrcs" in `/home/lh/chipyard/variable.mk`, with replacing the `$(TOP_FILE)` by using the netlist file, and removing `$(TOP_SMEMS_FILE)`.
Besides, I added the process library file to the `VCS_NONCC_OPTS` in `/home/lh/chipyard/vcs.mk`.
After, I run "make" in `/home/lh/chipyard/sims/vcs/`,when the verilog sources are compiled, I use `ctrl z` to terminate the program.And I use `$sdf_annotate("sdf file name",testDriver.testHarness.ldut);` to annotate in file `TestDriver.v`.
Finally , I run "make" again in  `/home/lh/chipyard/sims/vcs/`, and it faited.
Because  I don't quite understand how to do Gate-level simulation. So if you understand it ,can you teach me?
I have the netlist file ,sdf file and process library file. Thank you very much!

The following is the error message:
```
Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 35042
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 35097
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 35152
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 35207
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 35262
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 35475
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 36215
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 36281
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 36347
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 36413
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 36564
  Following variable has never been set any value.
  Source info: notifier

Lint-[VNGS] Variable never gets set
/home/lh/netlist/u055lscspmvbbr_sdf30.v, 36649
[... 125 lines cut ...]
  Port "auto_axi4yank_out_aw_bits_size" declared as output in module 
  "SimpleLazyModule_11" may need to be inout. Coercing to inout.

Lint-[PCTIO-L] Ports coerced to inout
/home/lh/netlist/ExampleRocketSystem_sta.v, 126183
"auto_axi4yank_out_ar_bits_size"
  Port "auto_axi4yank_out_ar_bits_size" declared as output in module 
  "SimpleLazyModule_11" may need to be inout. Coercing to inout.

Lint-[PCTIO-L] Ports coerced to inout
/home/lh/netlist/ExampleRocketSystem_sta.v, 123551
"auto_out_aw_bits_size"
  Port "auto_out_aw_bits_size" declared as output in module "TLToAXI4" may 
  need to be inout. Coercing to inout.

Lint-[PCTIO-L] Ports coerced to inout
/home/lh/netlist/ExampleRocketSystem_sta.v, 123575
"auto_out_ar_bits_size"
  Port "auto_out_ar_bits_size" declared as output in module "TLToAXI4" may 
  need to be inout. Coercing to inout.

/home/lh/synopsys/anzhuang/vcs/bin/vcs: line 8256: 79247 Killed                  /home/lh/synopsys/anzhuang/vcs/linux64/bin/vcs1 -Mcc=gcc -Mcplusplus=g++ -Masflags= -Mcfl=' -pipe -fPIC -I/home/lh/RV-installation/rocket-tool/include -I/home/lh/ai-chip/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/lh/synopsys/anzhuang/vcs/include ' -Mxcflags=' -pipe -fPIC -I/home/lh/synopsys/anzhuang/vcs/include' -Mldflags=" -Wl,--no-as-needed -rdynamic -Wl,-E " -Mout=/home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig -Mamsrun="" -Mvcsaceobjs="" -Mobjects=" /home/lh/synopsys/anzhuang/vcs/linux64/lib/libvirsim.so /home/lh/synopsys/anzhuang/vcs/linux64/lib/liberrorinf.so /home/lh/synopsys/anzhuang/vcs/linux64/lib/libsnpsmalloc.so /home/lh/synopsys/anzhuang/verdi/share/PLI/VCS/LINUXAMD64/pli.a /home/lh/ai-chip/chipyard/tools/DRAMSim2/libdramsim.a /home/lh/RV-installation/rocket-tool/lib/libfesvr.a " -Mexternalobj="" -Msaverestoreobj=/home/lh/synopsys/anzhuang/vcs/linux64/lib/vcs_save_restore_new.o -Mcrt0="" -Mcrtn="" -Mcsrc="" -Msyslibs="-ldl -lm " -full64 -Xnotice=1 -line -P /home/lh/synopsys/anzhuang/verdi/share/PLI/VCS/LINUXAMD64/novas.tab +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -error=noZMMCM -timescale=1ns/10ps -q -q +v2k +vcs+lic+wait +notimingchecks +nospecify +vc+list +systemverilogext+.sv+.svi+.svh+.svt -assert svaext +libext+.sv +v2k +verilog2001ext+.v95+.vt+.vp +libext+.v -Xcbug=0x1 -Xpiyushb1=0x80 +libext+.v +define+VCS +define+CLOCK_PERIOD=1.0 +define+MEM_INIT_ZERO +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT  [... cut ...] m.TestHarness.GemminiAcceleratorConfig/freechips.rocketchip.system.TestHarness.GemminiAcceleratorConfig.harness.mems.v" -v /home/lh/netlist/u055lscspmvbbh_sdf30.v -v /home/lh/netlist/u055lscspmvbbl_sdf30.v -v /home/lh/netlist/u055lscspmvbbr_sdf30.v
Makefile:50: recipe for target '/home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig' failed
make: *** [/home/lh/ai-chip/chipyard/sims/vcs/simv-freechips.rocketchip.system-GemminiAcceleratorConfig] Error 137
```

jerryz123, 2020-08-19:
The Killed sometimes indicates out of memory error.

lihao1915, 2020-08-20:
> The Killed sometimes indicates out of memory error.

Thank you very much!This problem was solved after I added memory. And it appeared some new errors,I think this is due to my lack of knowledge of the Gate-level simulation

closed 2020-08-25 by alonamid