ucb-bar/chipyard #660: What Boards are you using when working on Chipyard?

0ena, 2020-08-25:
Hi,

 I'm interested in using the Chipyard Framework to create a SoC that will be able to run Linux on it, based on the Rocket-Chip. I have recently started looking at RISC-V architectures so I am an amateur on this.

Since I am planning to buy an FPGA capable of supporting such project, I was wondering what Boards people working with Chipyard are using, in order to build on their implementations.
Do you have any Boards to suggest?

I think this thread will be really helpful for users that are new to the RISC-V family such as my self.
Thank you in advance for your responses.

Kind regards,
Nassos

faddat, 2020-08-25:
Hi 0ena, 

Around February I said out to do the same thing.  I have ordered a Genesys2 and an Arty T7100, but they won't arrive till September.  

But I am not sure if those support the rocket-chip or not.  

As for creating an SOC in Chipyard, it's a **dream**!  Here is an example:

https://github.com/faddat/chipyard/commit/a512b3b75e34c9f471c5131130ae208b497f434a

I was just playing around and so I defined the most complex and silly thing possible with every possible feature.  Nonetheless, it worked great!

If you are interested in learning more about the build process, I'd suggest that you check out https://gitlab.com/tendermint-hardware/core.  Over there I have an automatic build system for many types of risk five CPU cores. I also have Docker images for most common RISCV core build environments. My repository is pretty much aimed at amateurs, and at using entry into RISCV. It was quite challenging when I started down this path.  

Do you know if you need a hand with anything or want to chat.

0ena, 2020-08-26:
Hi @faddat ,

thank you very much for your reply.

I am leaning also towards getting a Genesys 2 board. I believe it has enough resources to handle even the SiFive U500 implementation of Rocket-Chip (it includes 4 Tiles (cores) and takes 146573 LUTS on a VC707 Virtex-board, the Kintex of Genesys 2 has 203800 LUTs in total). But I would like to know what others may have used to implement a  RISC-V SoC capable of running Linux.
So from your message I conclude that you haven't downloaded yet your implementation on an FPGA, am I right?

Thank you very much for your link and your attempt to streamline the process of building a RISC-V core. I will definitely look at it. I have a Verilog/VHDL background and I am completely new to Chisel/Chisel generators. My intention is not only to build a working SoC, but to be able to understand the generated Verilog (?) or the initial Chisel and be able to modify it.

My mail is on my git-profile and I am always open to discuss.

For anyone else, please feel free to share your experience with boards that were used to implement a RISC-V core capable of running Linux.
I will make updates on this also once I get my board and download my implementation.

alonamid, 2020-08-26:
The FPGA aspects of Chipyard have so far been focused on emulation/simulation using FireSim (i.e. including timing-accurate IO and peripheral modeling), as opposed to FPGA prototyping (synthesizing the RTL directly to an FPGA board). In that sense, we have been using Amazon's AWS F1 instances for FPGAs.

There is some current ongoing work to have a built-in FPGA prototyping flow within Chipyard, likely targeting boards such as the VCU118 or Zynq as the first stages. We do not have an estimated timeline for that yet.

I would recommend having a look at the SiFive [Freedom Platform](https://github.com/sifive/freedom) and [FPGA-shells](https://github.com/sifive/fpga-shells) for a list of boards you could probably try to get working with some hacking.

0ena, 2020-08-26:
Hi @alonamid ,

thank you very much for your informative reply.
I have a question regarding the boards you say you are targeting as a team, if you can answer it.

Is there a specific reason for choosing VCU118 board? It is a massive board in terms of resources (and it comes with high cost).

I have spent the last days checking the Freedom Project that you mentioned, they have indeed done a good job in terms of implementing a RISC-V on an FPGA board, using Rocket-Chip. They are also targeting on their part a Virtex board, namely VC707 which also has more than enough resources for their implementation.

alonamid, 2020-08-26:
The reason for these particular boards are a mix of availability (that's what we have) and requirements.
We don't use FPGA boards just for prototyping, but also as a platform to interface with fabricated test chips and boards. Therefore, they need to have certain interfaces and components on the boards that not all simple boards have.

closed 2020-08-27 by alonamid