ucb-bar/chipyard #665: How to add a apb port in ChipTop?

bg193, 2020-08-30:
I want to add apb master port in ChipTop like AXI4 DRAM controller port , how to do it?

jerryz123, 2020-08-30:
The error indicates that you have undriven IOs in your digital system. See the WithBlackBoxSimMem IOBinder in chipyard/generators/src/main/scala/IOBinders.scala for an example of how the axi4 IO is passed into the TestHarness and connected to a simulated DRAM model.

bg193, 2020-08-30:
Have add WithSimAPBMMIO :
```
class WithSimAPBMMIO extends OverrideIOBinder({
   (system: CanHaveMasterAPBMMIOPort with BaseSubsystem) => {
      SimAPBMem.connectMMIO(system)(system.p);
      Nil
   }
})
```
 and simulated memory model:
```
class SimAPBMem(edge: APBEdgeParameters, size: BigInt)(implicit p: Parameters) extends SimpleLazyModule {
  val node = APBMasterNode(List(edge.master))
  val srams = AddressSet.misaligned(0, size).map { aSet =>
    LazyModule(new APBRAM(
      address = aSet,
      beatBytes = edge.bundle.dataBits/8))
  }
  val xbar = new APBFanout()
  srams.foreach{ s => s.node := xbar.node }
  xbar.node := node
  val io_apb = InModuleBody { node.makeIOs() }
}

object SimAPBMem {
  def connectMMIO(dut: CanHaveMasterAPBMMIOPort)(implicit p: Parameters): Seq[SimAPBMem] = {
    dut.mmio_apb.zip(dut.mmioAPBNode.in).map { case (io, (_, edge)) =>
      // test harness size capped to 4KB (ignoring p(ExtMem).get.master.size)
      println(io.toNamed)
      val mmio_mem = LazyModule(new SimAPBMem(edge, size = 4096))
      Module(mmio_mem.module).suggestName("mmio_apb")
      mmio_mem.io_apb.head <> io
      mmio_mem
    }
  }
}

```

Still meet error :
[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[ChipTop.scala 71:68:chipyard.TestHarness.RocketConfig.fir@277156.4] : [module ChipTop]  Reference system is not fully initialized.
[error]    : system.mmio_apb.0.prdata <= VOID
[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[ChipTop.scala 71:68:chipyard.TestHarness.RocketConfig.fir@277156.4] : [module ChipTop]  Reference system is not fully initialized.
[error]    : system.mmio_apb.0.pready <= VOID
[error] firrtl.passes.CheckInitialization$RefNotInitializedException:  @[ChipTop.scala 71:68:chipyard.TestHarness.RocketConfig.fir@277156.4] : [module ChipTop]  Reference system is not fully initialized.
[error]    : system.mmio_apb.0.pslverr <= VOID
[error] firrtl.passes.PassException: 3 errors detected!

jerryz123, 2020-08-30:
This line 
```CanHaveMasterAPBMMIOPort with BaseSubsystem```
should be just
```CanHaveMasterAPBMMIOPort```

Sorry, the IOBinders system does not support compound types, this bug may be present in Chipyard 1.3.0, but it will be fixed in the next release.

bg193, 2020-08-31:
Revise as you said, but still meet that error.

closed 2020-08-31 by bg193

AD738560581, 2025-01-10:
Hello， did you have added APB successful？ I also want to added APB in the chipyard