ucb-bar/chipyard #668: Reduce the pbus frequency

bg193, 2020-09-03:
In chipyard , pbus work at same frequency with core clock, how to reduce the pbus frequency to 100Mhz?

jerryz123, 2020-09-03:
You mean simulate a multiclock system? There is active work in progress towards that goal.

bg193, 2020-09-03:
Similar to this goal, can you give me some Suggestions on how to modify it?

colinschmidt, 2020-09-04:
It is going to be a lot of modifications in the current state of the repo to get this to work. You would need to insert a clock divider somewhere using its output to drive the pbus and components its attached to, and then ensure that there is either a rational or async crossing between the pbus and whichever bus in your system is driving the pbus.

It is much simpler to wait for the feature to be supported in a future chipyard release, if you have the time. This configuration is certainly one we would like to support.

bg193, 2020-09-05:
@colinschmidt Thank you very much for your valuable advice

closed 2020-09-05 by bg193