ucb-bar/chipyard #687: Using DRAMSIM2 to test memory access latency inchipyard/sims/verilator

Zhouwk7, 2020-10-07:
I have used two ways to access 1024 consecutive integer data by client node attaching to systembus, previous version chipyard using SIMAXIMEM and current version chipyard having DRAMSIM2. But I found they have similar memory access about 3000 cycles for 1024 consecutive integer data access !!! As far as I know, SIMAXIMEM memory access request is 1 cycle because of implemented using SRAM. However, DRAMSIM2 is a cycle accurate DRAM simulator, its memory access latency should be similar to the actual situation. Do you know what the problem is? And how can I using DRAMSIM2 to test memory access? Here are some code fragments:
The node I have attached to bus:

class RoCCDMAStride(opcode: OpcodeSet, val regBufferNum: Int = 1024)(implicit p: Parameters) extends LazyRoCC(opcode){

    val dmaController = LazyModule(new DMAController)
override val tlNode = dmaController.node
…
}

The benchmark I used to record clock cycle 、node configure and 1024 consecutive integer data access:

     start = rdcycle();
    asm volatile("fence");
    ROCC_INSTRUCTION_SS(0,a,config,0);
    asm volatile("fence":::"memory");
    end = rdcycle();
    printf("Read %d longlong num cost: %lld cycles\n", size, end-start);

My verilator simulation command

./simulator-chipyard-RoCCDMAStrideRocketConfig  -dramsim ./tests/DMAStrideTest/main.riscv 2>output/RoCCDMAStride

colinschmidt, 2020-10-07:
You need to use `+dramsim` as defined in the makefiles:
https://github.com/ucb-bar/chipyard/blob/19152d3b73f1c2255ce378501f6c01f9363b75e9/variables.mk#L142

jerryz123, 2020-10-07:
The systembus sits in front of the L2, so you may be measuring L2 access latency instead.

colinschmidt, 2020-10-07:
In the past I've used https://github.com/ucb-bar/ccbench to measure cache size and latency based on software loads and stores.

Zhouwk7, 2020-10-08:
> In the past I've used https://github.com/ucb-bar/ccbench to measure cache size and latency based on software loads and stores.

Thanks, it's a very good resource for measure multi-code cache size and latency，and the ccbench also uses DRAMSIM2 as DRAM simulation. This means DRAMSIM2 can work well  and the rocket-chip DRAM latency is about 70 cycles.

Zhouwk7, 2020-10-08:
> The systembus sits in front of the L2, so you may be measuring L2 access latency instead.

Maybe you are right. When the rocket-chip is configureed to "new freechips.rocketchip.subsystem.WithInclusiveCache ++"，the cycle is about 3000 cycles，but nearly 40000 cycles without “new freechips.rocketchip.subsystem.WithInclusiveCache ++". 
However, something went worry when I run the emulator with “new freechips.rocketchip.subsystem.WithInclusiveCache ++". Here is the debug output during the simulation.
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 35241
== Loading device model file 'dramsim2_ini/DDR3_micron_64M_8B_x4_sg15.ini' == 
== Loading system model file 'dramsim2_ini/system.ini' == 
===== MemorySystem 0 =====
CH. 0 TOTAL_STORAGE : 4096MB | 1 Ranks | 16 Devices per rank
DRAMSim2 Clock Frequency =1Hz, CPU Clock Frequency=1Hz
*** FAILED *** (tohost = 1337)
*** FAILED *** (code = 1337, seed 1602116184) after 261972 cycles`

my command for testing the emulator

`./simulator-chipyard-RoCCDMAStrideRocketConfig  +dramsim +max-cycles=10000000 ./tests/DMAStrideTest/main.riscv 2>output/RoCCDMAStride`

By the way, do you know how can I do to access DRAM memory directly avoiding flushing L2$.

Zhouwk7, 2020-10-08:
> You need to use `+dramsim` as defined in the makefiles:
> https://github.com/ucb-bar/chipyard/blob/19152d3b73f1c2255ce378501f6c01f9363b75e9/variables.mk#L142

jerryz123, 2020-12-07:
Are you able to provide more information on the issue? That failure could be caused by many things.

If you access DRAM without flushing L2, your view of memory will be incoherent.

closed 2021-01-30 by alonamid