ucb-bar/chipyard #691: make failed when using VCS 2016

JunningWu, 2020-10-12:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* branch master, Hash: 19152d3
*OS:* Linux chipyard-vm 5.4.0-49-generic #53~18.04.1-Ubuntu SMP Mon Sep 21 14:12:39 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
*Other:* Vmware

**What is the current behavior?**

I can simulate with verilator, when using VCS, after runing **make or make debug**
```
recompiling module data_arrays_0_0
recompiling module l2_tlb_ram
recompiling module Queue_36_inTestHarness
recompiling module UARTAdapter_inTestHarness
recompiling module cc_dir_ext
recompiling module split_cc_dir_ext
recompiling module split_data_arrays_0_ext
recompiling module split_tag_array_ext
recompiling module split_tag_array_0_ext
recompiling module split_data_arrays_0_0_ext
All of 191 modules done
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/mm.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/mm_dramsim2.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/remote_bitbang.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimDRAM.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimDTM.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimJTAG.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimSerial.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimUART.cc
g++ -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -std=c++11 -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/uart.cc
cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C
/usr/bin/ld: /home/chipyard/software/synopsys/vcs2016/linux64/lib/vcs_save_restore_new.o: relocation R_X86_64_32S against undefined symbol `_sigintr' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
Makefile:104: recipe for target 'product_timestamp' failed
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
Make exited with status 2
Makefile:50: recipe for target '/home/chipyard/work/chipyard/chipyard/sims/vcs/simv-chipyard-RocketConfig' failed
make: *** [/home/chipyard/work/chipyard/chipyard/sims/vcs/simv-chipyard-RocketConfig] Error 2
```

**What is the expected behavior?**

**Other information**
i guess something related to fPIC or PIE, where to disable Flags?

colinschmidt, 2020-10-12:
This seems like an issue with your C compiler. What version of `g++` are you using?

JunningWu, 2020-10-13:
```
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

abejgonzalez, 2020-10-13:
This looks related: https://github.com/chipsalliance/rocket-chip/issues/1377. If you want to add/change flags you can refer to https://github.com/ucb-bar/chipyard/blob/19152d3b73f1c2255ce378501f6c01f9363b75e9/vcs.mk#L6-L10

JunningWu, 2020-10-13:
@abejgonzalez is it ok when changing g++ version to 4.8 according to [Error while run "make -j4 run" in /vsim](https://github.com/chipsalliance/rocket-chip/issues/1377) ?

JunningWu, 2020-10-15:
when changed gcc & g++ from 7.5 to 4.8, I could not build verilator, 
```
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.debug'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -include /home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.plusArgs -include /home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/verilator.h -include /home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.debug/VTestHarness.h   -c -o SimDRAM.o /home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/SimDRAM.cc
g++: error: unrecognized command line option ‘-faligned-new’
VTestHarness.mk:74: recipe for target 'SimDRAM.o' failed
make[1]: *** [SimDRAM.o] Error 1
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.debug'
Makefile:138: recipe for target '/home/chipyard/work/chipyard/chipyard/sims/verilator/simulator-chipyard-RocketConfig-debug' failed
make: *** [/home/chipyard/work/chipyard/chipyard/sims/verilator/simulator-chipyard-RocketConfig-debug] Error 2
```

abejgonzalez, 2020-10-15:
I would 1st try to get VCS working... I think the Verilator issue is different (I think that should work with gcc 7.5). The VCS issue seems that you can't find the DRAM sim header(s) (`mm_dramsim2.h`, `mm.h`) that is found in the `tools/DRAMSim2` submodule. Do you have that submodule checked out? If so, then I would try to see if the header is included in the VCS compilation. It should be found in the `generated-src` for your project.

JunningWu, 2020-10-16:
VCS is working well. 
And I added sims/vcs/generated-src/chipyard.TestHarness.RocketConfig directory to VCS, like this, which includes headers mm_dramsim2.h & mm.h 
```
VCS_CC_OPTS = \
	-CC "-I$(RISCV)/include" \
	-CC "-I$(dramsim_dir)" \
        -CC "-I$(build_dir)" \
	-CC "-std=c++11" \
	-CC "$(EXTRA_SIM_CC_FLAGS)" \
        -cpp g++-4.8 -cc gcc-4.8 \
        -CC "-LDFLAGS -Wl,-no-as-needed"
```
VCS compilation still failed with the same **undefined reference** error.

when I check the declaration and usage for **getMemorySystemInstance**, I wonder whether **"results"** matches **unsigned megsOfMemory**？

```
//////vcs/generated-src/chipyard.TestHarness.RocketConfig/mm_dramsim2.cc, line 52
mem = getMemorySystemInstance(memory_ini, system_ini, ini_dir, "results", size/(1024*1024));
```

```
///////DRAMSim.h, line 64
MultiChannelMemorySystem *getMemorySystemInstance(const string &dev, const string &sys, const string &pwd, const string &trc, unsigned megsOfMemory, std::string *visfilename=NULL);
```

colinschmidt, 2020-10-16:
Do you have access to a newer version of VCS? If you can get a 2018+ version of VCS you should be able to use a modern GCC.

abejgonzalez, 2020-10-16:
Can you include the entire log from when VCS is invoked? 

While this is highly unlikely IMO, maybe the `-CC "-I` part is not working... since that should include the proper folder that holds the header files. I would try to `man` or find the guide for your VCS and double check that the `-I` works for your VCS version... 

You can also try to manually run the VCS command and find another way to include the build folder with the headers.

JunningWu, 2020-10-17:
According to vcs ug, ``` -CC "-I ``` works well for VCS 2016.

```
-CC <options>
   Works the same as -CFLAGS.
-CFLAGS <options>
   Pass options to C compiler. Multiple -CFLAGS are allowed. Allows
   passing of C compiler optimization levels.
```

The entire log after VCS command:
```
rm -rf csrc && vcs -full64 -notice -line -CC "-I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include" -CC "-I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2" -CC "-I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig" -CC "-std=c++11" -CC "" -cpp g++-4.8 -cc gcc-4.8 -CC "-LDFLAGS -Wl,-no-as-needed" /home/chipyard/work/chipyard/chipyard/tools/DRAMSim2/libdramsim.a /home/chipyard/work/chipyard/chipyard/riscv-tools-install/lib/libfesvr.a +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -error=noZMMCM -timescale=1ns/10ps -quiet -q +rad +v2k +vcs+lic+wait +vc+list -f /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/sim_files.common.f -sverilog +systemverilogext+.sv+.svi+.svh+.svt -assert svaext +libext+.sv +v2k +verilog2001ext+.v95+.vt+.vp +libext+.v -debug_pp +incdir+/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.v /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.v /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.mems.v /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.mems.v +libext+.v +define+VCS +define+CLOCK_PERIOD=1.0 +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN  -o /home/chipyard/work/chipyard/chipyard/sims/vcs/simv-chipyard-RocketConfig-debug \
+define+DEBUG \
-debug_pp

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

Lint-[NS] Null statement
/home/chipyard/software/synopsys/vcs2016/etc/sva/rec_ltl_classes_package.svp, 2

[... 536 lines cut ...]
recompiling module split_data_arrays_0_0_ext
All of 192 modules done
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/mm.cc
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/mm_dramsim2.cc
[... cut ...]
mm_dramsim2.cc:(.text+0xcef): undefined reference to `DRAMSim::getMemorySystemInstance(std::string const&, std::string const&, std::string const&, std::string const&, unsigned int, std::string*)'
collect2: error: ld returned 1 exit status
Makefile:104: recipe for target 'product_timestamp' failed
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
Make exited with status 2
Makefile:54: recipe for target '/home/chipyard/work/chipyard/chipyard/sims/vcs/simv-chipyard-RocketConfig-debug' failed
make: *** [/home/chipyard/work/chipyard/chipyard/sims/vcs/simv-chipyard-RocketConfig-debug] Error 2
```

JunningWu, 2020-10-17:
[run.log](https://github.com/ucb-bar/chipyard/files/5394874/run.log)

JunningWu, 2020-10-17:
> Do you have access to a newer version of VCS? If you can get a 2018+ version of VCS you should be able to use a modern GCC.

I am not sure whether this error is related to older version gcc&g++?

JunningWu, 2020-10-17:
I can compile when I comment ``` mm.cc, mm_dramsim2.cc, SimDRAM.cc ``` ,
```
recompiling module split_data_arrays_0_ext
recompiling module split_tag_array_ext
recompiling module split_tag_array_0_ext
recompiling module split_data_arrays_0_0_ext
All of 192 modules done
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
make[1]: Entering directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/remote_bitbang.cc
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimDTM.cc
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimJTAG.cc
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimSerial.cc
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/SimUART.cc
g++-4.8 -w  -pipe -fPIC -I/home/chipyard/work/chipyard/chipyard/riscv-tools-install/include -I/home/chipyard/work/chipyard/chipyard/tools/DRAMSim2 -I/home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig -std=c++11 -LDFLAGS -Wl,-no-as-needed -O -I/home/chipyard/software/synopsys/vcs2016/include    -c /home/chipyard/work/chipyard/chipyard/sims/vcs/generated-src/chipyard.TestHarness.RocketConfig/uart.cc
cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C [enabled by default]
/home/chipyard/work/chipyard/chipyard/sims/vcs/simv-chipyard-RocketConfig-debug up to date
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/sims/vcs/csrc'
```
and get the executable file ``` simv-chipyard-RocketConfig-debug ```

JunningWu, 2020-11-10:
Still unable to solve this problem.

abejgonzalez, 2020-11-10:
Frankly, I'm not exactly sure what the issue is. The next thing I would check is that you have the `DRAMSim2.h` header pointed to with some `-I` and the `libdramsim2.a` library built and pointed to with `-L`.

JunningWu, 2020-12-03:
After changing to a higher version VCS(2018), I can compile and run the simulation.
So I guess the Chipyard project needs Ubunut18.04 & VCS2018, basically.

closed 2020-12-03 by JunningWu