ucb-bar/chipyard #694: Bump Chisel3 to 3.4.0 (and FIRRTL to 1.4)

timsnyder-siv, 2020-10-15:
**Impact**: unknown
Not sure what you guys would consider the impact.

**Description**

Chisel 3.4 was relased 10 days ago https://github.com/freechipsproject/chisel3/releases/tag/v3.4.0 and has made it into Rocket-Chip master.  At some point, someone else will want to pick it up and it looks like I might be the first one trying to.  Starting this ticket as a place to discuss

**What is a motivating example for changing the behavior?**
I need to do this for my organization's internal reasons.  I don't see an existing ticket or PR for this work.  I'm going to get the portions of Chipyard/FireSim I use working with it and can share at least those changes with the larger community.

chick, 2020-10-16:
I'm interested in in working on this as part of internalizing the chipyard system. I'm coming up for air after chisel release and would like to devote more time to this. I have a [barstools pr](https://github.com/ucb-bar/barstools/pull/87) that ports it to 3.4
So just a small start.

timsnyder-siv, 2020-10-19:
Thanks for the pointer @chick .  I have a branch of FireSim & Chipyard that builds `FireSimRocketConfig` and passing the [MIDAS-level run-asm-tests](https://docs.fires.im/en/latest/Advanced-Usage/Debugging-in-Software/RTL-Simulation.html#midas-level-simulation).  Note, I am focusing on getting the internals of FireSim updated for use with my deisgns.  In my branch, I stripped out many unused portions of ChipYard so that I could get FireSim working with only RocketChip.

I currently have AutoCounters disabled by commenting out [the registration of the FireSimPropertyLibrary](https://github.com/ucb-bar/chipyard/blob/46ea9005389019b158f07d21dceb4038f7b2ca1b/generators/firechip/src/main/scala/FireSim.scala#L142) otherwise, I hit 
```
Exception in thread "main" java.lang.AssertionError: assertion failed: ~F1Shim|FireSim/lazyModule_:ChipTop/system:DigitalTop/subsystem_fbus:FrontBus/coupler_from_port_named_blkdevcontroller:TLInterconnectCoupler
_6/fixer:TLFIFOFixer_2>_T_4 should be renamed exactly once. Suggested renames: List()
``` 

I'll do some more testing of my branch and push what I have in the next day or two.

alonamid, 2021-01-21:
Done in release 1.4

closed 2021-01-21 by alonamid