ucb-bar/chipyard #697: Make Failed Using VCS E-2011.03

CaWen, 2020-10-16:
I got the similar problem

**Impact:** rtl

**Tell us about your environment:** 
Chipyard Version: 1.3.0
OS: Linux 4.18.0-193.19.1.el8_2.x86_64 (Red Hat)
Other: gcc version 8.3.1 20191121 (Red Hat 8.3.1-5) (GCC) 

**What is the current behavior?**
When I run **make CONFIG=SmallBoomConfig**

```
awk '{print ;}' /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/sim_files.f /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/firrtl_black_box_resource_files.top.f /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/firrtl_black_box_resource_files.harness.f | sort -u | grep -v '.*\.\(svh\|h\)$' > /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/sim_files.common.f
rm -rf csrc && vcs -full64 -notice -line -CC "-I/home/cwu0220/riscvproj/chipyard/riscv-tools-install/include" -CC "-I/home/cwu0220/riscvproj/chipyard/tools/DRAMSim2" -CC "-std=c++11" -CC "" /home/cwu0220/riscvproj/chipyard/tools/DRAMSim2/libdramsim.a /home/cwu0220/riscvproj/chipyard/riscv-tools-install/lib/libfesvr.a +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -error=noZMMCM -timescale=1ns/10ps -quiet -q +rad +v2k +vcs+lic+wait +vc+list -f /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/sim_files.common.f -sverilog +systemverilogext+.sv+.svi+.svh+.svt -assert svaext +libext+.sv +v2k +verilog2001ext+.v95+.vt+.vp +libext+.v -debug_pp +incdir+/home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/chipyard.TestHarness.SmallBoomConfig.top.v /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/chipyard.TestHarness.SmallBoomConfig.harness.v /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/chipyard.TestHarness.SmallBoomConfig.top.mems.v /home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/chipyard.TestHarness.SmallBoomConfig.harness.mems.v +libext+.v +define+VCS +define+CLOCK_PERIOD=1.0 +define+RESET_DELAY=777.7 +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN  -o /home/cwu0220/riscvproj/chipyard/sims/vcs/simv-chipyard-SmallBoomConfig \
-debug_pp

Warning-[UNKWN_OPTVSIM] Unknown option passed
  Ignoring unknown option '-error=PCWM-L' passed to 'vcs' and continuing
  compilation.

Warning-[UNKWN_OPTVSIM] Unknown option passed
  Ignoring unknown option '-error=noZMMCM' passed to 'vcs' and continuing
  compilation.

DirectC Release 1.0 Copyright (c) 1991-2011 by Synopsys Inc.

Error-[SE] Syntax error
  Following verilog source has syntax error :
  "/home/cwu0220/riscvproj/chipyard/sims/vcs/generated-src/chipyard.TestHarness.SmallBoomConfig/mm.cc",
  3: token is '#'
  #include "mm.h"
   ^

1 error
make: *** [Makefile:50: /home/cwu0220/riscvproj/chipyard/sims/vcs/simv-chipyard-SmallBoomConfig] Error 255
```

**What is the expected behavior?**

**Other information**

CaWen, 2020-10-16:
The issue is similar to #691

baichen318, 2020-10-16:
I have solved this problem by modifying the generated VCS simulation command directly. However, I met another problem.
In this VCS version, sim_files.common.f contains some non-verilog files (i.e., *.cc), you need to remove these non-verilog files from sim_files.common.f (just leave verilog files in sim_files.common.f) and append these non-verilog files to the VCS command directly since the flag -f can noly handle verilog files rather than *.cc files.
When I could pass this problem, another new problem appeared. VCS showed that `Error-[SVA-SSTINSC] SVA-system task in non-SVA context`. In fact, I have passed the flag -sverilog to the simulation command but I failed. Currently I do not how to figure it out.

colinschmidt, 2020-10-16:
Do you have access to a newer version of VCS? The issues with `.cc` files in `.f` was fixed in 2014. Then you can follow along with #691 and use a specific version of GCC. If you can get a 2018+ version of VCS you should be able to use a modern GCC.

baichen318, 2020-10-19:
> I have solved this problem by modifying the generated VCS simulation command directly. However, I met another problem.
> In this VCS version, sim_files.common.f contains some non-verilog files (i.e., [...]

Hi, I find that when I use VCS M-2017.03, all problems are gone. So, I suggest you using this version. Thank you.

CaWen, 2020-10-19:
> Hi, I find that when I use VCS M-2017.03, all problems are gone. So, I suggest you using this version. Thank you.

Thank you for your advice. I upgrade my VCS to 2020.03 and also find no bug in this version.

closed 2020-10-19 by CaWen

Shahed-ur-Rahman, 2025-03-12:
Hi, CaWen. Can you please give me the RTL code of smallBoom core.

Shahed-ur-Rahman, 2025-03-12:
onfig   | tee /home/labdoo/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.SmallBoomconfig/chipyard.harness.TestHarness.SmallBoomconfig.chisel.log)
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/labdoo/chipyard/.java_tmp
Exception in thread "main" java.lang.Exception: Unable to find part "chipyard.SmallBoomconfig" from "ArraySeq(chipyard.SmallBoomconfig)", did you misspell it or specify the wrong package path?
	at chipyard.stage.HasChipyardStageUtils.$anonfun$getConfig$1(StageUtils.scala:18)
	at scala.collection.immutable.ArraySeq.foldRight(ArraySeq.scala:237)
	at chipyard.stage.HasChipyardStageUtils.getConfig(StageUtils.scala:13)
	at chipyard.stage.HasChipyardStageUtils.getConfig$(StageUtils.scala:12)
	at chipyard.stage.phases.PreElaboration.getConfig(PreElaboration.scala:17)
	at chipyard.stage.phases.PreElaboration.transform(PreElaboration.scala:28)
	at chipyard.stage.phases.PreElaboration.transform(PreElaboration.scala:17)
	at firrtl.options.DependencyManager.$anonfun$transform$5(DependencyManager.scala:280)
	at firrtl.Utils$.time(Utils.scala:53)
	at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:280)
	at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
	at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
	at scala.collection.immutable.List.foldLeft(List.scala:79)
	at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
	at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
	at firrtl.options.PhaseManager.transform(DependencyManager.scala:443)
	at chipyard.stage.ChipyardStage.run(ChipyardStage.scala:64)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:42)
	at firrtl.options.Stage$$anon$1.transform(Stage.scala:42)
	at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:45)
	at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:183)
	at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:179)
	at scala.collection.immutable.List.foldLeft(List.scala:79)
	at firrtl.options.Stage.$anonfun$transform$2(Stage.scala:45)
	at logger.Logger$.$anonfun$makeScope$4(Logger.scala:148)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:59)
	at logger.Logger$.makeScope(Logger.scala:146)
	at logger.Logger$.makeScope(Logger.scala:125)
	at firrtl.options.Stage.transform(Stage.scala:45)
	at firrtl.options.Stage.execute(Stage.scala:56)
	at firrtl.options.StageMain.main(Stage.scala:69)
	at chipyard.Generator.main(Generator.scala)
Caused by: java.lang.ClassNotFoundException: chipyard.SmallBoomconfig
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:641)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:391)
	at java.base/java.lang.Class.forName(Class.java:382)
	at chipyard.stage.HasChipyardStageUtils.$anonfun$getConfig$1(StageUtils.scala:15)
	... 31 more
make: *** [/home/labdoo/chipyard/common.mk:135: /home/labdoo/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.SmallBoomconfig/chipyard.harness.TestHarness.SmallBoomconfig.fir] Error 1

This is problem i am getting while make command. Can anyone please identify the cause of the problem ?

sdudyl, 2025-04-24:
> onfig | tee /home/labdoo/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.SmallBoomconfig/chipyard.harness.TestHarness.SmallBoomconfig.chisel.log) Picked up JAVA_TOOL_OPTIONS: [...]

Hi, have you solved this problem?