ucb-bar/chipyard #698: Bypassing ChipTop IOs or generating DigitalTop directly

lsteveol, 2020-10-16:
We have our own IOs and other IPs that we want to instantiate in the "ChipTop". We would have our own version of "ChipTop" in this case, which we'll call ``MyRealChipTop``. **Is there a way to create an ``OverrideIOBinder`` which can just punch through all of the ports on ``DigitalTop`` so I can just use the chpyard generated ``ChipTop`` as-is?**  

Currently I'm just building as-is but only instantiating ``DigitalTop`` in my design (ignoring the chipyard generated ``ChipTop``). Our other IPs are verilog based so we would hand instantiate the Chipyard ``DigitalTop`` inside our chiptop along with the other IPs, like below:
[image]

Another option is to just generate the ``DigitalTop`` directly. I have tried to do this by passing the ``TOP=``, ``MODEL=``, and ``VLOG_MODEL=`` however I seem to get some errors for the initial ``chipyard.Generator``. (The errors are different for chipyard v1.3 and the dev/ branch). This may not really be possible, but figured I'd post to see if someone had a solution.

jerryz123, 2020-10-16:
You should be able to generate a verilog file with only DigitalTop if you set `TOP=DigitalTop` only.

abejgonzalez, 2020-10-16:
Additionally, if you want to have an IOBinder that just passes through the IOs, you can create something like this:

```
class WithUARTIOPassthrough extends OverrideIOBinder({
  (system: HasPeripheryUARTModuleImp) => {
    val io_uart_pins_temp = system.uart.zipWithIndex.map { case (dio, i) => IO(dio.cloneType).suggestName(s"uart_$i") }
    (io_uart_pins_temp zip system.uart).map { case (io, sysio) =>
      io <> sysio
    }
    (io_uart_pins_temp, Nil)
  }
})
```

lsteveol, 2020-10-16:
Thank you for both of the quick replies.

@jerryz123 

With commit 19152d3b73f1c2255ce378501f6c01f9363b75e9 I ran the following
``make TOP=DigitalTop -j8``

However I don't see any verilog produced
```
steven@ubuntu:/prj/projects/chipyard/sims/verilator/generated-src/chipyard.DigitalTop.RocketConfig$ ls
chipyard.DigitalTop.RocketConfig.0x0.0.regmap.json         chipyard.DigitalTop.RocketConfig.anno.json         chipyard.DigitalTop.RocketConfig.json         emulator.cc        SimDRAM.cc
chipyard.DigitalTop.RocketConfig.0x0.1.regmap.json         chipyard.DigitalTop.RocketConfig.core.config       chipyard.DigitalTop.RocketConfig.l2.json      mm.cc              SimDTM.cc
chipyard.DigitalTop.RocketConfig.0x2000000.0.regmap.json   chipyard.DigitalTop.RocketConfig.d                 chipyard.DigitalTop.RocketConfig.memmap.json  mm_dramsim2.cc     sim_files.f
chipyard.DigitalTop.RocketConfig.0x2010000.0.regmap.json   chipyard.DigitalTop.RocketConfig.dromajo_params.h  chipyard.DigitalTop.RocketConfig.plusArgs     mm_dramsim2.h      SimJTAG.cc
chipyard.DigitalTop.RocketConfig.0x40.0.regmap.json        chipyard.DigitalTop.RocketConfig.dts               chipyard.DigitalTop.RocketConfig.rom.conf     mm.h               SimSerial.cc
chipyard.DigitalTop.RocketConfig.0x54000000.0.regmap.json  chipyard.DigitalTop.RocketConfig.fir               DigitalTop.anno.json                          remote_bitbang.cc  verilator.h
chipyard.DigitalTop.RocketConfig.0xc000000.0.regmap.json   chipyard.DigitalTop.RocketConfig.graphml           EICG_wrapper.v                                remote_bitbang.h

```

With commit 023d8096a9c8d77a229ca4364ae93fb574581ff8 I see the following:

```
[info] running barstools.tapeout.transforms.GenerateTopAndHarness -o /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.v -tho /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.v -i /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.fir --syn-top DigitalTop --harness-top TestHarness -faf /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.anno.json -tsaof /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.anno.json -tdf /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/firrtl_black_box_resource_files.top.f -tsf /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.fir -thaof /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.anno.json -hdf /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/firrtl_black_box_resource_files.harness.f -thf /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.fir --infer-rw --repl-seq-mem -c:TestHarness:-o:/prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.mems.conf -thconf /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.mems.conf -td /prj/projects/chipyard-dev/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -ll error
------------------------------------------------------------------------------
Warning: firrtl.Driver is deprecated since 1.2!
Please switch to firrtl.stage.FirrtlMain
------------------------------------------------------------------------------
Computed transform order in: 441.7 ms
[error] (run-main-0) firrtl.FirrtlInternalException: Wrong number of sources or sinks!
[error] firrtl.FirrtlInternalException: Wrong number of sources or sinks!
[error]         at firrtl.Utils$.error(Utils.scala:453)
[error]         at firrtl.passes.wiring.WiringTransform.execute(WiringTransform.scala:86)
[error]         at firrtl.Transform.$anonfun$runTransform$4(Compiler.scala:385)
[error]         at firrtl.Utils$.time(Utils.scala:171)
[error]         at firrtl.Transform$.runTransform(Compiler.scala:203)
[error]         at firrtl.Transform.runTransform(Compiler.scala:385)
[error]         at firrtl.Transform.runTransform$(Compiler.scala:384)
[error] apeout /at firrtl.passes.wiring.WiringTransform.runTransform(WiringTransform.scala:41)
[error] 	at firrtl.passes.memlib.ReplSeqMem.$anonfun$execute$1(ReplaceMemTransform.scala:148)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] 	at firrtl.passes.memlib.ReplSeqMem.execute(ReplaceMemTransform.scala:148)
[error] 	at firrtl.Transform.transform(Compiler.scala:305)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 71 s (01:11), completed Oct 16, 2020, 10:28:42 AM
make: *** [/prj/projects/chipyard-dev/chipyard/common.mk:134: firrtl_temp] Error 1

```

@abejgonzalez 
I'm a beginner/intermediate with Chisel/Scala but it seems like most of these passthroughs are per interface based on the ``CanHave*`` traits. Is there an easy way to just do something like an ``WithAllPassThrough``? I tried the ``WithDontTouch`` which just tied off the ``DigitalTop`` ports.

jerryz123, 2020-10-16:
I was unable to replicate your issue on the `master` branch. After running `make TOP=DigitalTop`, I see
```
$ ls generated-src/chipyard.TestHarness.RocketConfig/
chipyard.TestHarness.RocketConfig.0x0.0.regmap.json         chipyard.TestHarness.RocketConfig.dts                chipyard.TestHarness.RocketConfig.plusArgs       mm.cc               sim_files.f
chipyard.TestHarness.RocketConfig.0x0.1.regmap.json         chipyard.TestHarness.RocketConfig.fir                chipyard.TestHarness.RocketConfig.rom.conf       mm_dramsim2.cc      SimJTAG.cc
chipyard.TestHarness.RocketConfig.0x2000000.0.regmap.json   chipyard.TestHarness.RocketConfig.graphml            chipyard.TestHarness.RocketConfig.top.anno.json  mm_dramsim2.h       SimSerial.cc
chipyard.TestHarness.RocketConfig.0x2010000.0.regmap.json   chipyard.TestHarness.RocketConfig.harness.anno.json  chipyard.TestHarness.RocketConfig.top.fir        mm.h                SimSerial.v
chipyard.TestHarness.RocketConfig.0x40.0.regmap.json        chipyard.TestHarness.RocketConfig.harness.fir        chipyard.TestHarness.RocketConfig.top.mems.conf  plusarg_reader.v    SimUART.cc
chipyard.TestHarness.RocketConfig.0x54000000.0.regmap.json  chipyard.TestHarness.RocketConfig.harness.mems.conf  chipyard.TestHarness.RocketConfig.top.mems.v     remote_bitbang.cc   SimUART.v
chipyard.TestHarness.RocketConfig.0xc000000.0.regmap.json   chipyard.TestHarness.RocketConfig.harness.mems.v     chipyard.TestHarness.RocketConfig.top.v          remote_bitbang.h    TestDriver.v
chipyard.TestHarness.RocketConfig.anno.json                 chipyard.TestHarness.RocketConfig.harness.v          EICG_wrapper.v                                   SimDRAM.cc          TestHarness.anno.json
chipyard.TestHarness.RocketConfig.core.config               chipyard.TestHarness.RocketConfig.json               firrtl_black_box_resource_files.harness.f        SimDRAM.v           uart.cc
chipyard.TestHarness.RocketConfig.d                         chipyard.TestHarness.RocketConfig.l2.json            firrtl_black_box_resource_files.top.f            SimDTM.cc           uart.h
chipyard.TestHarness.RocketConfig.dromajo_params.h          chipyard.TestHarness.RocketConfig.memmap.json        IOCell.v                                         sim_files.common.f
```
Where `.top.v` contains the DigitalTop module. The generated verilator/vcs simulator will also still work correctly. From the snippet you posted, it looks like you may have entered `MODEL=DigitalTop` instead?

On the `dev` branch, the fix is a little more involved. You have to add this additional config fragment
```
class WithNoIOBinders extends Config((site, here, up) => {
  case IOBinders => Map()
})
```
So your config will need to look like the following before `make TOP=DigitalTop` will work. Note that these changes will prevent the resulting VCS/verilator simulator from working correctly. If you don't care about the verilator executable, and just want to grab the verilog, then this approach is fine.

```
class TestRocketConfig extends Config(
  new chipyard.iobinders.WithNoIOBinders ++
  new chipyard.iobinders.WithDontTouchPorts ++
  new chipyard.RocketConfig)
```

I think if you don't want to use any of the ChipTop features, then setting `TOP=DigitalTop` is the easiest way to produce a verilog file with a DigitalTop module with all ports exposed. I don't recommend trying to create passthrough IOBinders for each type.

We'll fix the problem in `dev` so that in the future, `make TOP=DigitalTop` will work without requiring changes.

lsteveol, 2020-10-16:
Thanks for the feedback. I will investigate on my end.

lsteveol, 2020-10-21:
I haven't had much time to debug, but I saw #699 so I'm going to go ahead and close this since I believe that PR will resolve my issues.

Thanks for the feedback

closed 2020-10-21 by lsteveol