ucb-bar/chipyard #700: ClockDividerN DIV parameter has no default

lsteveol, 2020-10-21:
**Impact**: rtl 

**Tell us about your environment:**
*Chipyard Version:* dev/ Hash: 0c46ed167633ebe948ac746f9e0eccd9ab69b361
*OS:*  CentOS7
*Other:*  Using Xcelium 20.09-s001

**What is the current behavior?**
ClockDividerN.sv ``DIV`` parameter has no default

```
module ClockDividerN #(parameter DIV)(output logic clk_out = 1'b0, input clk_in);
                                    |
xmvlog: *E,SVVMAP (/prj/wavious/dev/frontend/chip/wh2/workareas/sbridges/wh240/rtl/sub/chipyard/chipyard/sims/verilator/generated-src/chipyard.TestHarness.WH240RocketConfigTest/ClockDividerN.sv,8|36): Omitting the assignment to constant_param_expression in a param_assignment in a parameter_port_list currently is not supported.
```

**What is the expected behavior?**
Should ``DIV`` be defaulted to a specific value?

**Other information**

I don't have access to VCS/QuestaSim so I don't know if this works on those simulators.

Cadence has the following in their support page

**Definition**

This error message comes because the parameter without default values was not supported in the versions prior to19.09.

**Solution**

In simulator versions 19.09 onward, the support for parameters without default values is provided under the following environments variables:

Value Parameters without default values:  CADENCE_ENABLE_AVSREQ_44905_PHASE_1 
Type Parameters without default values:   CADENCE_ENABLE_AVSREQ_63188_PHASE_1 
Even though declaring parameters without a default value is allowed, the parameters must be assigned a value during instantiation. Else, an elaboration error will occur.

The new elaboration error that occurs when no value is provided during instantiation is NODEFP.

Example:

```
module ma#( parameter int p1 );
   initial $display("p1 = %d", p1);
endmodule

module mb;    
  ma #( )obj(); //needs value
endmodule
```

Error:

```
run ma #( )obj(); //needs value          
|
xmelab: *E,NODEFP (./no_def_param.sv,7|10): No default value is passed to parameter mb.obj@ma<module>.p1

```

lsteveol, 2020-10-21:
It's not a huge problem for me, but it may make sense to change the default.

davidbiancolin, 2020-11-05:
Resolved by #706.

PR #706 (2020-11-03): [clocking] Provide a default div for ClockDividerN sv implementation
Resolves #700.

**Related issue**: #700 

**Type of change**: bug fix 

**Impact**: rtl change
closed 2020-11-05 by davidbiancolin