ucb-bar/chipyard #712: Error making with MegaBoomConfig

Lingrui98, 2020-11-05:
I ran ```make CONFIG=MegaBoomConfig``` in sims/verilator, and it seems to stuck in sbt launch:

```
[warn]                                                                                                                                                                                                                                                           
[warn]  Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.                                                                                                                            
[warn]          com.github.gseitz:sbt-protobuf:0.6.5 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                         
[warn]          com.eed3si9n:sbt-buildinfo:0.9.0 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                             
[warn]          com.eed3si9n:sbt-assembly:0.14.10 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                            
[warn]          org.scalastyle:scalastyle-sbt-plugin:1.0.0 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                   
[warn]          com.eed3si9n:sbt-unidoc:0.4.3 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                                
[warn]          com.typesafe:sbt-mima-plugin:0.7.0 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                           
[warn]          org.scoverage:sbt-scoverage:1.6.1 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                            
[warn]          com.typesafe.sbt:sbt-site:1.4.0 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                              
[warn]          ch.epfl.scala:sbt-scalafix:0.9.15 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                            
[warn]          com.simplytyped:sbt-antlr4:0.8.2 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                                             
[warn]          com.thoughtworks.sbt-api-mappings:sbt-api-mappings:3.0.0 (sbtVersion=1.0, scalaVersion=2.13)                                                                                                                                                     
[warn]                                                                                                                                                                                                                                                           
[warn]  Note: Unresolved dependencies path:                                                                                                                                                                                                                      
[error] sbt.librarymanagement.ResolveException: Error downloading com.github.gseitz:sbt-protobuf;sbtVersion=1.0;scalaVersion=2.13:0.6.5                                                                                                                          
[error]   Not found                                                                                                                                                                                                                                              
[error]   Not found                                                                                                             
[error]   not found: https://repo1.maven.org/maven2/com/github/gseitz/sbt-protobuf_2.13_1.0/0.6.5/sbt-protobuf-0.6.5.pom                                                                                                                                         
[error]   not found: /home/glr/.ivy2/local/com.github.gseitz/sbt-protobuf/scala_2.13/sbt_1.0/0.6.5/ivys/ivy.xml                                                                                                                                                  
[error]   download error: Caught javax.net.ssl.SSLHandshakeException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed (PKIX path validation failed: java.security.cert.CertPathValidatorException: validity che
ck failed) while downloading https://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/com.github.gseitz/sbt-protobuf/scala_2.13/sbt_1.0/0.6.5/ivys/ivy.xml                                                                                            
[error]   not found: https://download.eclipse.org/jgit/maven/com/github/gseitz/sbt-protobuf_2.13_1.0/0.6.5/sbt-protobuf-0.6.5.pom                                                                                                                                
[error]   not found: https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.github.gseitz/sbt-protobuf/scala_2.13/sbt_1.0/0.6.5/ivys/ivy.xml                                                                                                                
[error]   not found: https://repo.typesafe.com/typesafe/ivy-releases/com.github.gseitz/sbt-protobuf/scala_2.13/sbt_1.0/0.6.5/ivys/ivy.xml                                                                                                                        
[error] Error downloading com.eed3si9n:sbt-buildinfo;sbtVersion=1.0;scalaVersion=2.13:0.9.0                                     
[error]   Not found                                                                                                             
[error]   Not found                                                                                                             
[error]   not found: https://repo1.maven.org/maven2/com/eed3si9n/sbt-buildinfo_2.13_1.0/0.9.0/sbt-buildinfo-0.9.0.pom           
[error]   not found: /home/glr/.ivy2/local/com.eed3si9n/sbt-buildinfo/scala_2.13/sbt_1.0/0.9.0/ivys/ivy.xml                                                                                                                                                      
[error]   download error: Caught javax.net.ssl.SSLHandshakeException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed (PKIX path validation failed: java.security.cert.CertPathValidatorException: validity che
ck failed) while downloading https://scalasbt.artifactoryonline.com/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-buildinfo/scala_2.13/sbt_1.0/0.9.0/ivys/ivy.xml                                                                                                
[error]   not found: https://download.eclipse.org/jgit/maven/com/eed3si9n/sbt-buildinfo_2.13_1.0/0.9.0/sbt-buildinfo-0.9.0.pom                                                                                                                                   
[error]   not found: https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-buildinfo/scala_2.13/sbt_1.0/0.9.0/ivys/ivy.xml                                                                                                                    
[error]   not found: https://repo.typesafe.com/typesafe/ivy-releases/com.eed3si9n/sbt-buildinfo/scala_2.13/sbt_1.0/0.9.0/ivys/ivy.xml                                                                                                                            
[... cut ...]
[error]   not found: https://download.eclipse.org/jgit/maven/com/eed3si9n/sbt-assembly_2.13_1.0/0.14.10/sbt-assembly-0.14.10.pom                                                                                                                                 
[error]   not found: https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-assembly/scala_2.13/sbt_1.0/0.14.10/ivys/ivy.xml                                                                                                                   
```

thus I cannot finish the build process

lsteveol, 2020-11-06:
Did you follow the following install instructions for your specific distro?
https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

Lingrui98, 2020-11-06:
> Did you follow the following install instructions for your specific distro?
> https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

Yes, I followed exactly the Ubuntu/Debian part since I'm running on Ubuntu 20.04

jerryz123, 2023-02-17:
This should be resolved with the new conda-based flow.

closed 2023-02-17 by jerryz123