ucb-bar/chipyard #714: How to push a BOOM Core through the physical design (VLSI flow) ?

baichen318, 2020-11-10:
Hi all,

I would like to ask a question how I can push a BOOM core rather than the entire SoC through the physical design (from synthesis to simulation after P&R)?

I have read the documentation (i.e., https://chipyard.readthedocs.io/en/latest/VLSI/Tutorial.html) and I also follow these steps one by one. After all steps done, I didn't encounter problems. However, I met some problems and I could not figure them out when I push a BOOM core rather than the entire SoC through the flow.

I use the command `make buildfile MACROCOMPILER_MODE='--mode synflops' CONFIG=SmallBoomConfig VLSI_TOP=BoomCore` to push a BOOM core rather than the entire SoC to the physical design flow. The synthesis goes well without any error or warning. However, I met a problem in P&R which said that `NET io_ifu_sfence_bits_addr[0] has more than one top-level logical pin which has no physical port (pin geometries) or has not been placed`. Enclosed please find one part of the log file.
My system information: Linux 3.10.0-862.14.4.el7.x86_64 and Innovus version: v16.24-s058_1.

Thank you very much!

[innovus.log](https://github.com/ucb-bar/chipyard/files/5515902/innovus.log)

yunchen-juuuump, 2020-11-10:
Hi,

I will also run BOOM's gate-level simulation in the future. Could I ask you some questions? First, do you modify the Hammer IR? If no, is that mean you just install and initialize the ASAP7? Compared with the tutorial, the only thing you changed is the **make buildfile**'s flags? Last, should I manually move the whole **generated-src** in sims/vcs/ to the vlsi/?

Sorry cannot help you...

Thanks!

baichen318, 2020-11-10:
> Hi,
>
> I will also run BOOM's gate-level simulation in the future. Could I ask you some questions? First, do you modify the Hammer IR? If no, is that mean you just install and initialize the ASAP7? [...]

Hi @yunchen-juuuump 

Thank you for your attention.

In fact, I made some modifications to push the BOOM core through the VLSI flow. I modified Makefile under `vlsi` folder (e.g., remove example.v from the build procedure). Then I remove some unrelated placement constraints for SHA3 accelerator since a single BOOM core rather than the whole SoC doesn't include co-accelerators. After some modifications, my synthesis goes well but problems appear during P&R stage. I still cannot figure them out.

yunchen-juuuump, 2020-11-11:
Hi, @baichen318 

It's weird. I actually can run the gate-level simulation without any error by using DC...But I now also would like to focus on Hammer and ASAP7. It's that possible for you to upload your **vlsi/** to your repo? Maybe I can check the difference between synthesized files generated by DC and ASAP7. Besides, it will be really helpful to me for my next stage work.

Thank you very much!

baichen318, 2020-11-12:
> Hi, @baichen318
>
> It's weird. I actually can run the gate-level simulation without any error by using DC...But I now also would like to focus on Hammer and ASAP7. It's that possible for you to upload [...]

Hi, @yunchen-juuuump 

I have updated my modifications to synthesize a BOOM core under `vlsi` folder. Please check the branch `boom` of the repo. https://github.com/ucb-bar/chipyard.git. In fact, I use genus to do the job rather than DC. If you have set up the Chipyard repo., then you can execute the command `make syn MACROCOMPILER_MODE='--mode synflops' CONFIG=SmallBoomConfig VLSI_TOP=BoomCore` to generate BoomCore.mapped.v by genus tool.

baichen318, 2020-11-12:
Hi @yunchen-juuuump ,

I have updated my modifications to synthesize a single BOOM core. Please check the branch `boom` of the repo. https://github.com/baichen318/chipyard.git. In fact, I use genus rather than DC. When you have set up the Chipyard repo, you may use `make buildfile MACROCOMPILER_MODE='--mode synflops' CONFIG=SmallBoomConfig VLSI_TOP=BoomCore` to synthesize a BOOM core. You may obtain BoomCore.mapped.v generated by genus. Thank you.

yunchen-juuuump, 2020-11-12:
> Hi @yunchen-juuuump ,
>
> I have updated my modifications to synthesize a single BOOM core. Please check the branch `boom` of the repo. https://github.com/baichen318/chipyard.git. In fact, I use genus [...]

@baichen318 Thanks! Did you try to use`SimAXIMem` to generate the boom? When I use DC to run the gate-level simulation, I find the DRAMSim2 may have problems. But they can be solved by using `SimAXIMem`. I'm not sure whether it is related to your issue. Besides, you can also try to use chipyard 1.2.0, which is much more stable I think, although 1.2.0 is not SonicBoom.  By the way, if you want to communicate with me to talk more about the chipyard, you can send me the email(bjcheny1995@gmail.com), although I'm just a beginner at the chipyard. Thanks again!

baichen318, 2020-11-12:
> > Hi @yunchen-juuuump ,
> > I have updated my modifications to synthesize a single BOOM core. Please check the branch `boom` of the repo. https://github.com/baichen318/chipyard.git. In fact, I use genus rather than DC. When you have set up the Chipyard repo, you may use `make buildfile MACROCOMPILER_MODE='--mode synflops' CONFIG=SmallBoomConfig VLSI_TOP=BoomCore` to synthesize a BOOM core. You may obtain BoomCore.mapped.v generated by genus. Thank you.
> 
> @baichen318 Thanks! Did you try to use`SimAXIMem` to generate the boom? When I use DC to run the gate-level simulation, I find the DRAMSim2 may have problems. But they can be solved by using `SimAXIMem`. I'm not sure whether it is related to your issue. Besides, you can also try to use chipyard 1.2.0, which is much more stable I think, although 1.2.0 is not SonicBoom. By the way, if you want to communicate with me to talk more about the chipyard, you can send me the email([bjcheny1995@gmail.com](mailto:bjcheny1995@gmail.com)), although I'm just a beginner at the chipyard. Thanks again!

@yunchen-juuuump  Thank you for your advice! I am also a beginner at Chipyard. I haven't tried to use `SimAXIMem`. In fact, I don't encounter a problem of `DRAMSim2`. Maybe you could check the environment? Thank you again.

baichen318, 2020-12-02:
Hi all,

I can now push BoomCore through the VLSI flow by using the command `make par MACROCOMPILER_MODE='--mode synflops' CONFIG=SmallBoomConfig VLSI_TOP=BoomCore`. After this step, I can get the GDSII file of BoomCore. You can visit the branch `boom` of my repo : https://github.com/baichen318/chipyard.git to view the modifiled Hammer IR.

Thank you very much!

closed 2020-12-02 by baichen318