ucb-bar/chipyard #723: sbt compile too long

pxs7, 2020-11-21:
when make verilog in vlsi dir, the following infomation had kept too long  without change

  100.0% [##########] 2.3 MiB (9.2 KiB / s)
https://repo1.maven.org/maven2/org/scala-sbt/zinc-compile_2.12/1.3.0/zinc-compile_2.12-1.3.0.jar
  100.0% [##########] 16.6 KiB (12.2 KiB / s)
https://repo1.maven.org/maven2/org/scala-sbt/tasks_2.12/1.3.2/tasks_2.12-1.3.2.jar
  100.0% [##########] 83.6 KiB (11.7 KiB / s)
https://repo1.maven.org/maven2/org/scala-sbt/main_2.12/1.3.2/main_2.12-1.3.2.jar
  100.0% [##########] 2.4 MiB (10.0 KiB / s)
https://repo1.maven.org/maven2/org/scala-sbt/protocol_2.12/1.3.2/protocol_2.12-1.3.2.jar
  100.0% [##########] 432.9 KiB (12.2 KiB / s)
https://repo1.maven.org/maven2/org/scala-sbt/zinc-persist_2.12/1.3.0/zinc-persist_2.12-1.3.0.jar
  100.0% [##########] 1.7 MiB (8.5 KiB / s)
https://repo1.maven.org/maven2/org/scala-sbt/sbt/1.3.2/sbt-1.3.2.jar
  100.0% [##########] 64.9 KiB (252 B / s)
https://repo1.maven.org/maven2/com/github/os72/protoc-jar/3.5.1.1/protoc-jar-3.5.1.1.jar
  100.0% [##########] 9.1 MiB (8.8 KiB / s)

alonamid, 2020-11-21:
How long is "long"? Did you  have any issues with your internet connections? what is your environment (OS, etc.)? are you on a network file system?

pxs7, 2020-11-22:
it takes about one hour， nothing changed。 

OS： ubuntu

alonamid, 2021-01-30:
Is this issue persisting? Are you able to reproduce it on another machine?
Is a network file system involved anywhere in the process?

jerryz123, 2023-02-17:
Closing due to inactivity.

closed 2023-02-17 by jerryz123