ucb-bar/chipyard #724: Rocket Chip SOC Modules Connection

Uzair540, 2020-11-21:
I am mapping Rocket chip (chips alliance repository) on FPGA and I am encountering an error of many Outputs as I am not connecting SOC blocks to it, I have  to connect my Rocket chip to different SOC modules in order to map to FPGA.Can you please elaborate me Is that possible through this chip yard repository that I can run rocket chip directly to FPGA by generating it .v file. Or should I have to write additional code (wrapper file) in order to connect rocket chip to its  SOC module.
What is the difference between chip yard  rocket chip Repository and chips alliance repository.

alonamid, 2020-11-21:
In order to run a Chipyard configuration as an FPGA prototype (in contrast to FireSim FPGA-accelerated simulation) you need to be able to map the IOs to the FPGAs IOs and peripherals. Chipyard currently does not support that. We hope to have support for this in the next Chipyard release based on the SiFive fpga-shells repository (https://github.com/sifive/fpga-shells). If you would like to attempt doing a similar thing yourself before the next Chipyard release, you will indeed need to write additional code, and I would recommend studying the SiFive fpga-shells repository.

Chipyard is not a "rocket-chip repository". Chipyard uses rocket-chip as a generator library, integrated together with several other generator projects and tools (BOOM, Hwacha, Gemmini, FireSim, Hammer, just to name a few). Rocket-chip is a submodule within Chipyard, and that submodule points to the chips-alliance repository. So the answer to your questions is that there is no difference between the rocket-chip within Chipyard and the rocket-chip Chips Alliance repository, because the Chipyard rocket-chip submodule points the the Chips Alliance rocket-chip repository.

Uzair540, 2020-11-23:
Should I have to write code in Scala/Chisel or Verilog for connection of these modules ?

lsteveol, 2020-11-23:
The `IOCell`s that are included at the `ChipTop` can be replaced with FPGA IOs. Depending on your configuration you will have interfaces for DRAM, FBUS, etc. These would need to be connected to a respective DRAM/On Chip Memory or some other AXI bus. 

I would suggest using Verilog to begin with if you aren't comfortable with Chisel/Scala.

closed 2020-12-07 by alonamid