ucb-bar/chipyard #726: Hwacha simulation error using VCS

XiayuanW, 2020-11-27:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* <-- 1.3.0, branch: master, Hash: f387c4b -->
*OS:* <-- `Linux 3.10.0-1062.31.2.el7.x86_64 x86_64` -->
*Other:* 
https://chipyard.readthedocs.io/en/latest/Simulation/Software-RTL-Simulation.html
use VCS to run the simulation
`make SUB_PROJECT=hwacha run-asm-tests`
**What is the current behavior?**
make: *** [../chipyard/sims/vcs/output/freechips.rocketchip.system.TestHarness.HwachaConfig/(some_file_name).out] Error 255

In the .out file:
simv-freechips.rocketchip.system-HwachaConfig: ../fesvr/dtm.cc:61: uint32_t dtm_t::do_command(dtm_t::req): Assertion `resp_buf.resp == 0' failed.
**What is the expected behavior?**

**Other information**

jerryz123, 2020-11-30:
I have seen similar errors in the past when there is mismatch between the FESVR version and the Chipyard/rocketchip version. Can you confirm what commit of riscv-isa-sim you are using?

XiayuanW, 2020-11-30:
Hi, 

Thanks! 

It's `esp-isa-sim at commit 13384cac1e54828200067ff890f564a505a4ebb3`.
https://github.com/ucb-bar/esp-isa-sim/commit/13384cac1e54828200067ff890f564a505a4ebb3

In addition, we could run the rocketchip simulation correctly by running 
`make run-asm-tests`

But we met the problem when running the Hwacha simulation.

closed 2020-12-13 by XiayuanW