ucb-bar/chipyard #727: Integration with ETH Zürich clusters

mbelda, 2020-11-28:
Hello, 

I am working allready with Chipyard but now I want to integrate it with an Alveo FPGA Cluster from ETH Zürich. 

Is it possible or Chipyard integration is only implemented for FireSim right now?

Thank you very much!
Regards, MªJosé.

alonamid, 2020-11-30:
Chipyard currently only supports FireSim.
There is ongoing work to support FPGA prototyping on local FPGAs (hopefully in the next Chipyard release). The initial one will likely be a VCU118.
There is interest in support FireSim simulations on local Alveo's, but that is not something that is currently implemented.

abejgonzalez, 2020-12-01:
To add a bit more here, if you do any work with https://github.com/sifive/fpga-shells to integrate the Alveo boards, then we might use/help with that in the future.

alonamid, 2021-01-30:
You might want to have a look at the local FPGA support for VCU118 in the Chipyard 1.4 release.
You could try to build on top of that with fpga-shells for Alveo boards.

Closing this issue for lack of additional activity.

closed 2021-01-30 by alonamid