ucb-bar/chipyard #730: How to speed up compilation

hz1490919302, 2020-12-02:
I use Ubuntu 20.04 with 8G of memory. Every time I compile with `make config = smallboomconfig`, the speed is slow. I only change the _/chipyard/generator/boom_ folder, with other files do not change，and only want to get the generated boom simulator. 
How can I speed up the compilation process, such as setting compilation options or other optimization measures? thank you

jerryz123, 2020-12-02:
How long is that compile taking?
You may want to try compiling with Bloop instead of SBT. Just add `ENABLE_BLOOP=1` like `make CONFIG=SmallBoomConfig ENABLE_BLOOP=1`

hz1490919302, 2020-12-02:
Each compilation takes about 30 minutes, and the _tapeout_ process takes a lot of time.

I use _make - j_ and it doesn't seem to speed it up. Google told me that _ccache_ might be a good idea. I will also try enable_ bloop. I hope they can speed up my operation.

jerryz123, 2020-12-02:
The tapeout.GenerateTopAndHarness process is expected to take a lot of time.

abejgonzalez, 2020-12-03:
Another thing to check is your JAVA and SBT settings. Sometimes if you are downloading libraries/jars your cache is over the network leading to low performance. You can see some of the things we set here for CI:

https://github.com/ucb-bar/chipyard/blob/70d41996dd738cd4b32e2e845644d62c9fbe14be/.circleci/defaults.sh#L37

In the future, we are looking to support SBT's "thin client" which is similar to BLOOP.

hz1490919302, 2020-12-03:
> How long is that compile taking?
> You may want to try compiling with Bloop instead of SBT. Just add `ENABLE_BLOOP=1` like `make CONFIG=SmallBoomConfig ENABLE_BLOOP=1`

Hello, Chipyard's master branch does not support bloop now, so which branch supports BLOOp?

abejgonzalez, 2020-12-03:
The `dev` branch currently supports `bloop`. But as Jerry stated, after you compile all the sources for the project, the FIRRTL step (`GenerateTopAndHarness`) will take the longest and that cannot be changed by moving to `bloop` over `sbt`.

hz1490919302, 2020-12-04:
> The `dev` branch currently supports `bloop`. But as Jerry stated, after you compile all the sources for the project, the FIRRTL step (`GenerateTopAndHarness`) will take the longest and that cannot be changed by moving to `bloop` over `sbt`.

Thank you for your answer, I will close this issue.

closed 2020-12-04 by hz1490919302