ucb-bar/chipyard #733: Error when compiling with "make debug CONFIG=SmallNVDLARocketConfig"

JunningWu, 2020-12-03:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* branch master, Hash: cca3cd0 
*OS:* Linux chipyard-vm 5.4.0-49-generic #53~18.04.1-Ubuntu SMP Mon Sep 21 14:12:39 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
*Other:* Vmware

**What is the current behavior?**
when running the make command: `make debug CONFIG=SmallNVDLARocketConfig` under `sims/vcs`, I get the following error message:
```
make[1]: /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/../../../../../scripts/insert-includes.py: Command not found
Makefile:47: recipe for target '/home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/nvdla_small.preprocessed.v' failed
make[1]: Leaving directory '/home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources'
make[1]: *** [/home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/nvdla_small.preprocessed.v] Error 127
[error] java.lang.IllegalArgumentException: requirement failed: Failed to run pre-processing step
[error]         ...
[error]         at nvidia.blocks.ip.dla.nvdla.<init>(nvdla.scala:98)
[error]         at nvidia.blocks.dla.NVDLA$$anon$1.$anonfun$u_nvdla$1(NVDLA.scala:88)
[error]         at chisel3.Module$.do_apply(Module.scala:52)
[error]         at nvidia.blocks.dla.NVDLA$$anon$1.<init>(NVDLA.scala:88)
[error]         at nvidia.blocks.dla.NVDLA.module$lzycompute(NVDLA.scala:86)
[error]         at nvidia.blocks.dla.NVDLA.module(NVDLA.scala:86)
[error] hipyard at nvidia.blocks.dla.NVDLA.module(NVDLA.scala:21)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:166)
[error] 	at chisel3.Module$.do_apply(Module.scala:52)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:166)
[error] 	at scala.collection.immutable.List.flatMap(List.scala:338)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:164)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:163)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImp.instantiate(LazyModule.scala:193)
[error] 	at freechips.rocketchip.diplomacy.LazyModuleImp.<init>(LazyModule.scala:194)
[error] 	at freechips.rocketchip.subsystem.BareSubsystemModuleImp.<init>(BaseSubsystem.scala:31)
[error] 	at freechips.rocketchip.subsystem.BaseSubsystemModuleImp.<init>(BaseSubsystem.scala:130)
[error] 	at chipyard.SubsystemModuleImp.<init>(Subsystem.scala:98)
[error] 	at chipyard.SystemModule.<init>(System.scala:37)
[error] 	at chipyard.DigitalTopModule.<init>(DigitalTop.scala:33)
[error] 	at chipyard.DigitalTop.module$lzycompute(DigitalTop.scala:30)
[error] 	at chipyard.DigitalTop.module(DigitalTop.scala:30)
[error] 	at chipyard.DigitalTop.module(DigitalTop.scala:15)
[error] 	at chipyard.BaseChipTop.$anonfun$system$2(ChipTop.scala:71)
[error] 	at chisel3.Module$.do_apply(Module.scala:52)
[error] 	at chipyard.BaseChipTop.$anonfun$system$1(ChipTop.scala:71)
[error] 	at chisel3.withClockAndReset$.apply(MultiClock.scala:25)
[error] 	at chipyard.BaseChipTop.<init>(ChipTop.scala:71)
[error] 	at chipyard.ChipTop.<init>(ChipTop.scala:120)
[error] 	at chipyard.BuildTop$$anonfun$$lessinit$greater$1.$anonfun$apply$1(TestHarness.scala:14)
[error] 	at chisel3.Module$.do_apply(Module.scala:52)
[error] 	at chipyard.BuildTop$$anonfun$$lessinit$greater$1.apply(TestHarness.scala:14)
[error] 	at chipyard.BuildTop$$anonfun$$lessinit$greater$1.apply(TestHarness.scala:14)
[error] 	at chipyard.TestHarness.<init>(TestHarness.scala:25)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:31)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 19 s, completed Dec 3, 2020, 6:00:38 PM
/home/chipyard/work/chipyard/chipyard/common.mk:70: recipe for target 'generator_temp' failed
```

**What is the expected behavior?**

**Other information**
After running `chmod +x insert-includes.py` according to [](https://stackoverflow.com/questions/59513258/why-is-the-following-sh-script-saying-py-command-not-found-when-the-script-is), the error remains.

abejgonzalez, 2020-12-03:
I cloned a fresh repo at that commit, ran the same command, and I was able to pass that point. 

You can diagnose this a bit more by just running the NVDLA makefile:
```
make -C generators/nvdla/src/main/resources default NVDLA_TYPE=small
```

`insert-includes.py` is found in the `scripts` area (like you said) so I'm not exactly sure whats going on here.

JunningWu, 2020-12-04:
I get the same error when running the NVDLA makefile:
```
chipyard@chipyard-vm:~/work/chipyard/chipyard/generators/nvdla/src/main/resources$ pwd
/home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources
```

```
chipyard@chipyard-vm:~/work/chipyard/chipyard/generators/nvdla/src/main/resources$ make default NVDLA_TYPE=small
mkdir -p /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/
cat /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/defines/defs.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/top/NV_NVDLA_partition_o.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/top/NV_NVDLA_partition_c.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/top/NV_NVDLA_partition_m.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/top/NV_NVDLA_partition_a.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/top/NV_NVDLA_partition_p.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/apb2csb/NV_NVDLA_apb2csb.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDMA_WT_8ATMM_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDP_DP_data_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDP_DP_intpinfo_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDP_DP_sumpd_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDP_RDMA_lat_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDP_RDMA_ro_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_CDP_WDMA_dat_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_PDP_RDMA_lat_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_PDP_RDMA_ro_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_PDP_SDPIN_ro_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_PDP_WDMA_DAT_fifo.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_BRDMA_cq_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_BRDMA_lat_fifo_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_ERDMA_cq_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_ERDMA_lat_fifo_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_MRDMA_cq_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_MRDMA_EG_lat_fifo_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_NRDMA_cq_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/fifos/NV_NVDLA_SDP_NRDMA_lat_fifo_lib.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_assembly_buffer.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_assembly_ctrl.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_CALC_int8.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_calculator.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_delivery_buffer.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_delivery_ctrl.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_dual_reg.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_regfile.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_single_reg.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_CACC_slcg.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/cacc/NV_NVDLA_cacc.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/car/NV_NVDLA_core_reset.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/car/NV_NVDLA_reset.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small/vmod/nvdla/car/NV_NVDLA_sync3d_c.v /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/vsrc/small [... cut ...]
make: /home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/../../../../../scripts/insert-includes.py: Command not found
Makefile:47: recipe for target '/home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/nvdla_small.preprocessed.v' failed
make: *** [/home/chipyard/work/chipyard/chipyard/generators/nvdla/src/main/resources/nvdla_small.preprocessed.v] Error 127
```

BTW, I have set up the repo according to [Initial Repository Setup](https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html)

JunningWu, 2020-12-04:
After running the `./insert-includes.py`, I have found the reason.

`/usr/bin/python` is not exist.

I only have `python3`.

Solved.

closed 2020-12-04 by JunningWu

krabo0om, 2023-06-05:
I have just experienced the same issue. Our system does not have a `/usr/bin/python`, only `python2` and `python3`. The actual solution is to change the shebang to `#!/usr/bin/env python` https://github.com/ucb-bar/chipyard/blob/518dacdb524a591dab3263152bcc849e8e0106ab/scripts/insert-includes.py#L1

Do we need a PR or can someone just push the fix?