ucb-bar/chipyard #735: Release 1.3.0 libgloss installation error

SaltedFishLZ, 2020-12-08:
**Impact**: unknown | other
Hi, I am following the tutorial to set up chipyard (https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html), but it seems `libgloss` cannot be configured correctly and my installation is aborted.

**Tell us about your environment:**
*Chipyard Version:* 1.3.0, Hash: c576a7e76717376b7a2cbbc0a531f51382ff87f1
*OS:* `Ubuntu 18.04 Docker Image` 
*Other:* 
https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

**What is the current behavior?**

After running `./scripts/build-toolchains.sh riscv-tools`, the compilation process will be aborted due to an error, which shows "configure: error: The target architecture of the C compiler does not appear to be RISC-V. If cross-compiling, specify a target triplet such as '--host=riscv64-unknown-elf'."

**What is the expected behavior?**

**Other information**

Here is my log
```
riscv64-unknown-elf-objdump --disassemble-all --disassemble-zeroes --section=.text --section=.text.startup --section=.text.init --section=.data rv64mi-p-sbreak > rv64mi-p-sbreak.dump
make[1]: Leaving directory '/chipyard/toolchains/riscv-tools/riscv-tests/build/isa'
==>  Installing riscv-tests
mkdir -p benchmarks
mkdir -p isa
/usr/bin/make -C benchmarks -f /chipyard/toolchains/riscv-tools/riscv-tests/build/../benchmarks/Makefile src_dir=/chipyard/toolchains/riscv-tools/riscv-tests/build/../benchmarks XLEN=64
/usr/bin/make -C isa -f /chipyard/toolchains/riscv-tools/riscv-tests/build/../isa/Makefile src_dir=/chipyard/toolchains/riscv-tools/riscv-tests/build/../isa XLEN=64
make[1]: Entering directory '/chipyard/toolchains/riscv-tools/riscv-tests/build/benchmarks'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/chipyard/toolchains/riscv-tools/riscv-tests/build/benchmarks'
make[1]: Entering directory '/chipyard/toolchains/riscv-tools/riscv-tests/build/isa'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/chipyard/toolchains/riscv-tools/riscv-tests/build/isa'
install -d /chipyard/riscv-tools-install/riscv64-unknown-elf/share/riscv-tests/isa
install -d /chipyard/riscv-tools-install/riscv64-unknown-elf/share/riscv-tests/benchmarks
install -p -m 644 `find isa -maxdepth 1 -type f` /chipyard/riscv-tools-install/riscv64-unknown-elf/share/riscv-tests/isa
install -p -m 644 `find benchmarks -maxdepth 1 -type f` /chipyard/riscv-tools-install/riscv64-unknown-elf/share/riscv-tests/benchmarks
=>  Starting libgloss build
==>  Initializing libgloss submodule
Cloning into '/chipyard/toolchains/libgloss'...
Submodule path 'toolchains/libgloss': checked out '04b249764b1bc0b784b9f7ff422084f4b53a07ef'
==>  Configuring libgloss
checking for riscv64-unknown-elf-gcc... gcc-5
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc-5 accepts -g... yes
checking for gcc-5 option to accept ISO C89... none needed
checking for riscv64-unknown-elf-ar... riscv64-unknown-elf-ar
checking for riscv64-unknown-elf-size... riscv64-unknown-elf-size
configure: error: in `/chipyard/toolchains/libgloss/build':
configure: error: The target architecture of the C compiler does not appear to be RISC-V.
If cross-compiling, specify a target triplet such as '--host=riscv64-unknown-elf'.
See `config.log' for more details
```

a0u, 2020-12-13:
Overriding `CC=gcc-5 CXX=g++-5` in the environment forces `configure` to pick up the native gcc instead of the cross-compiler in the `AC_PROG_CC` check.  This behavior is not unique to libgloss-htif; you'll find that it breaks cross-compilation of other autoconf-based packages.

But you do bring up an uncommon use case where it could be useful to independently specify configure variables for the host toolchain build, without affecting the build process for RISC-V target code.  One possible solution is to add `--host-cc` and `--host-cxx` options to the `build-toolchains.sh` script.

a0u, 2020-12-13:
```sh
CC= CXX=  SRCDIR="$(pwd)/toolchains" module_all libgloss --prefix="${RISCV}/riscv64-unknown-elf" --host=riscv64-unknown-elf
``` 
would also fix this.

closed 2020-12-13 by SaltedFishLZ