ucb-bar/chipyard #737: Bump Scala versions to 2.12.12

timsnyder-siv, 2020-12-09:
There is a desire to use portions of chipyard and it's submodules (notably firesim) with vscode+metals.  To do so, @azidar points out that metals wants at least version 2.12.10.

Would it be possible to bump the scala version consistantly to 2.12.latest for the upcoming Chipyard relase?

davidbiancolin, 2020-12-09:
I think most of the work for this will be done in 3.4 bump.

colinschmidt, 2020-12-09:
see also: https://github.com/ucb-bar/plsi-mdf/pull/6
and: https://github.com/ucb-bar/barstools/pull/87#issuecomment-742004946

timsnyder-siv, 2020-12-09:
2.12.10 is probably fine.  When I tried 2.12.12, it looked like the macro-paradise plugin version would also have to be updated to 2.1.1.

alonamid, 2021-01-30:
Done in the Chipyard 1.4 release

closed 2021-01-30 by alonamid